natario1/CameraView #579: How do focus on an area rather than a point?

shedeed1, 2019-09-04:
### How do I?
Ok so I've been trying to request focus before I take a specific picture because my application requires taking pictures of highest detail possible. Here is what I do:

`camera.startAutoFocus((camera.getX() + camera.getWidth()) / 2, (camera.getY() + camera.getHeight()) / 2);`

Then on its callback I take my picture:

```
public void onAutoFocusEnd(boolean successful, @NonNull PointF point) {
                if (successful)
                    camera.takePicture();
}
```

The problem with this approach that it focuses on a single point and (sometimes) blurs out other areas so I lose a lot of detail in the picture.

My question is: is there an approach to focus on the whole picture before I take it? Or focus on a certain rectangle in the picture?

Thank you for your hard work in this library, it's really one of the best libraries I have ever used on Android in my 2 years of development.

### Version used
2.2.0

natario1, 2019-09-04:
Thanks! 

I'm not sure about 2.2.0, but the current master branch already does this for you. It will be available in the next version.

About focusing to a rectangle you might want to add this idea to #487, maybe one day we'll support this.

shedeed1, 2019-09-05:
Oh ok. Waiting for the next release then! Thanks!

closed 2019-09-11 by natario1