natario1/CameraView #581: restart record video has error。

Jack-Xiao, 2019-09-05:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.2.0*
- Camera engine used: *camera2/both*
- Reproducible in official demo app: *no*
- Device / Android version: *xiaomi note7 pro, API 28*

when start record video, it's ok
step1: start record video  camera.takevideo(File(path))
step2: stop record video camera.stopVideo()
step3: onVideoTaken callback method, take a replace fragment override 
step4: close the fragment, the camera preview is ok
step5: start record video again, it has break.
error log:
E/CameraCaptureSession: Session 1: Exception while stopping repeating: 
    android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): The camera device has encountered a serious error
        at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:2448)
        at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:1116)
        at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:527)
        at com.otaliastudios.cameraview.engine.Camera2Engine.onStopBind(Camera2Engine.java:606)
        at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:490)
        at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:487)
        at com.otaliastudios.cameraview.engine.Step$4.then(Step.java:155)
        at com.otaliastudios.cameraview.engine.Step$4.then(Step.java:150)
        at com.google.android.gms.tasks.zzf.run(Unknown Source:2)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:130)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute(WorkerHandler.java:103)
        at com.google.android.gms.tasks.zze.onComplete(Unknown Source:6)
        at com.google.android.gms.tasks.zzr.zza(Unknown Source:19)
        at com.google.android.gms.tasks.zzu.zze(Unknown Source:132)
        at com.google.android.gms.tasks.zzu.continueWithTask(Unknown Source:74)
        at com.otaliastudios.cameraview.engine.Step.doStop(Step.java:150)
        at com.otaliastudios.cameraview.engine.Step.doStop(Step.java:145)
        at com.otaliastudios.cameraview.engine.CameraEngine.stopBind(CameraEngine.java:487)
        at com.otaliastudios.cameraview.engine.CameraEngine.access$600(CameraEngine.java:120)
        at com.otaliastudios.cameraview.engine.CameraEngine$18$1$3.then(CameraEngine.java:805)
        at com.otaliastudios.cameraview.engine.CameraEngine$18$1$3.then(CameraEngine.java:802)
        at com.google.android.gms.tasks.zzf.run(Unknown Source:2)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:130)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute(WorkerHandler.java:103)
        at com.google.android.gms.tasks.zze.onComplete(Unknown Source:6)
        at com.google.android.gms.tasks.zzr.zza(Unknown Source:19)
        at com.google.android.gms.tasks.zzu.zze(Unknown Source:132)
        at com.google.android.gms.tasks.zzu.continueWithTask(Unknown Source:74)
        at com.otaliastudios.cameraview.engine.CameraEngine$18$1.call(CameraEngine.java:802)
        at com.otaliastudios.cameraview.engine.CameraEngine$18$1.call(CameraEngine.java:799)
        at com.otaliastudios.cameraview.engine.Step$4.then(Step.java:155)
        at com.otaliastudios.cameraview.engine.Step$4.then(Step.java:150)
        at com.google.android.gms.tasks.zzf.run(Unknown Source:2)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:130)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute(WorkerHandler.java:103)
        at com.google.android.gms.tasks.zze.onComplete(Unknown Source:6)
        at com.google.android.gms.tasks.zzr.zza(Unknown Source:19)
        at com.google.android.gms.tasks.zzu.zze(Unknown Source:132)
        at com.google.android.gms.tasks.zzu.continueWithTask(Unknown Source:74)
        at com.otaliastudios.cameraview.engine.Step.doStop(Step.java:150)
        at com.otaliastudios.cameraview.engine.Step.doStop(Step.java:145)
        at com.otaliastudios.cameraview.engine.CameraEngine$18.run(CameraEngine.java:799)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:201)
        at android.os.HandlerThread.run(HandlerThread.java:65)

natario1, 2019-09-11:
Please post your code.

jianjiankedizai, 2019-09-12:
CameraView version: 2.2.0
Camera engine used: *camera2
Reproducible in official demo app: no
Device / Android version: xiaomi hongminote2, API 25

 android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): The camera device has encountered a serious error
        at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:2023)
        at android.hardware.camera2.impl.CameraDeviceImpl.configureStreamsChecked(CameraDeviceImpl.java:399)
        at android.hardware.camera2.impl.CameraCaptureSessionImpl$IdleDrainListener.onDrained(CameraCaptureSessionImpl.java:745)
        at android.hardware.camera2.utils.TaskDrainer$1.run(TaskDrainer.java:210)
        at android.os.Handler.handleCallback(Handler.java:754)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:163)
        at android.os.HandlerThread.run(HandlerThread.java:61)

Similar error，
this is error happen when  first time call take video

Jack-Xiao, 2019-09-18:
in xml:
<com.otaliastudios.cameraview.CameraView
                android:id="@+id/camera"
                android:layout_width="match_parent"
                android:layout_height="match_parent"
                android:layout_gravity="center"
                android:keepScreenOn="true"
                app:cameraAudio="on"
                app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
                app:cameraEngine="camera2"
                app:cameraExperimental="true"
                app:cameraFacing="back"
                app:cameraFlash="auto"
                app:cameraGestureLongTap="none"
                app:cameraGesturePinch="zoom"
                app:cameraGestureScrollHorizontal="filterControl1"
                app:cameraGestureScrollVertical="exposureCorrection"
                app:cameraGestureTap="autoFocus"
                app:cameraGrid="off"
                app:cameraMode="video"
                app:cameraPlaySounds="true"
                app:cameraPreview="glSurface" />
######
start record code:
camera.takePictureSnapshot()
camera.takeVideo(File(path))
#######
and stop video:
camera.stopVideo()
#######
in CameraListener
override fun onVideoTaken(result: VideoResult) {
            super.onVideoTaken(result)
            val imagePath = AppConstant.getVideoPath("image_url.png")
            platyFragment = PlayFragment.getInstance(path, imagePath, result)
            val transaction = childFragmentManager.beginTransaction()
            transaction.add(cn.com.wewave.bibi.R.id.fragment_play_instance, platyFragment)
                .commit()

            fragment_record.visibility = View.GONE
            fragment_play.visibility = View.VISIBLE
        }

#######
May be the reason is childFragment play the record video。However, when i replace activity，it also happen.

Jack-Xiao, 2019-09-20:
@jianjiankedizai  have resolver for this issue.

jianjiankedizai, 2019-09-23:
@Jack-Xiao ok, thanks

closed 2019-10-20 by stale[bot]