natario1/CameraView #582: Exception: Filters are only supported by the GL_SURFACE preview. Current:SURFACE

uncledoc, 2019-09-05:
How to reproduce:
1. configure CameraView in a layout
```xml
<com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="wrap_content"
        android:keepScreenOn="true"
        app:cameraAudio="off"
        app:cameraMode="picture"
        app:cameraPlaySounds="false"
        app:cameraPreview="surface" />
```
2. Run Activity with the xml layout

p.s. Be attentive, i set  app:cameraPreview="**surface**"

dylanrjames, 2019-09-10:
Same thing happens with `texture`.

It appears that `setFilter` in `CameraView` is asking that `NoFilter` be sent to clear, though it doesn't excuse `NoFilter` when it checks for `surface` or `texture` compatibility.

Since `initialize()` always calls `setFilter()`, even when the `FilterParser` returns `NoFilter` as part of the default case, `surface` and `texture` previews can't even be initialized and are entirely broken as far as I can tell.

natario1, 2019-09-11:
Thanks for investigating @dylanrjames !

commit 580c373c (2019-09-11): Fix #582
commit 5619dbdc (2019-10-01): Small improvements (#588)

* Abort actions on exit

* Fix VideoRecorder state

* Add missing metering tests

* Fix default vertex shader definition

* Fix #582

* Make Camera2Engine more open

* Add timestampUs in Filter.draw signature

* Add timestampUs in BaseFilter.onPreDraw, onDraw, onPostDraw

* Change README

* Fix FrameProcessing segmentation fault

* Fix #590 by throwing

* Fix StackOverflowErrors in AudioMediaEncoder (Fixes #594)

* Enforce 100 character limits

* Improve encoder selection

* Simplify DeviceEncoders class

* Improve overlay add/remove

* Add changelog info

* Add more [... cut]
closed 2019-10-01 by natario1