natario1/CameraView #585: Camera Preview not shown on some devies

MoustafaElsaghier, 2019-09-08:
### Describe the bug
I'm working on an android app that uses the lib as a small box with some specification of zoom, rotates and scales the app works well on many devices but I get some feedback from clients that the box did not open on their devices and i can't recognize why that happens.

- CameraView version: *1.6.0*
- Camera engine used: *both*
- Reproducible in official demo app: *my devices no*
- Device / Android version: *redmi note 7, API 28*

### To Reproduce
i open the activity to get the camera box upon a background

### Expected behavior
expected that the camera opens normally as on other devices
### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
<com.otaliastudios.cameraview.CameraView
            android:id="@+id/back_camera_CameraView"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:layout_gravity="center"
            app:cameraCropOutput="true"
            app:cameraPlaySounds="false" />
```

I know that is not enough details to discover what happens but hope you could help me

shedeed1, 2019-09-09:
You could start by updating to the latest version first.

MoustafaElsaghier, 2019-09-09:
client refuse to do that as it requires many changes and he not accept that -_-

natario1, 2019-09-10:
If we were to found a bug and fix it, that would be published in a new version anyway @MoustafaElsaghier ! But I'm sure this has already been fixed at some point.

MoustafaElsaghier, 2019-09-10:
sorry but, i misunderstand what you mean .. is it fixed in new versions or what?

natario1, 2019-09-10:
I don't know! I think it is already fixed, but you should upgrade your app to discover.

closed 2019-09-11 by natario1

MoustafaElsaghier, 2019-09-18:
i did that in a copy of my project and now client camera box opens with a black screen