natario1/CameraView #592: "Height not supported after adjustment."

brnogz, 2019-09-15:
### How do I?
Used cameraview as in the example code below. 
```
<com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:keepScreenOn="true"
            app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
            app:cameraGesturePinch="zoom"
            app:cameraGestureTap="autoFocus"
            app:cameraMode="picture" />
```
When i submit app google play store, pre-release test show exception on nokia 1. stack trace and device specs is also added below.
```
FATAL EXCEPTION: CameraViewEngine
Process: lol.playz.android, PID: 19409
java.lang.RuntimeException: Height not supported after adjustment. Desired:96 Range:[144, 144]
	at com.otaliastudios.cameraview.internal.DeviceEncoders.getSupportedVideoSize(DeviceEncoders.java:209)
	at com.otaliastudios.cameraview.video.FullVideoRecorder.onPrepareMediaRecorder(FullVideoRecorder.java:100)
	at com.otaliastudios.cameraview.video.Full1VideoRecorder.onPrepareMediaRecorder(Full1VideoRecorder.java:44)
	at com.otaliastudios.cameraview.video.FullVideoRecorder.prepareMediaRecorder(FullVideoRecorder.java:45)
	at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:170)
	at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:62)
	at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:339)
	at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1151)
	at android.os.Handler.handleCallback(Handler.java:790)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:164)
	at android.os.HandlerThread.run(HandlerThread.java:65)
```
**Screen Size** 480 × 854
**Screen Density** (dpi) 240
**Model Name** Nokia 1
**RAM** 1024 MB
**Manufacturer** Nokia
**OpenGL ES** Version 3.1
**Android Version** Android 8.1 (Go edition)
**Native Platform** armeabi-v7a
**Locale** tr
**CPU Make** Mediatek
**CPU Model** MT6737M

### Version used
2.2.0

How can I make it work on nokia 1 ? I want to make it on low end devices, so I'm concerned about this.

adaonder, 2019-09-16:
Nokia 3 Android 9 also has the same error. No solution?
Date: 15 Eyl 2019 15:09:00

Version used
2.2.0

My Log:

```
Fatal Exception: java.lang.RuntimeException: Height not supported after adjustment. Desired:96 Range:[144, 144]
       at com.otaliastudios.cameraview.internal.DeviceEncoders.getSupportedVideoSize + 209(DeviceEncoders.java:209)
       at com.otaliastudios.cameraview.video.FullVideoRecorder.onPrepareMediaRecorder + 100(FullVideoRecorder.java:100)
       at com.otaliastudios.cameraview.video.Full1VideoRecorder.onPrepareMediaRecorder + 44(Full1VideoRecorder.java:44)
       at com.otaliastudios.cameraview.video.FullVideoRecorder.prepareMediaRecorder + 45(FullVideoRecorder.java:45)
       at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart + 170(FullVideoRecorder.java:170)
       at com.otaliastudios.cameraview.video.VideoRecorder.start + 62(VideoRecorder.java:62)
       at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo + 339(Camera1Engine.java:339)
       at com.otaliastudios.cameraview.engine.CameraEngine$23.run + 1151(CameraEngine.java:1151)
       at android.os.Handler.handleCallback + 873(Handler.java:873)
       at android.os.Handler.dispatchMessage + 99(Handler.java:99)
       at android.os.Looper.loop + 193(Looper.java:193)
       at android.os.HandlerThread.run + 65(HandlerThread.java:65)
```

brnogz, 2019-09-18:
@natario1 I didn't find enough time to play with it but I guess it is related with the codecs but I also wonder if it helps i change view sizes?

natario1, 2019-10-02:
I suggest you guys try with latest version 2.3.0.

closed 2019-10-02 by natario1