natario1/CameraView #595: Focus on tap continues to repeat multiple times

ivan200, 2019-09-18:
Device: Meizu 15
Os: Android 7.1.1
Camera2 Api support: Legacy.
CameraView version: 2.2.0

In example app, if i use camera2 engine, after tap, focus keep trying to detect like 5-10 seconds and it seems like jumping focusing.
In camera1 engine all working right, half second to focus on touched area.
Seems like situation mentioned here: https://stackoverflow.com/questions/41649691
But dont sure.

ivan200, 2019-09-30:
I attach an example reproduce it
https://imgur.com/UKxsPTU

natario1, 2019-10-02:
Thanks for reporting @ivan200 . I think this is a device specific issue, or maybe specific to legacy devices. I don't own this particular device so I don't know if a fix exists, and I don't know which one. 

This is not actionable for me. If you find any solution feel free to re-open this issue. Sorry

closed 2019-10-02 by natario1

HiDream, 2020-06-05:
@ivan200 Have you solved it？I met this problem too..

ivan200, 2020-06-05:
@HiDream 
solved it like this:
```
if (allowCamera2Support()) {
    cameraView.engine = Engine.CAMERA2
}
```
and
```
fun allowCamera2Support(): Boolean {
    if (Build.VERSION.SDK_INT < Build.VERSION_CODES.LOLLIPOP) return false

    (activity!!.getSystemService(CAMERA_SERVICE) as? CameraManager)
            ?.apply {
                cameraIdList.firstOrNull()?.let {
                    val characteristics = getCameraCharacteristics(it)
                    val support = characteristics.get(INFO_SUPPORTED_HARDWARE_LEVEL)
                    if(support == CameraMetadata.INFO_SUPPORTED_HARDWARE_LEVEL_FULL ||
                            support == CameraMetadata.INFO_SUPPORTED_HARDWARE_LEVEL_3){
                        return true
                    }
                }
            }
    return false
}
```