natario1/CameraView #596: Front camera producing mirror image

AkshayDandale, 2019-09-23:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *version number*
- Camera engine used: *camera1/camera2/both*
- Reproducible in official demo app: *yes/no*
- Device / Android version: *Pixel, API 28*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Go to '...'
2. Click on '....'
3. See error

### Expected behavior
A clear and concise description of what you expected to happen.

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
<CameraView
    android:layout_width="match_parent"
    android:layout_height="match_parent"       
    ...>
</CameraView>
```

### Screenshots
If applicable, add screenshots to help explain your problem.

### Logs
Use `CameraLogger.setLogLevel(LEVEL_VERBOSE)` to see all logs into LogCat.

Use `CameraLogger.registerLogger()` to export to file or crash reporting service.

### APK
Link to a Github repo where the bug is reproducible.

denizsubasi, 2019-09-26:
i use this code. 
 
 if (cameraView..facing == Facing.FRONT) {
                     bitmap.flip(-1f,1f,bitmap.width/2f,bitmap.height /2f)
                      }

 private fun Bitmap.flip(x: Float, y: Float, cx: Float, cy: Float): Bitmap {
        val matrix = Matrix().apply { postScale(x, y, cx, cy) }
        return Bitmap.createBitmap(this, 0, 0, width, height, matrix, true)
    }

natario1, 2019-09-28:
Please search into closed issues.

closed 2019-09-28 by natario1