natario1/CameraView #602: Camera1 api video recording not working with front camera

maheshchunkhade, 2019-09-30:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *version number*
- Camera engine used: *camera1/camera2/both*
- Reproducible in official demo app: *yes/no*
- Device / Android version: *Pixel, API 28*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Go to '...'
2. Click on '....'
3. See error

### Expected behavior
A clear and concise description of what you expected to happen.

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
<CameraView
    android:layout_width="match_parent"
    android:layout_height="match_parent"       
    ...>
</CameraView>
```

### Screenshots
If applicable, add screenshots to help explain your problem.

### Logs
start: Error while starting media recorder. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.

### APK
Expection #5 in sample app. 
Moto G 6

natario1, 2019-10-02:
@maheshchunkhade can you try with latest version `2.3.0` and also post complete logs from the demo app?

maheshchunkhade, 2019-10-03:
I am having same issue with latest version 2.3.0. Below are logs from demo app. 

I am using Camera1 with front camera on moto(g6) India version.

`2019-10-03 10:44:25.773 20940-20940/com.otaliastudios.cameraview.demo W/DemoApp: Recording for 5 seconds...
2019-10-03 10:44:25.792 20940-20940/com.otaliastudios.cameraview.demo V/CameraEngine: takeVideo scheduling
2019-10-03 10:44:25.793 20940-20974/com.otaliastudios.cameraview.demo V/CameraEngine: takeVideo performing. BindState: 2 isTakingVideo: false
2019-10-03 10:44:25.812 20940-20974/com.otaliastudios.cameraview.demo I/DeviceEncoders: findDeviceEncoder - type: video/avc encoders: 2
2019-10-03 10:44:25.812 20940-20974/com.otaliastudios.cameraview.demo I/DeviceEncoders: Enabled. Found video encoder: OMX.qcom.video.encoder.avc
2019-10-03 10:44:25.813 20940-20974/com.otaliastudios.cameraview.demo I/DeviceEncoders: findDeviceEncoder - type: audio/mp4a-latm encoders: 1
2019-10-03 10:44:25.813 20940-20974/com.otaliastudios.cameraview.demo I/DeviceEncoders: Enabled. Found audio encoder: OMX.google.aac.encoder
2019-10-03 10:44:25.814 20940-20974/com.otaliastudios.cameraview.demo I/DeviceEncoders: getSupportedVideoSize - inputSize: 2304x1728 adjustedSize: 1450x1088
2019-10-03 10:44:25.814 20940-20974/com.otaliastudios.cameraview.demo I/DeviceEncoders: getSupportedVideoBitRate - inputRate: 17000000 adjustedRate: 17000000
2019-10-03 10:44:25.814 20940-20974/com.otaliastudios.cameraview.demo I/DeviceEncoders: getSupportedAudioBitRate - inputRate: 128000 adjustedRate: 128000
2019-10-03 10:44:25.814 20940-20974/com.otaliastudios.cameraview.demo I/DeviceEncoders: getSupportedVideoFrameRate - inputRate: 30 adjustedRate: 30
2019-10-03 10:44:25.828 20940-20974/com.otaliastudios.cameraview.demo E/MediaRecorder: start failed: -19
2019-10-03 10:44:25.829 20940-20974/com.otaliastudios.cameraview.demo W/FullVideoRecorder: start: Error while starting media recorder. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:206)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:76)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:356)
        at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1212)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2019-10-03 10:44:25.829 20940-20974/com.otaliastudios.cameraview.demo I/CameraCallbacks: dispatchOnVideoRecordingEnd
2019-10-03 10:44:25.830 20940-20974/com.otaliastudios.cameraview.demo E/MediaRecorder: stop called in an invalid state: 0
2019-10-03 10:44:25.831 20940-20974/com.otaliastudios.cameraview.demo W/FullVideoRecorder: stop: Error while closing media recorder. java.lang.IllegalStateException
    java.lang.IllegalStateException
        at android.media.MediaRecorder.stop(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStop(FullVideoRecorder.java:221)
        at com.otaliastudios.cameraview.video.VideoRecorder.stop(VideoRecorder.java:90)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:212)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:76)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:356)
        at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1212)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2019-10-03 10:44:25.835 20940-20974/com.otaliastudios.cameraview.demo E/CameraEngine: onVideoResult result is null: something went wrong. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:206)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:76)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:356)
        at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1212)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2019-10-03 10:44:25.836 20940-20974/com.otaliastudios.cameraview.demo I/CameraCallbacks: dispatchError com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
        at com.otaliastudios.cameraview.engine.CameraEngine.onVideoResult(CameraEngine.java:1277)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onVideoResult(Camera1Engine.java:398)
        at com.otaliastudios.cameraview.video.VideoRecorder.dispatchResult(VideoRecorder.java:115)
[... cut ...]
2019-10-03 10:45:14.781 20940-20940/com.otaliastudios.cameraview.demo I/CameraCallbacks: onDeviceOrientationChanged 270
2019-10-03 10:45:14.781 20940-20940/com.otaliastudios.cameraview.demo I/Angles: Angles changed: sensorOffset: 90 displayOffset: 0 deviceOrientation: 270
2019-10-03 10:45:15.766 20940-20940/com.otaliastudios.cameraview.demo I/CameraCallbacks: onDeviceOrientationChanged 0
2019-10-03 10:45:15.766 20940-20940/com.otaliastudios.cameraview.demo I/Angles: Angles changed: sensorOffset: 90 displayOffset: 0 deviceOrientation: 0
`

natario1, 2019-10-03:
@maheshchunkhade what happens if you remove these lines? 

https://github.com/natario1/CameraView/blob/master/cameraview/src/main/java/com/otaliastudios/cameraview/video/FullVideoRecorder.java#L104-L133

maheshchunkhade, 2019-10-04:
@natario1 still same issue , logs as below

2019-10-04 11:40:30.894 13269-13323/com.otaliastudios.cameraview.demo V/Frame: Frame with time 1570169430893 is being released.
2019-10-04 11:40:30.903 13269-13269/com.otaliastudios.cameraview.demo W/DemoApp: Recording for 5 seconds...
2019-10-04 11:40:30.918 13269-13269/com.otaliastudios.cameraview.demo V/CameraEngine: takeVideo scheduling
2019-10-04 11:40:30.919 13269-13324/com.otaliastudios.cameraview.demo V/CameraEngine: takeVideo performing. BindState: 2 isTakingVideo: false
2019-10-04 11:40:30.960 13269-13324/com.otaliastudios.cameraview.demo E/MediaRecorder: start failed: -19
2019-10-04 11:40:30.961 13269-13324/com.otaliastudios.cameraview.demo W/FullVideoRecorder: start: Error while starting media recorder. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:209)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:76)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:356)
        at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1212)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2019-10-04 11:40:30.961 13269-13324/com.otaliastudios.cameraview.demo I/CameraCallbacks: dispatchOnVideoRecordingEnd
2019-10-04 11:40:30.962 13269-13324/com.otaliastudios.cameraview.demo E/MediaRecorder: stop called in an invalid state: 0
2019-10-04 11:40:30.966 13269-13324/com.otaliastudios.cameraview.demo W/FullVideoRecorder: stop: Error while closing media recorder. java.lang.IllegalStateException
    java.lang.IllegalStateException
        at android.media.MediaRecorder.stop(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStop(FullVideoRecorder.java:224)
        at com.otaliastudios.cameraview.video.VideoRecorder.stop(VideoRecorder.java:90)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:215)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:76)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:356)
        at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1212)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2019-10-04 11:40:30.969 13269-13324/com.otaliastudios.cameraview.demo E/CameraEngine: onVideoResult result is null: something went wrong. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:209)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:76)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:356)
        at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1212)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2019-10-04 11:40:30.972 13269-13324/com.otaliastudios.cameraview.demo I/CameraCallbacks: dispatchError com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
        at com.otaliastudios.cameraview.engine.CameraEngine.onVideoResult(CameraEngine.java:1277)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onVideoResult(Camera1Engine.java:398)
        at com.otaliastudios.cameraview.video.VideoRecorder.dispatchResult(VideoRecorder.java:115)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStop(FullVideoRecorder.java:239)
        at com.otaliastudios.cameraview.video.VideoRecorder.stop(VideoRecorder.java:90)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:215)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:76)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:356)
        at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1212)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
     Caused by: java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:209)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:76) 
[... cut ...]
2019-10-04 11:40:30.974 13269-13323/com.otaliastudios.cameraview.demo V/CameraCallbacks: dispatchFrame: dispatching 1570169430973 to processors.
2019-10-04 11:40:30.974 13269-13323/com.otaliastudios.cameraview.demo V/Frame: Frame with time 1570169430973 is being released.
2019-10-04 11:40:30.993 13269-13269/com.otaliastudios.cameraview.demo I/DemoApp: Video taken. Processing...
2019-10-04 11:40:31.002 13269-13269/com.otaliastudios.cameraview.demo W/DemoApp: onVideoRecordingEnd!
2019-10-04 11:40:31.002 13269-13269/com.otaliastudios.cameraview.demo W/DemoApp: Got CameraException #5

natario1, 2019-10-04:
What happens if you remove audio? You can do this through the edit panel in the demo app, set it to Off.

Thanks for your help

maheshchunkhade, 2019-10-04:
@natario1  its not working with audio off setting.

But when i change video encoder setting to H263 its working properly. Seems bit strange why its not take device_default video encoder.
is there any relation between video encoder and video format ?

natario1, 2019-10-04:
So this is like #467 . When you use `DEVICE_DEFAULT` with no changes, what is the value of the `videoType` variable? Also what is the value of `stub.size` ? You could set a breakpoint in FullVideoRecorder line 96 to check this.

There is a 1-1 relation between encoders and format, each codec encodes a different format.

natario1, 2019-10-04:
Just as reference for the future, these are useful links for debugging:

- [mediarecorder.cpp](https://android.googlesource.com/platform/frameworks/av/+/master/media/libmedia/mediarecorder.cpp) (MediaRecorder interface) 
- [StagefrightRecorder.cpp](https://android.googlesource.com/platform/frameworks/av/+/master/media/libmediaplayerservice/StagefrightRecorder.cpp) (MediaRecorder implementation)
- [MPEG4Writer.cpp](https://android.googlesource.com/platform/frameworks/av/+/master/media/libstagefright/MPEG4Writer.cpp) (video file writer)
- [CameraSource.cpp](https://android.googlesource.com/platform/frameworks/av/+/master/media/libstagefright/CameraSource.cpp) (used with Camera1)

maheshchunkhade, 2019-10-04:
videoType = video/avc
stub.size = 1088x1450
mProfile.fileFormat =  2

natario1, 2019-10-04:
@maheshchunkhade could you also tell me if this works with Camera2?

maheshchunkhade, 2019-10-04:
With camera 2 its working, below are values with debugging 
videoType  = video/avc
mProfile.fileFormat = 2
stub.size = 1080x1920
audio=off

natario1, 2019-10-04:
Going back to Camera1, another thing that I'd like you to try is to comment out these lines which I mentioned before... https://github.com/natario1/CameraView/blob/master/cameraview/src/main/java/com/otaliastudios/cameraview/video/FullVideoRecorder.java#L105-L133 

**and** also adding `flip = !flip;` just before `mMediaRecorder.setVideoSize()`.

maheshchunkhade, 2019-10-04:
2019-10-04 18:17:53.074 32254-32399/? D/ClClient: Not sending keepalive.  Current connection state=STOPPED
2019-10-04 18:17:55.808 2044-2044/com.otaliastudios.cameraview.demo W/DemoApp: Recording for 5 seconds...
2019-10-04 18:17:55.824 2044-2044/com.otaliastudios.cameraview.demo V/CameraEngine: takeVideo scheduling
2019-10-04 18:17:55.824 2044-2167/com.otaliastudios.cameraview.demo V/CameraEngine: takeVideo performing. BindState: 2 isTakingVideo: false
2019-10-04 18:17:55.827 1284-1922/? W/StagefrightRecorder: stop while neither recording nor paused
2019-10-04 18:17:55.840 1284-1922/? W/StagefrightRecorder: Max file size is not positive: 0 bytes. Disabling file size limit.
2019-10-04 18:17:55.840 1284-1922/? W/StagefrightRecorder: Target file size (0 bytes) is too small to be respected
2019-10-04 18:17:55.841 1284-1922/? W/StagefrightRecorder: Target duration (5000000 us) too short to be respected
2019-10-04 18:17:55.844 1284-1922/? W/StagefrightRecorder: Intended video encoding frame height (2304) is too large and will be set to (2160)
2019-10-04 18:17:55.846 1284-1922/? E/CameraSource: Video dimension (1728x2160) is unsupported
2019-10-04 18:17:55.846 1284-1922/? I/ExtendedCameraSource: Careating : ExtendedCameraSource
2019-10-04 18:17:55.846 1275-2452/? E/Camera2ClientBase: getParameters: attempt to use a locked camera from a different process (old pid 0, new pid 1284)
2019-10-04 18:17:55.847 1275-2452/? E/Camera2ClientBase: setParameters: attempt to use a locked camera from a different process (old pid 0, new pid 1284)
2019-10-04 18:17:55.848 1284-1922/? E/ExtendedUtils: Failed to set parameter - cache-video-buffers
2019-10-04 18:17:55.848 2044-2167/com.otaliastudios.cameraview.demo E/MediaRecorder: start failed: -19
2019-10-04 18:17:55.850 2044-2167/com.otaliastudios.cameraview.demo W/FullVideoRecorder: start: Error while starting media recorder. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:209)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:76)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:356)
        at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1212)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2019-10-04 18:17:55.850 2044-2167/com.otaliastudios.cameraview.demo I/CameraCallbacks: dispatchOnVideoRecordingEnd
2019-10-04 18:17:55.850 2044-2167/com.otaliastudios.cameraview.demo E/MediaRecorder: stop called in an invalid state: 0
2019-10-04 18:17:55.851 2044-2167/com.otaliastudios.cameraview.demo W/FullVideoRecorder: stop: Error while closing media recorder. java.lang.IllegalStateException
    java.lang.IllegalStateException
        at android.media.MediaRecorder.stop(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStop(FullVideoRecorder.java:224)
        at com.otaliastudios.cameraview.video.VideoRecorder.stop(VideoRecorder.java:90)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:215)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:76)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:356)
        at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1212)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2019-10-04 18:17:55.851 1284-1922/? W/StagefrightRecorder: stop while neither recording nor paused
2019-10-04 18:17:55.852 1275-2452/? E/Camera2ClientBase: setPreviewCallbackFlag: attempt to use a locked camera from a different process (old pid 0, new pid 2044)
2019-10-04 18:17:55.853 2044-2167/com.otaliastudios.cameraview.demo E/CameraEngine: onVideoResult result is null: something went wrong. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:209)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:76)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:356)
        at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1212)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2019-10-04 18:17:55.854 2044-2167/com.otaliastudios.cameraview.demo I/CameraCallbacks: dispatchError com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
        at com.otaliastudios.cameraview.engine.CameraEngine.onVideoResult(CameraEngine.java:1277)
[... cut ...]
2019-10-04 18:18:14.507 2057-2101/? W/BroadcastQueue: Background execution not allowed: receiving Intent { act=android.intent.action.DROPBOX_ENTRY_ADDED flg=0x10 (has extras) } to com.google.android.gms/.chimera.GmsIntentOperationService$PersistentTrustedReceiver
2019-10-04 18:18:14.525 3575-3344/? E/AQD:PackageSync: Error
2019-10-04 18:18:19.356 3207-3226/? V/FA: Inactivity, disconnecting from the service
2019-10-04 18:18:23.074 32254-32399/? D/ClClient: Not sending keepalive.  Current connection state=STOPPED

natario1, 2019-10-04:
@maheshchunkhade thanks for following up so fast.

Could you
- post the values of `cameraOptions.getSupportedVideoSizes()` when using Camera1?
- post the values of `cameraOptions.getSupportedVideoSizes()` when using Camera2?

These should be taken after you move to video mode and after you choose the correct camera (front?).

maheshchunkhade, 2019-10-04:
**Camera-2**

0 = {Size@11667} "960x1920"
1 = {Size@11668} "960x1280"
2 = {Size@11669} "640x1280"
3 = {Size@11670} "768x1024"
4 = {Size@11671} "480x640"
5 = {Size@11672} "288x352"
6 = {Size@11673} "144x176"
7 = {Size@11674} "360x640"
8 = {Size@11675} "720x1280"
9 = {Size@11676} "480x720"
10 = {Size@11677} "240x320"
11 = {Size@11678} "180x320"
12 = {Size@11679} "1080x1920"
13 = {Size@11680} "1080x1440"
14 = {Size@11681} "600x800"

**Camera 1 -**

0 = {Size@11712} "1920x3840"
1 = {Size@11713} "1728x2304"
2 = {Size@11714} "1536x2048"
3 = {Size@11715} "960x1920"
4 = {Size@11716} "960x1280"
5 = {Size@11717} "640x1280"
6 = {Size@11718} "768x1024"
7 = {Size@11719} "288x352"
8 = {Size@11720} "720x1280"
9 = {Size@11721} "1458x2592"
10 = {Size@11722} "1080x1440"
11 = {Size@11723} "1632x3264"
12 = {Size@11724} "480x640"
13 = {Size@11725} "144x176"
14 = {Size@11726} "360x640"
15 = {Size@11727} "1836x3264"
16 = {Size@11728} "480x720"
17 = {Size@11729} "240x320"
18 = {Size@11730} "1944x2592"
19 = {Size@11731} "1080x1920"
20 = {Size@11732} "600x800"

natario1, 2019-10-04:
@maheshchunkhade would you like to test the `full-video-improvements` branch? I think it might solve your issue.

maheshchunkhade, 2019-10-05:
@natario1  heay thanks, its working now with camera1 with front camera but substantial amount quality lost in video of both camera1 & camera2. in main branch quality is perfect. What is see in loss due frame rate. 
What fixes you made to fix this crash? 
Do you need videos check quality or understand problem?

maheshchunkhade, 2019-10-05:
https://we.tl/t-SOMJrQveXJ , here is sample file

natario1, 2019-10-05:
@maheshchunkhade can you pull that branch and try again? I just reordered the operations correctly.

maheshchunkhade, 2019-10-07:
@natario1  its working now thanks for your effort.  You can merge this branch to master whenever you ready.

PR #617 (2019-10-07): takeVideo improvements
Fixes #602 
Fixes #467
closed 2019-10-07 by natario1