natario1/CameraView #606: Front Camera - Error Close

lucazin, 2019-10-01:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.20*
- Camera engine used: *camera1/*
- Reproducible in official demo app: *no*
- Device / Android version: *A70, API 28*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Open Camera
2. Change to frontal camera
3. Start recording a video
4. And sometimes stop close the activity and sometimes its normal..

### Expected behavior
Recording normal

### XML layout
  
    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/RecordVideo"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:layout_gravity="center"
        android:keepScreenOn="true"
        android:visibility="visible"
        app:cameraAudio="on"
        app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
        app:cameraEngine="camera1"
        app:cameraExperimental="true"
        app:cameraFacing="back"
        app:cameraFlash="auto"
        app:cameraGestureLongTap="autoFocus"
        app:cameraGesturePinch="zoom"
        app:cameraGestureScrollHorizontal="filterControl1"
        app:cameraGestureScrollVertical="exposureCorrection"
        app:cameraGestureTap="autoFocus"
        app:cameraGrid="off"
        app:cameraMode="video"
        app:cameraPlaySounds="true"
        app:cameraPreview="glSurface"
        app:layout_constraintBottom_toBottomOf="parent"
        app:layout_constraintEnd_toEndOf="parent"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintTop_toTopOf="parent">

```xml
<CameraView
    android:layout_width="match_parent"
    android:layout_height="match_parent"       
    ...>
</CameraView>
```

### Screenshots
not have

### Logs
`2019-09-30 21:02:03.740 749-20558/? E/AudioFlinger: AudioFlinger_APM : (AudioFlinger::openRecord) persist.app.permission.monitor is true
2019-09-30 21:02:03.848 939-10359/? E/OMX-VENC: ERROR: unsupported index 1870659589
2019-09-30 21:02:03.953 696-21445/? E/ACDB-LOADER: Error: ACDB_CMD_GET_AUDPROC_INSTANCE_GAIN_DEP_STEP_TABLE_SIZE Returned = -19
2019-09-30 21:02:03.953 696-21445/? E/ACDB-LOADER: Error: ACDB AudProc vol returned = -19
2019-09-30 21:02:03.953 696-21445/? E/ACDB-LOADER: Error: ACDB_CMD_GET_AFE_INSTANCE_COMMON_TABLE_SIZE Returned = -19
2019-09-30 21:02:03.953 696-21445/? E/ACDB-LOADER: Error: ACDB AFE returned = -19
2019-09-30 21:02:03.968 19535-21430/br.com.oneplace A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x6300750064006f in tid 21430 (VideoEncoder), pid 19535 (br.com.oneplace)
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG: Build fingerprint: 'samsung/a70qxtc/a70q:9/PPR1.180610.011/A705MNXXU3ASG7:user/release-keys'
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG: Revision: '12'
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG: ABI: 'arm64'
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG: pid: 19535, tid: 21430, name: VideoEncoder  >>> br.com.oneplace <<<
[... 192 lines cut ...]
2019-09-30 21:02:06.725 21481-21481/br.com.oneplace E/LoadedApk: Unable to instantiate appComponentFactory
    java.lang.ClassNotFoundException: Didn't find class "android.support.v4.app.CoreComponentFactory" on path: DexPathList[[zip file "/system/framework/org.apache.http.legacy.boot.jar", zip file "/data/app/br.com.oneplace-hiRlls7vTg_OSFAphlvOrA==/base.apk"],nativeLibraryDirectories=[/data/app/br.com.oneplace-hiRlls7vTg_OSFAphlvOrA==/lib/arm64, /data/app/br.com.oneplace-hiRlls7vTg_OSFAphlvOrA==/base.apk!/lib/arm64-v8a, /system/lib64]]
        at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:134)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        at android.app.LoadedApk.createAppFactory(LoadedApk.java:241)
        at android.app.LoadedApk.createOrUpdateClassLoaderLocked(LoadedApk.java:770)
        at android.app.LoadedApk.getClassLoader(LoadedApk.java:849)
        at android.app.LoadedApk.getResources(LoadedApk.java:1090)
        at android.app.ContextImpl.createAppContext(ContextImpl.java:2563)
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6144)
        at android.app.ActivityThread.access$1200(ActivityThread.java:240)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1797)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7094)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:975)
2019-09-30 21:02:06.951 21494-21494/? E/[[WEATHER](161085@5)]: {[289822919A985EADBE56F8FDFF616C43225A3B2279C9A39418D1BC000569344A1DC1508B7A3B3C1BACE5BB711A677EB7E8D339FAD5086C6397E81A4F064CE4E9]}
2019-09-30 21:02:06.954 21494-21494/? E/[[WEATHER](161085@5)]: {[43EB1C024052B99D53E991930172379630519257355A8D28DD16151FD22314B0705629E7512927CE37D04AEECA3D7536061CCEAADE99499591CE63087FB099F7E59E883323AB4EA0530BD1E3C63000A785DC95222A4B209560BE487EB3F2E4C8CD6194FD67686619D64F1674B0BC25C1052F811A8D7602DA61C4ECD245DEBF24704667FBA64D7703B2354A68D5596929F0D18B4328D2DBC3DEF85BE6FBBB08B9B0E2FBAF31AA352905370D3805293658CE97A5DB66F7A07EB4C7AF467367D7F581054C3E353ED961EB3AF5C14BA6028A]}
2019-09-30 21:02:07.234 620-667/? E/Netd: getNetworkForDns: getNetId from enterpriseCtrl is netid 0`

### APK
Link to a Github repo where the bug is reproducible.

lucazin, 2019-10-01:
> 2019-09-30 22:37:01.993 18022-19613/br.com.oneplace A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 19613 (VideoEncoder), pid 18022 (br.com.oneplace)
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG: Build fingerprint: 'samsung/a70qxtc/a70q:9/PPR1.180610.011/A705MNXXU3ASG7:user/release-keys'
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG: Revision: '12'
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG: ABI: 'arm64'
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG: pid: 18022, tid: 19613, name: VideoEncoder  >>> br.com.oneplace <<<
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG: Cause: null pointer dereference
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG:     x0  0000000000000001  x1  000000797fae3b28  x2  0000000000250000  x3  0000000000000001
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG:     x4  000000797fabf688  x5  0000000000000005  x6  0000000000000227  x7  7e807d807d807d80
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG:     x8  0000000000000000  x9  0000000000000000  x10 0000000000002ba6  x11 000000797fae3b28
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG:     x12 0000000000000000  x13 0000000000000020  x14 0000007971ccef80  x15 000000797fcca498
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG:     x16 0000000000000000  x17 0000007980d55e30  x18 000000796f03e798  x19 0000007975c301c0
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG:     x20 0000000000000000  x21 0000000000000000  x22 0000000000000006  x23 000000798a027b18
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG:     x24 0000000000000000  x25 0000000000000000  x26 0000000000000000  x27 000000796f03ed18
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG:     x28 0000000000000000  x29 000000796f03ed00
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG:     sp  000000796f03e9f0  lr  000000797fac8ff4  pc  000000797fac92a0
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG: backtrace:
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #00 pc 00000000002622a0  /vendor/lib64/egl/libGLESv2_adreno.so (EsxResource::UpdateSubResourceDirect(EsxContext*, unsigned int, int, EsxBox const*, EsxSurfaceAccessDesc const*, int)+1800)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #01 pc 0000000000262d48  /vendor/lib64/egl/libGLESv2_adreno.so (EsxTextureObject::PerformShadowCopy(EsxContext*)+608)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #02 pc 000000000024fa4c  /vendor/lib64/egl/libGLESv2_adreno.so (EsxContext::ValidateProgramTexUnitBoundObjs(EsxProgram*, int)+724)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #03 pc 000000000024c648  /vendor/lib64/egl/libGLESv2_adreno.so (EsxContext::ValidateGfxState(EsxDrawDescriptor const*)+520)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #04 pc 00000000002a4350  /vendor/lib64/egl/libGLESv2_adreno.so (EsxContext::DrawArraysInstanced(EsxPrimType, int, unsigned int, unsigned int)+200)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #05 pc 0000000000441a28  /system/framework/arm64/boot-framework.oat (offset 0x41e000) (android.opengl.GLES10.glDrawArrays [DEDUPED]+168)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #06 pc 00000000000a6924  /dev/ashmem/dalvik-jit-code-cache_18022_18022 (deleted) (com.otaliastudios.cameraview.filter.BaseFilter.onDraw+52)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #07 pc 00000000000aa0d8  /dev/ashmem/dalvik-jit-code-cache_18022_18022 (deleted) (com.otaliastudios.cameraview.filter.BaseFilter.draw+200)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #08 pc 0000000000096e6c  /dev/ashmem/dalvik-jit-code-cache_18022_18022 (deleted) (com.otaliastudios.cameraview.internal.egl.EglViewport.drawFrame+284)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #09 pc 000000000009fc28  /dev/ashmem/dalvik-jit-code-cache_18022_18022 (deleted) (com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onFrame+2264)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #10 pc 0000000000559388  /system/lib64/libart.so (art_quick_invoke_stub+584)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #11 pc 00000000000d02c8  /system/lib64/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+200)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #12 pc 0000000000280cc0  /system/lib64/libart.so (art::interpreter::ArtInterpreterToCompiledCodeBridge(art::Thread*, art::ArtMethod*, art::ShadowFrame*, unsigned short, art::JValue*)+344)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #13 pc 000000000027acc8  /system/lib64/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+968)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #14 pc 000000000052971c  /system/lib64/libart.so (MterpInvokeDirect+296)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #15 pc 000000000054bb14  /system/lib64/libart.so (ExecuteMterpImpl+14484)
2019-09-30 22:37:02.249 19630-19630/? A/DEBUG:     #16 pc 000000000002fbc6  /dev/ashmem/dalvik-classes110.dex extracted in memory from /data/app/br.com.oneplace-W53Wtq21mXbWUxU2ynHBPQ==/base.apk!classes110.dex_18022_18022 (deleted) (com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onEvent+90)
[... cut ...]
2019-09-30 22:37:02.250 19630-19630/? A/DEBUG:     #37 pc 000000000045fe0c  /system/lib64/libart.so (art::InvokeVirtualOrInterfaceWithJValues(art::ScopedObjectAccessAlreadyRunnable const&, _jobject*, _jmethodID*, jvalue*)+424)
2019-09-30 22:37:02.250 19630-19630/? A/DEBUG:     #38 pc 000000000048b22c  /system/lib64/libart.so (art::Thread::CreateCallback(void*)+1120)
2019-09-30 22:37:02.250 19630-19630/? A/DEBUG:     #39 pc 0000000000091c3c  /system/lib64/libc.so (__pthread_start(void*)+36)
2019-09-30 22:37:02.250 19630-19630/? A/DEBUG:     #40 pc 0000000000024200  /system/lib64/libc.so (__start_thread+68)

lucazin, 2019-10-01:
When i change to front camera and start recording ...this is cenario

**Build fingerprint: 'samsung/a70qxtc/a70q:9/PPR1.180610.011/A705MNXXU3ASG7:user/release-keys'
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG: Revision: '12'
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG: ABI: 'arm64'
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG: pid: 18022, tid: 19613, name: VideoEncoder  >>> br.com.oneplace <<<
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
2019-09-30 22:37:02.148 19630-19630/? A/DEBUG: Cause: null pointer**

lucazin, 2019-10-01:
i will try back to 1.6.0 version to check if i have same problem...  =/

natario1, 2019-10-02:
@lucazin can you also post logs **before** the crash?

lucazin, 2019-10-02:
Sending now...   U want before the log I send here?

natario1, 2019-10-02:
Yes! You could set the level to `Info`.

lucazin, 2019-10-02:
this is the log when i open the camera ( frontal camera default) 

`2019-10-02 06:33:22.444 2098-2098/? I/StackScroller: ADJUST STACK.TRANSLATION
2019-10-02 06:33:24.477 1250-1719/? I/InputReader: Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.2987 ] when=8743.981827
2019-10-02 06:33:24.479 1250-1718/? I/InputDispatcher: Delivering touch to (2098): action: 0x4, displayId: 0, '9f46511', toolType: 1
2019-10-02 06:33:24.479 1250-1718/? I/clp-JNI: Add pinfo) 0
2019-10-02 06:33:24.479 1250-1718/? I/InputDispatcher: Delivering touch to (31595): action: 0x0, displayId: 0, 'c5a105f', toolType: 1
2019-10-02 06:33:24.528 1250-1719/? I/InputReader: Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=8744.032334
2019-10-02 06:33:24.528 1250-1718/? I/clp-JNI: Add pinfo) c5a105f br.com.oneplace/aplicativo.oneplace.Docker.Container.MainActivity (server) : AppWindowToken{8b23f11 token=Token{bf82338 ActivityRecord{129f39b u0 br.com.oneplace/aplicativo.oneplace.Docker.Container.MainActivity t53}}} : 31595 : 10224 : 1
2019-10-02 06:33:24.528 1250-1718/? I/clp-JNI: Add pinfo) 1
2019-10-02 06:33:24.528 1250-1718/? I/InputDispatcher: Delivering touch to (31595): action: 0x1, displayId: 0, 'c5a105f', toolType: 1
2019-10-02 06:33:24.542 1250-2467/? I/ActivityManager: START u0 {act=null typ=null flg=0x0 cmp=ComponentInfo{br.com.oneplace/aplicativo.oneplace.Fragments.AppEstabelecimento.Default_Features.VideoComercial.VideoComercial}} from uid 10224
2019-10-02 06:33:24.544 693-723/? E/ANDR-PERF-MPCTL: Invalid profile no. 0, total profiles 0 only
2019-10-02 06:33:24.555 738-803/? I/SurfaceFlinger: id=6637 createSurf (4800x4800),2 flag=4, AppWindowToken{ee1b7c4 token=Token{3da12d7 ActivityRecord{2b7256 u0 br.com.oneplace/aplicativo.oneplace.Fragments.AppEstabelecimento.Default_Features.VideoComercial.VideoComercial t53}}}#0
2019-10-02 06:33:24.563 686-686/? I/sensors-hal: activate:129, activate handle 21 
2019-10-02 06:33:24.563 686-686/? I/sensors-hal: activate:134, android.sensor.accelerometer/21 en=0
2019-10-02 06:33:24.565 686-686/? I/sensors-hal: deactivate:240, disconnect qmi to disable sensor
2019-10-02 06:33:24.565 686-686/? I/sensors-hal: activate:145, android.sensor.accelerometer/21 en=0 completed
2019-10-02 06:33:24.568 2367-8806/? I/NfcService: setForegroundDispatch disable
2019-10-02 06:33:24.569 1250-14020/? I/ActivityManager: mTimer start at this point for Notification solution
2019-10-02 06:33:24.569 1250-14020/? I/ActivityManager: mTimer scheduling done
2019-10-02 06:33:24.572 1250-14020/? I/SemPersonaManager: SecureFolder personaId = 150
2019-10-02 06:33:24.577 31595-31595/br.com.oneplace W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@417aef6
2019-10-02 06:33:24.601 31595-31595/br.com.oneplace I/FFmpeg: Loading FFmpeg for armv7 CPU
2019-10-02 06:33:24.604 31595-31595/br.com.oneplace I/DecorView: mWindow.mActivityCurrentConfig is null.
2019-10-02 06:33:24.605 31595-31595/br.com.oneplace I/DecorView: createDecorCaptionView >> DecorView@c206f01[], isFloating: false, isApplication: true, hasWindowDecorCaption: false, hasWindowControllerCallback: true
2019-10-02 06:33:24.693 1250-14020/? I/SamsungAlarmManager: setLocked to kernel - T:2 / 20191002T063437, SetElapsed=35544114, nowELAPSED=35471017
2019-10-02 06:33:24.693 1250-14020/? I/SamsungAlarmManager: setLocked to kernel - T:3 / 20191002T063400, SetElapsed=35506324, nowELAPSED=35471017
2019-10-02 06:33:24.712 1250-14020/? I/SamsungAlarmManager: setLocked to kernel - T:2 / 20191002T063437, SetElapsed=35544114, nowELAPSED=35471036
2019-10-02 06:33:24.712 1250-14020/? I/SamsungAlarmManager: setLocked to kernel - T:3 / 20191002T063400, SetElapsed=35506324, nowELAPSED=35471036
2019-10-02 06:33:24.732 1250-12463/? I/SensorService: createSensorEventConnection package name com.instabug.library.util.c
2019-10-02 06:33:24.733 686-686/? I/sensors-hal: batch:202, android.sensor.accelerometer/21, period=200000000, max_latency=0
2019-10-02 06:33:24.733 686-686/? I/sensors-hal: batch:211, android.sensor.accelerometer/21, period=200000000, max_latency=0 request completed
2019-10-02 06:33:24.733 686-686/? I/sensors-hal: activate:129, activate handle 21 
2019-10-02 06:33:24.733 686-686/? I/sensors-hal: activate:134, android.sensor.accelerometer/21 en=1
2019-10-02 06:33:24.734 686-686/? I/sensors-hal: ssc_enable_rt_task:260, ssc_enable_rt_task : true
2019-10-02 06:33:24.738 686-686/? I/sensors-hal: activate:145, android.sensor.accelerometer/21 en=1 completed
2019-10-02 06:33:24.739 686-31881/? I/sensors-hal: handle_indication_realtime:353,  SCHED_FIFO(10) for qmi_cbk
2019-10-02 06:33:24.742 1250-12463/? I/SensorService: createSensorEventConnection package name android.view.OrientationEventListener
2019-10-02 06:33:24.743 686-686/? I/sensors-hal: flush:231, android.sensor.accelerometer/21
2019-10-02 06:33:24.744 686-686/? I/sensors-hal: flush:237, android.sensor.accelerometer/21 completed
2019-10-02 06:33:24.750 826-1864/? I/CameraService: CameraService::connect call (PID -1 "br.com.oneplace", camera ID 1) for HAL version default and Camera API version 1
2019-10-02 06:33:24.751 1250-12463/? I/EDMNativeHelperService: isCameraEnabled
2019-10-02 06:33:24.751 738-803/? I/SurfaceFlinger: id=6638 createSurf (4800x4800),2 flag=4, f3e85cf br.com.oneplace/aplicativo.oneplace.Fragments.AppEstabelecimento.Default_Features.VideoComercial.VideoComercial#0
2019-10-02 06:33:24.751 826-1864/? E/CameraService: checkTemperature (0)
[... cut ...]
2019-10-02 06:33:32.504 696-1860/? E/CamX: [ERROR][STATS_AFD] afd_process.cpp:1124: afd_process_detect_flicker Return! num_peaks:0
2019-10-02 06:33:32.937 1250-1314/? I/EDMNativeHelperService: isCameraEnabled
2019-10-02 06:33:33.383 1250-1250/? I/MotionRecognitionService: Plugged
2019-10-02 06:33:33.404 2098-2098/? I/StackScroller: ADJUST STACK.TRANSLATION
2019-10-02 06:33:33.404 2098-2098/? I/StackScroller: ADJUST STACK.TRANSLATION
2019-10-02 06:33:33.404 2098-2098/? I/StackScroller: ADJUST STACK.TRANSLATION
2019-10-02 06:33:33.943 1250-2365/? I/EDMNativeHelperService: isCameraEnabled
`

lucazin, 2019-10-02:
before crash in  info level 

`2019-10-02 06:34:34.634 869-32011/? I/MediaCodec: setCodecState state : 0
2019-10-02 06:34:34.634 869-32011/? I/ResourceManagerService: getMediaResourceInfo resourceType : 2, size : 1
2019-10-02 06:34:34.634 869-32011/? I/ResourceManagerService: writeResourceInfo (PID : 31595, clientID : 3979418112, non-secure-codec/video-codec:1, 708x1440(fps:25) - SoftCodec : No, Encorder : No)
2019-10-02 06:34:34.636 7261-7280/? I/SemMediaResourceHelper: makeMediaResourceInfo mOwnResourceEventExcluded : false, mPid : 7261
2019-10-02 06:34:34.636 7261-7280/? I/SemMediaResourceHelper: [1] makeMediaResourceInfo resourceType : 2 isSecured : false, pid : 31595, client id : 3979418112
2019-10-02 06:34:34.637 7261-7296/? I/SemMediaResourceHelper: onState
2019-10-02 06:34:34.637 7261-7296/? I/DrmMediaResourceHelper: onStateChanged size = 1
2019-10-02 06:34:34.637 7261-7296/? I/DrmMediaResourceHelper: resource type: 2,,,isSecured :  falsegetCodecState : 1getpid : 31595
2019-10-02 06:34:34.786 686-23101/? I/sensors-hal: handle_sns_client_event:78, auto_rotation_event: 255
2019-10-02 06:34:34.790 1250-1684/? I/CAE: parse(SensorHubParserProvider.java:192) - buffer size = 4
2019-10-02 06:34:34.790 1250-1684/? I/CAE: parse(SensorHubParserProvider.java:203) - 1, 1, 7, -1,
2019-10-02 06:34:34.792 1250-1684/? I/CAE: display(ContextProvider.java:391) - Angle=[-1]
2019-10-02 06:34:34.813 1250-2031/? E/TaskPersister: File error accessing recents directory (directory doesn't exist?).
2019-10-02 06:34:34.844 686-23101/? I/sensors-hal: handle_sns_client_event:83, auto_rotation_event: mode:255, type:1:1
2019-10-02 06:34:34.963 686-23101/? I/sensors-hal: handle_sns_client_event:78, auto_rotation_event: 0
2019-10-02 06:34:34.967 1250-1684/? I/CAE: parse(SensorHubParserProvider.java:192) - buffer size = 4
2019-10-02 06:34:34.967 1250-1684/? I/CAE: parse(SensorHubParserProvider.java:203) - 1, 1, 7, 0,
2019-10-02 06:34:34.969 1250-1684/? I/CAE: display(ContextProvider.java:391) - Angle=[0]
2019-10-02 06:34:35.268 1250-1719/? I/InputReader: Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.2990 ] when=8814.770011
2019-10-02 06:34:35.269 1250-1718/? I/InputDispatcher: Delivering touch to (2098): action: 0x4, displayId: 0, '9f46511', toolType: 1
2019-10-02 06:34:35.269 1250-1718/? I/clp-JNI: Add pinfo) 0
2019-10-02 06:34:35.269 1250-1718/? I/InputDispatcher: Delivering touch to (31595): action: 0x0, displayId: 0, '940ba5 ', toolType: 1
2019-10-02 06:34:35.336 1250-1719/? I/InputReader: Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=8814.840546
2019-10-02 06:34:35.336 1250-1718/? I/clp-JNI: Add pinfo) 940ba5 br.com.oneplace/aplicativo.oneplace.Fragments.AppEstabelecimento.Default_Features.VideoComercial.VideoComercialPreview (server) : AppWindowToken{a24e270 token=Token{bd1ccb3 ActivityRecord{480ba22 u0 br.com.oneplace/aplicativo.oneplace.Fragments.AppEstabelecimento.Default_Features.VideoComercial.VideoComercialPreview t53}}} : 31595 : 10224 : 1
2019-10-02 06:34:35.336 1250-1718/? I/clp-JNI: Add pinfo) 1
2019-10-02 06:34:35.336 1250-1718/? I/InputDispatcher: Delivering touch to (31595): action: 0x1, displayId: 0, '940ba5 ', toolType: 1
2019-10-02 06:34:35.347 686-1635/? I/sensors-hal: batch:202, android.sensor.accelerometer/21, period=200000000, max_latency=0
2019-10-02 06:34:35.347 686-1635/? I/sensors-hal: batch:211, android.sensor.accelerometer/21, period=200000000, max_latency=0 request completed
2019-10-02 06:34:35.356 1250-2561/? I/SemPersonaManager: SecureFolder personaId = 150
2019-10-02 06:34:35.369 31595-31595/br.com.oneplace I/FirebasePerformance: Session ID - f893ccb4ef664b15be44798d1a12ea7f
2019-10-02 06:34:35.372 1250-2561/? I/SensorService: createSensorEventConnection package name com.instabug.library.util.c
2019-10-02 06:34:35.373 686-1635/? I/sensors-hal: flush:231, android.sensor.accelerometer/21
2019-10-02 06:34:35.374 686-1635/? I/sensors-hal: flush:237, android.sensor.accelerometer/21 completed
2019-10-02 06:34:35.385 738-1528/? I/SurfaceFlinger: id=6671 createSurf (1080x2324),1 flag=4, br.com.oneplace/aplicativo.oneplace.Fragments.AppEstabelecimento.Default_Features.VideoComercial.VideoComercial[31595]#0
2019-10-02 06:34:35.386 826-1864/? I/CameraService: CameraService::connect call (PID -1 "br.com.oneplace", camera ID 1) for HAL version default and Camera API version 1
2019-10-02 06:34:35.388 1250-2561/? I/EDMNativeHelperService: isCameraEnabled
2019-10-02 06:34:35.390 826-1864/? E/CameraService: checkTemperature (0)
2019-10-02 06:34:35.390 826-1864/? I/CameraService: CameraService::validateClientPermissionsLocked is ok : calling pid 31595, calling uid 10224, cameraservice pid=826, device user 0, currently allowed device users: 0, 150
2019-10-02 06:34:35.390 826-1864/? I/CameraService: CameraService::handleEvictionsLocked
2019-10-02 06:34:35.390 826-1864/? I/CameraService: wouldEvictLocked:getCurrentCostLocked()=0, cost=100, totalCost=100, MaxCost=100
2019-10-02 06:34:35.390 826-1864/? I/CameraService: wouldEvictLocked: owner=31595, priority-score=0, highestPriorityOwner=31595, highestPriority-score=0
2019-10-02 06:34:35.390 826-1864/? I/CameraFlashlight: prepareDeviceOpen: prepare for device open(cameraId=1, package=br.com.oneplace)
2019-10-02 06:34:35.391 826-1864/? I/CameraService: onTorchStatusChangedLocked: Torch status changed for cameraId=0, newStatus=0
[... cut ...]
2019-10-02 06:34:36.444 673-32082/? E/ACDB-LOADER: Error: ACDB AudProc vol returned = -19
2019-10-02 06:34:36.444 673-32082/? E/ACDB-LOADER: Error: ACDB_CMD_GET_AFE_INSTANCE_COMMON_TABLE_SIZE Returned = -19
2019-10-02 06:34:36.444 673-32082/? E/ACDB-LOADER: Error: ACDB AFE returned = -19
2019-10-02 06:34:36.444 673-32082/? I/msm8974_platform: platform_get_custom_mtmx_params: no matching param with id 0 ip_ch 1 op_ch 1 uc_id 19 snd_dev 258
2019-10-02 06:34:36.451 31595-32069/br.com.oneplace W/br.com.oneplac: 0xebadde09 skipped times: 0`

lucazin, 2019-10-02:
@natario1  beginning of crash

`--------- beginning of crash
2019-10-02 06:34:36.452 31595-32069/br.com.oneplace A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x6300750064006f in tid 32069 (VideoEncoder), pid 31595 (br.com.oneplace)
2019-10-02 06:34:36.531 1250-2468/? I/EDMNativeHelperService: isCameraEnabled
2019-10-02 06:34:36.589 32086-32086/? I/crash_dump64: obtaining output fd from tombstoned, type: kDebuggerdTombstone
2019-10-02 06:34:36.589 965-965/? I//system/bin/tombstoned: received crash request for pid 32069
2019-10-02 06:34:36.591 32086-32086/? I/crash_dump64: performing dump of process 31595 (target tid = 32069)
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG: Build fingerprint: 'samsung/a70qxtc/a70q:9/PPR1.180610.011/A705MNXXU3ASG7:user/release-keys'
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG: Revision: '12'
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG: ABI: 'arm64'
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG: pid: 31595, tid: 32069, name: VideoEncoder  >>> br.com.oneplace <<<
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x6300750064006f
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG:     x0  0000000000000000  x1  0000000000000000  x2  0000000000000000  x3  00000007fed9f710
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG:     x4  00000077d17c5830  x5  0000000000000005  x6  0000000000000227  x7  8080808080808080
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG:     x8  ffffffffffe2d754  x9  0000000000000001  x10 0000000000000001  x11 8080808080808080
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG:     x12 8080808080808080  x13 8080808080808080  x14 0000000000000001  x15 00000007fed9f740
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG:     x16 0000000000000600  x17 0000007879d6a7e8  x18 00000077d13850e8  x19 00000077d1a466e0
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG:     x20 006300750064006f  x21 00000007fecd5000  x22 00000077d72f2500  x23 0000000000000000
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG:     x24 0000000000000001  x25 0000000000630069  x26 00000077ed827640  x27 0000000000000001
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG:     x28 0000000000000001  x29 00000077d1385330
2019-10-02 06:34:36.604 32086-32086/? A/DEBUG:     sp  00000077d1384ef0  lr  00000077e330fe90  pc  00000077e330f498
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG: backtrace:
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #00 pc 000000000026a498  /vendor/lib64/egl/libGLESv2_adreno.so (EsxBltLib::CopyHostPtrBltSw(EsxBltCopyHostPtr const*)+416)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #01 pc 0000000000261ff0  /vendor/lib64/egl/libGLESv2_adreno.so (EsxResource::UpdateSubResourceDirect(EsxContext*, unsigned int, int, EsxBox const*, EsxSurfaceAccessDesc const*, int)+1112)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #02 pc 0000000000262d48  /vendor/lib64/egl/libGLESv2_adreno.so (EsxTextureObject::PerformShadowCopy(EsxContext*)+608)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #03 pc 000000000024fa4c  /vendor/lib64/egl/libGLESv2_adreno.so (EsxContext::ValidateProgramTexUnitBoundObjs(EsxProgram*, int)+724)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #04 pc 000000000024c648  /vendor/lib64/egl/libGLESv2_adreno.so (EsxContext::ValidateGfxState(EsxDrawDescriptor const*)+520)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #05 pc 00000000002a4350  /vendor/lib64/egl/libGLESv2_adreno.so (EsxContext::DrawArraysInstanced(EsxPrimType, int, unsigned int, unsigned int)+200)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #06 pc 0000000000441a28  /system/framework/arm64/boot-framework.oat (offset 0x41e000) (android.opengl.GLES10.glDrawArrays [DEDUPED]+168)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #07 pc 0000000000072bc8  /dev/ashmem/dalvik-jit-code-cache_31595_31595 (deleted) (aplicativo.otaliastudios.cameraview.filter.BaseFilter.onDraw+56)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #08 pc 000000000007350c  /dev/ashmem/dalvik-jit-code-cache_31595_31595 (deleted) (aplicativo.otaliastudios.cameraview.filter.BaseFilter.draw+220)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #09 pc 00000000000701d8  /dev/ashmem/dalvik-jit-code-cache_31595_31595 (deleted) (aplicativo.otaliastudios.cameraview.internal.egl.EglViewport.drawFrame+296)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #10 pc 0000000000559388  /system/lib64/libart.so (art_quick_invoke_stub+584)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #11 pc 00000000000d02c8  /system/lib64/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+200)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #12 pc 0000000000280cc0  /system/lib64/libart.so (art::interpreter::ArtInterpreterToCompiledCodeBridge(art::Thread*, art::ArtMethod*, art::ShadowFrame*, unsigned short, art::JValue*)+344)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #13 pc 000000000027acc8  /system/lib64/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+968)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #14 pc 00000000005283dc  /system/lib64/libart.so (MterpInvokeVirtual+588)
2019-10-02 06:34:36.712 32086-32086/? A/DEBUG:     #15 pc 000000000054ba14  /system/lib64/libart.so (ExecuteMterpImpl+14228)
[... cut ...]
2019-10-02 06:34:38.153 867-31284/? I/OggExtractor: ~MyOggExtractor --
2019-10-02 06:34:38.153 867-31284/? I/OggExtractor: ~OggExtractor --
2019-10-02 06:34:38.158 867-31284/? I/OggExtractor: ~OggExtractor ++
2019-10-02 06:34:38.158 867-31284/? I/OggExtractor: ~MyOggExtractor ++ 
2019-10-02 06:34:38.159 867-31284/? I/OggExtractor: ~MyOggExtractor --
2019-10-02 06:34:38.159 867-31284/? I/OggExtractor: ~OggExtractor --
2019-10-02 06:34:38.160 826-4315/? E/CameraService: binderDied: Java client's binder died, removing it from the list of active clients`

lucazin, 2019-10-02:
@natario1  need some other logs?  I send correctly?

I want to help you to get this problem resolution, because is very important for me and my  work and for others here using this awesome lib!

lucazin, 2019-10-02:
@natario1 i think this log is more readable

`2019-10-02 06:46:26.671 2040-2153/br.com.oneplace W/FirebaseRemoteConfig: No value of type 'String' exists for parameter key 'sessions_max_length_minutes'.
2019-10-02 06:46:26.671 2040-2219/br.com.oneplace W/FirebaseRemoteConfig: No value of type 'String' exists for parameter key 'sessions_feature_enabled'.
    
    --------- beginning of system
2019-10-02 06:46:26.690 1250-2081/? I/ActivityManager: Killing 24229:com.google.process.gapps/u0a38 (adj 906): empty #33
2019-10-02 06:46:26.691 1250-1424/? W/libprocessgroup: kill(-24229, 9) failed: No such process
2019-10-02 06:46:26.714 1250-1424/? I/chatty: uid=1000(system) ActivityManager identical 4 lines
2019-10-02 06:46:26.720 1250-1424/? W/libprocessgroup: kill(-24229, 9) failed: No such process
2019-10-02 06:46:26.721 616-616/? I/Zygote: Process 24229 exited due to signal (9)
2019-10-02 06:46:26.725 1250-1424/? W/libprocessgroup: kill(-24229, 9) failed: No such process
2019-10-02 06:46:26.725 1250-1424/? I/libprocessgroup: Successfully killed process cgroup uid 10038 pid 24229 in 34ms
2019-10-02 06:46:26.727 1250-2328/? E/BartenderActivityManager: BarTender: pid : 1825 cached=true abnormal=false kill=false stay=false release=false
2019-10-02 06:46:26.892 1250-2328/? E/BartenderActivityManager: BarTender: pid : 1873 cached=true abnormal=false kill=false stay=false release=false
2019-10-02 06:46:27.144 1250-2081/? E/BartenderActivityManager: BarTender: pid : 1911 cached=true abnormal=false kill=false stay=false release=false
2019-10-02 06:46:27.308 1250-1719/? I/InputReader: Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.3001 ] when=9512.172758
2019-10-02 06:46:27.309 1250-1718/? I/InputDispatcher: Delivering touch to (2098): action: 0x4, displayId: 0, '9f46511', toolType: 1
2019-10-02 06:46:27.309 1250-1718/? I/clp-JNI: Add pinfo) 0
2019-10-02 06:46:27.309 1250-1718/? I/InputDispatcher: Delivering touch to (2040): action: 0x0, displayId: 0, '80d59be', toolType: 1
2019-10-02 06:46:27.386 1250-1719/? I/InputReader: Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=9512.252503
2019-10-02 06:46:27.386 1250-1718/? I/clp-JNI: Add pinfo) 80d59be br.com.oneplace/aplicativo.oneplace.Docker.Container.MainActivity (server) : AppWindowToken{275586a token=Token{d86d655 ActivityRecord{2d73d0c u0 br.com.oneplace/aplicativo.oneplace.Docker.Container.MainActivity t54}}} : 2040 : 10224 : 1
2019-10-02 06:46:27.386 1250-1718/? I/clp-JNI: Add pinfo) 1
2019-10-02 06:46:27.386 1250-1718/? I/InputDispatcher: Delivering touch to (2040): action: 0x1, displayId: 0, '80d59be', toolType: 1
2019-10-02 06:46:27.393 1250-2081/? E/BartenderActivityManager: BarTender: pid : 1900 cached=true abnormal=false kill=false stay=false release=false
2019-10-02 06:46:27.403 1250-11180/? I/ActivityManager: START u0 {act=null typ=null flg=0x0 cmp=ComponentInfo{br.com.oneplace/aplicativo.oneplace.Fragments.AppEstabelecimento.Default_Features.VideoComercial.VideoComercial}} from uid 10224
2019-10-02 06:46:27.415 738-1377/? I/SurfaceFlinger: id=6769 createSurf (4800x4800),2 flag=4, AppWindowToken{1948006 token=Token{5e9aae1 ActivityRecord{c1d9c48 u0 br.com.oneplace/aplicativo.oneplace.Fragments.AppEstabelecimento.Default_Features.VideoComercial.VideoComercial t54}}}#0
2019-10-02 06:46:27.422 686-686/? I/sensors-hal: activate:129, activate handle 21 
2019-10-02 06:46:27.422 686-686/? I/sensors-hal: activate:134, android.sensor.accelerometer/21 en=0
2019-10-02 06:46:27.424 686-686/? I/sensors-hal: deactivate:240, disconnect qmi to disable sensor
2019-10-02 06:46:27.424 686-686/? I/sensors-hal: activate:145, android.sensor.accelerometer/21 en=0 completed
2019-10-02 06:46:27.426 2367-8806/? I/NfcService: setForegroundDispatch disable
2019-10-02 06:46:27.427 1250-2081/? I/ActivityManager: mTimer start at this point for Notification solution
2019-10-02 06:46:27.428 1250-2081/? I/ActivityManager: mTimer scheduling done
2019-10-02 06:46:27.431 1250-2081/? I/SemPersonaManager: SecureFolder personaId = 150
2019-10-02 06:46:27.438 2040-2040/br.com.oneplace W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@431f302
2019-10-02 06:46:27.461 1250-2177/? E/Watchdog: !@Sync 316 [2019-10-02 06:46:27.461] FD count : 550
2019-10-02 06:46:27.463 2040-2040/br.com.oneplace I/FFmpeg: Loading FFmpeg for armv7 CPU
2019-10-02 06:46:27.466 2040-2040/br.com.oneplace I/DecorView: mWindow.mActivityCurrentConfig is null.
2019-10-02 06:46:27.467 2040-2040/br.com.oneplace I/DecorView: createDecorCaptionView >> DecorView@3d5ea14[], isFloating: false, isApplication: true, hasWindowDecorCaption: false, hasWindowControllerCallback: true
2019-10-02 06:46:27.550 1250-11180/? E/BartenderActivityManager: BarTender: pid : 1937 cached=true abnormal=false kill=false stay=false release=false
2019-10-02 06:46:27.580 1250-11180/? I/SensorService: createSensorEventConnection package name com.instabug.library.util.c
2019-10-02 06:46:27.581 686-686/? I/sensors-hal: batch:202, android.sensor.accelerometer/21, period=200000000, max_latency=0
2019-10-02 06:46:27.581 686-686/? I/sensors-hal: batch:211, android.sensor.accelerometer/21, period=200000000, max_latency=0 request completed
2019-10-02 06:46:27.581 686-686/? I/sensors-hal: activate:129, activate handle 21 
2019-10-02 06:46:27.581 686-686/? I/sensors-hal: activate:134, android.sensor.accelerometer/21 en=1
[... cut ...]
2019-10-02 06:46:35.601 1250-1630/? I/WindowManager: commitFinishDrawingLocked: mDrawState=READY_TO_SHOW Surface(name=br.com.oneplace/aplicativo.oneplace.Docker.Container.MainActivity[2598])/@0x1f0cc21
2019-10-02 06:46:35.601 1250-1630/? I/WindowManager: Cancelling animation restarting=false, leash=null, surface=Surface(name=3d7a3b3 br.com.oneplace/aplicativo.oneplace.Docker.Container.MainActivity)/@0x43f2846, parent=Surface(name=AppWindowToken{275586a token=Token{d86d655 ActivityRecord{2d73d0c u0 br.com.oneplace/aplicativo.oneplace.Docker.Container.MainActivity t54}}})/@

lucazin, 2019-10-02:
@natario1 do you think we can fix this problem?
thanks for you cooperation!

natario1, 2019-10-02:
Can you do this with the demo app? The demo app will add more interesting logs.

lucazin, 2019-10-02:
i will do now 

look this .. i think is acdb loader

`2019-10-02 06:46:31.570 673-2494/? I/audio_hw_utils: send_app_type_cfg_for_device CAPTURE app_type 69944, acdb_dev_id 56, sample_rate 48000, snd_device_be_idx 177
2019-10-02 06:46:31.570 673-2494/? E/ACDB-LOADER: Error: ACDB_CMD_GET_AUDPROC_INSTANCE_GAIN_DEP_STEP_TABLE_SIZE Returned = -19
2019-10-02 06:46:31.570 673-2494/? E/ACDB-LOADER: Error: ACDB AudProc vol returned = -19
2019-10-02 06:46:31.571 673-2494/? E/ACDB-LOADER: Error: ACDB_CMD_GET_AFE_INSTANCE_COMMON_TABLE_SIZE Returned = -19
2019-10-02 06:46:31.571 673-2494/? E/ACDB-LOADER: Error: ACDB AFE returned = -19
2019-10-02 06:46:31.571 673-2494/? I/msm8974_platform: platform_get_custom_mtmx_params: no matching param with id 0 ip_ch 1 op_ch 1 uc_id 19 snd_dev 258
2019-10-02 06:46:31.588 2040-2466/br.com.oneplace W/br.com.oneplac: 0xebadde09 skipped times: 0
    
    --------- beginning of crash
2019-10-02 06:46:31.590 2040-2466/br.com.oneplace A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 2466 (VideoEncoder), pid 2040 (br.com.oneplace)`

lucazin, 2019-10-02:
@natario1  **Here is the log cat  level INFO  in demo app** 

`2019-10-02 07:13:10.275 1250-1719/? I/InputReader: Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.3053 ] when=10196.788087
2019-10-02 07:13:10.276 1250-1718/? I/InputDispatcher: Delivering touch to (2098): action: 0x4, displayId: 0, '9f46511', toolType: 1
2019-10-02 07:13:10.276 1250-1718/? I/InputDispatcher: Delivering touch to (2098): action: 0x4, displayId: 0, '64440ab', toolType: 1
2019-10-02 07:13:10.276 1250-1718/? I/clp-JNI: Add pinfo) 0
2019-10-02 07:13:10.276 1250-1718/? I/InputDispatcher: Delivering touch to (6456): action: 0x0, displayId: 0, '26a5408', toolType: 1
2019-10-02 07:13:10.286 686-3812/? I/sensors-hal: handle_sns_client_event:78, auto_rotation_event: 0
2019-10-02 07:13:10.288 1250-1684/? I/CAE: parse(SensorHubParserProvider.java:192) - buffer size = 4
2019-10-02 07:13:10.288 1250-1684/? I/CAE: parse(SensorHubParserProvider.java:203) - 1, 1, 7, 0,
2019-10-02 07:13:10.288 1250-1684/? I/CAE: display(ContextProvider.java:391) - Angle=[0]
2019-10-02 07:13:10.345 686-3812/? I/sensors-hal: handle_sns_client_event:78, auto_rotation_event: 255
2019-10-02 07:13:10.346 1250-1684/? I/CAE: parse(SensorHubParserProvider.java:192) - buffer size = 4
2019-10-02 07:13:10.346 1250-1684/? I/CAE: parse(SensorHubParserProvider.java:203) - 1, 1, 7, -1,
2019-10-02 07:13:10.347 1250-1684/? I/CAE: display(ContextProvider.java:391) - Angle=[-1]
2019-10-02 07:13:10.356 1250-1719/? I/InputReader: Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=10196.868923
2019-10-02 07:13:10.356 1250-1718/? I/clp-JNI: Add pinfo) 26a5408 com.otaliastudios.cameraview.demo/com.otaliastudios.cameraview.demo.CameraActivity (server) : AppWindowToken{92525bc token=Token{e9f1daf ActivityRecord{6f5018e u0 com.otaliastudios.cameraview.demo/.CameraActivity t57}}} : 6456 : 10225 : 1
2019-10-02 07:13:10.356 1250-1718/? I/clp-JNI: Add pinfo) 1
2019-10-02 07:13:10.356 1250-1718/? I/InputDispatcher: Delivering touch to (6456): action: 0x1, displayId: 0, '26a5408', toolType: 1
2019-10-02 07:13:10.370 6456-6456/com.otaliastudios.cameraview.demo W/DemoApp: Recording snapshot for 5 seconds...
2019-10-02 07:13:10.379 1250-5351/? W/ContextImpl: Calling a method in the system process without a qualified user: android.app.ContextImpl.sendBroadcast:1067 com.android.server.notification.DisplayToast.sendIntentForToastDumpLog:156 com.android.server.notification.DisplayToast.outFile:106 com.android.server.notification.DisplayToast.out:62 com.android.server.notification.NotificationManagerService$12.enqueueToastForDisplay:2343 
2019-10-02 07:13:10.380 738-803/? I/SurfaceFlinger: id=7158 createSurf (4800x4800),2 flag=4, WindowToken{2d0b05e android.os.Binder@2a18599}#0
2019-10-02 07:13:10.385 6456-6495/com.otaliastudios.cameraview.demo I/Camera2Engine: onTakeVideoSnapshot rotation: 0 size: 1080x1440
2019-10-02 07:13:10.389 1250-5351/? I/FingerprintService: semGetIconBottomMargin called from pid=6456
2019-10-02 07:13:10.398 738-2772/? I/SurfaceFlinger: id=7159 createSurf (4800x4800),2 flag=4, d08f90c Toast#0
2019-10-02 07:13:10.406 738-803/? I/SurfaceFlinger: id=7160 createSurf (777x116),1 flag=4, Toast[6456]#0
2019-10-02 07:13:10.424 1250-1630/? I/WindowManager: commitFinishDrawingLocked: mDrawState=READY_TO_SHOW Surface(name=Toast[6456])/@0x3fbe255
2019-10-02 07:13:10.424 1250-1630/? I/WindowManager: Cancelling animation restarting=true, leash=null, surface=Surface(name=d08f90c Toast)/@0x5d299f8, parent=Surface(name=WindowToken{2d0b05e android.os.Binder@2a18599})/@0x3654ed1
2019-10-02 07:13:10.425 1250-1630/? I/WindowManager: Reparenting to leash, surface=Surface(name=d08f90c Toast)/@0x5d299f8
2019-10-02 07:13:10.425 738-1528/? I/SurfaceFlinger: id=7161 createSurf (4800x4800),2 flag=4, Surface(name=d08f90c Toast)/@0x5d299f8 - animation-leash#0
2019-10-02 07:13:10.427 6456-6508/com.otaliastudios.cameraview.demo I/SnapshotVideoRecorder: Starting the encoder engine.
2019-10-02 07:13:10.428 6456-6508/com.otaliastudios.cameraview.demo I/DeviceEncoders: findDeviceEncoder - type: video/avc encoders: 2
2019-10-02 07:13:10.428 6456-6508/com.otaliastudios.cameraview.demo I/DeviceEncoders: Enabled. Found video encoder: OMX.qcom.video.encoder.avc
2019-10-02 07:13:10.428 6456-6508/com.otaliastudios.cameraview.demo I/DeviceEncoders: findDeviceEncoder - type: audio/mp4a-latm encoders: 2
2019-10-02 07:13:10.428 6456-6508/com.otaliastudios.cameraview.demo I/DeviceEncoders: Enabled. Found audio encoder: OMX.SEC.naac.enc
2019-10-02 07:13:10.429 6456-6508/com.otaliastudios.cameraview.demo I/DeviceEncoders: getSupportedVideoSize - inputSize: 1080x1440 adjustedSize: 1080x1440
2019-10-02 07:13:10.429 6456-6508/com.otaliastudios.cameraview.demo I/DeviceEncoders: getSupportedVideoBitRate - inputRate: 3265920 adjustedRate: 3265920
2019-10-02 07:13:10.429 6456-6508/com.otaliastudios.cameraview.demo I/DeviceEncoders: getSupportedAudioBitRate - inputRate: 64000 adjustedRate: 64000
2019-10-02 07:13:10.432 6456-6508/com.otaliastudios.cameraview.demo I/DeviceEncoders: getSupportedVideoFrameRate - inputRate: 30 adjustedRate: 30
2019-10-02 07:13:10.435 1250-1518/? I/WindowManager: Cancelling animation restarting=true, leash=null, surface=null, parent=null
2019-10-02 07:13:10.435 1250-1518/? W/WindowManager: Unable to start animation, surface is null or no children.
2019-10-02 07:13:10.435 1250-1518/? I/WindowManager: Cancelling animation restarting=false, leash=null, surface=null, parent=null
[... cut ...]
2019-10-02 07:13:14.920 1250-1250/? I/MotionRecognitionService: Plugged
2019-10-02 07:13:14.946 2098-2098/? I/StackScroller: ADJUST STACK.TRANSLATION
2019-10-02 07:13:14.947 2098-2098/? I/StackScroller: ADJUST STACK.TRANSLATION
2019-10-02 07:13:14.947 2098-2098/? I/StackScroller: ADJUST STACK.TRANSLATION
2019-10-02 07:13:16.037 686-3812/? I/sensors-hal: handle_sns_client_event:83, auto_rotation_event: mode:255, type:1:1
2019-10-02 07:13:16.115 1250-1250/? W/WindowManager: removeWindowToken: Attempted to remove non-existing token: android.os.Binder@469

lucazin, 2019-10-02:
@natario1 Some parts for help

`2019-10-02 07:13:10.534 6456-7141/com.otaliastudios.cameraview.demo W/OMXUtils: do not know color format 0x7f000789 = 2130708361
2019-10-02 07:13:10.535 898-17076/? E/OMX-VENC: ERROR: unsupported index 1870659589

2019-10-02 07:13:10.547 6456-7141/com.otaliastudios.cameraview.demo I/ACodec: [OMX.qcom.video.encoder.avc] Now Loaded->Idle
2019-10-02 07:13:10.547 673-7150/? I/audio_hw_utils: send_app_type_cfg_for_device CAPTURE app_type 69944, acdb_dev_id 56, sample_rate 48000, snd_device_be_idx 177
2019-10-02 07:13:10.548 673-7150/? E/ACDB-LOADER: Error: ACDB_CMD_GET_AUDPROC_INSTANCE_GAIN_DEP_STEP_TABLE_SIZE Returned = -19
2019-10-02 07:13:10.548 673-7150/? E/ACDB-LOADER: Error: ACDB AudProc vol returned = -19
2019-10-02 07:13:10.548 673-7150/? E/ACDB-LOADER: Error: ACDB_CMD_GET_AFE_INSTANCE_COMMON_TABLE_SIZE Returned = -19
2019-10-02 07:13:10.548 673-7150/? E/ACDB-LOADER: Error: ACDB AFE returned = -19

2019-10-02 07:13:10.591 6456-7137/com.otaliastudios.cameraview.demo A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x6300750064006f in tid 7137 (VideoEncoder), pid 6456 (cameraview.demo)

2019-10-02 07:13:10.688 6456-7145/com.otaliastudios.cameraview.demo I/AudioMediaEncoder: read thread - eos: false - Read new audio frame. Bytes: 1024
2019-10-02 07:13:10.688 6456-7145/com.otaliastudios.cameraview.demo I/AudioMediaEncoder: read thread - eos: false - mLastTimeUs: 10197189645
2019-10-02 07:13:10.688 7155-7155/? I/crash_dump64: obtaining output fd from tombstoned, type: kDebuggerdTombstone
2019-10-02 07:13:10.689 965-965/? I//system/bin/tombstoned: received crash request for pid 7137
2019-10-02 07:13:10.690 6903-6903/? I/[UNWFC]-- Util: multiSim Device
2019-10-02 07:13:10.691 7155-7155/? I/crash_dump64: performing dump of process 6456 (target tid = 7137)
2019-10-02 07:13:10.695 6456-7146/com.otaliastudios.cameraview.demo I/AudioMediaEncoder: encoding thread - performing 1 pending operations.
2019-10-02 07:13:10.698 6456-7146/com.otaliastudios.cameraview.demo I/AudioMediaEncoder: encoding thread - performing pending operation for timestamp: 10197189645 - encoding.
2019-10-02 07:13:10.698 6456-7145/com.otaliastudios.cameraview.demo I/AudioMediaEncoder: read thread - eos: false - Read new audio frame. Bytes: 1024
2019-10-02 07:13:10.699 6456-7145/com.otaliastudios.cameraview.demo I/AudioMediaEncoder: read thread - eos: false - mLastTimeUs: 10197201254
2019-10-02 07:13:10.699 6456-7145/com.otaliastudios.cameraview.demo I/AudioMediaEncoder: read thread - eos: false - Read new audio frame. Bytes: 1024
2019-10-02 07:13:10.699 6456-7145/com.otaliastudios.cameraview.demo I/AudioMediaEncoder: read thread - eos: false - mLastTimeUs: 10197212864
2019-10-02 07:13:10.700 7155-7155/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2019-10-02 07:13:10.700 7155-7155/? A/DEBUG: Build fingerprint: 'samsung/a70qxtc/a70q:9/PPR1.180610.011/A705MNXXU3ASG7:user/release-keys'
2019-10-02 07:13:10.700 6456-7146/com.otaliastudios.cameraview.demo I/AudioMediaEncoder: encoding thread - performing pending operation for timestamp: 10197189645 - draining.
2019-10-02 07:13:10.700 6456-7146/com.otaliastudios.cameraview.demo I/MediaEncoder: AudioEncoder DRAINING - EOS: false
2019-10-02 07:13:10.703 6456-7145/com.otaliastudios.cameraview.demo I/AudioMediaEncoder: read thread - eos: false - Read new audio frame. Bytes: 1024
2019-10-02 07:13:10.704 6456-7145/com.otaliastudios.cameraview.demo I/AudioMediaEncoder: read thread - eos: false - mLastTimeUs: 10197224474
2019-10-02 07:13:10.704 6456-7145/com.otaliastudios.cameraview.demo I/AudioMediaEncoder: read thread - eos: false - Read new audio frame. Bytes: 1024
2019-10-02 07:13:10.705 6456-7145/com.otaliastudios.cameraview.demo I/AudioMediaEncoder: read thread - eos: false - mLastTimeUs: 10197236084
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG: Revision: '12'
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG: ABI: 'arm64'
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG: pid: 6456, tid: 7137, name: VideoEncoder  >>> com.otaliastudios.cameraview.demo <<<
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x6300750064006f
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x0  0000000000000000  x1  0000000000000000  x2  0000000000000000  x3  00000007fecdf710
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x4  00000077d7bda830  x5  0000000000000005  x6  0000000000000227  x7  8080808080808080
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x8  ffffffffffe2d754  x9  0000000000000001  x10 0000000000000001  x11 8080808080808080
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x12 8080808080808080  x13 8080808080808080  x14 0000000000000001  x15 00000007fecdf740
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x16 0000000000000600  x17 0000007879d6a7e8  x18 00000077d4ebd6f8  x19 00000077d85d6ff0
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x20 006300750064006f  x21 00000007fec15000  x22 00000077d869a080  x23 0000000000000000
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x24 0000000000000001  x25 0000000000630069  x26 00000077d9b0a388  x27 0000000000000001
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x28 0000000000000001  x29 00000077d4ebd940
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     sp  00000077d4ebd500  lr  00000077e330fe90  pc  00000077e330f498
2019-10-02 07:13:10.709 6456-6508/com.otaliastudios.cameraview.demo I/MediaEncoderEngine: Passing event to encoders: frame

`

[... cut ...]
2019-10-02 07:13:10.836 7155-7155/? A/DEBUG:     #38 pc 000000000045fe0c  /system/lib64/libart.so (art::InvokeVirtualOrInterfaceWithJValues(art::ScopedObjectAccessAlreadyRunnable const&, _jobject*, _jmethodID*, jvalue*)+424)
2019-10-02 07:13:10.836 7155-7155/? A/DEBUG:     #39 pc 000000000048b22c  /system/lib64/libart.so (art::Thread::CreateCallback(void*)+1120)
2019-10-02 07:13:10.836 7155-7155/? A/DEBUG:     #40 pc 0000000000091c3c  /system/lib64/libc.so (__pthread_start(void*)+36)
2019-10-02 07:13:10.836 7155-7155/? A/DEBUG:     #41 pc 0000000000024200  /system/lib64/libc.so (__start_thread+68)`

natario1, 2019-10-02:
I am attaching a file and hiding your comments, just to make this issue readable.

[Issue #606.txt](https://github.com/natario1/CameraView/files/3680871/Issue.606.txt)

Relevant parts:

```

2019-10-02 07:13:10.582 6456-7137/com.otaliastudios.cameraview.demo I/TextureMediaEncoder: onEvent - frameNumber: 2 timestampUs: 37856170158 - draining.
2019-10-02 07:13:10.582 6456-7137/com.otaliastudios.cameraview.demo I/MediaEncoder: VideoEncoder DRAINING - EOS: false
2019-10-02 07:13:10.582 6456-7137/com.otaliastudios.cameraview.demo I/TextureMediaEncoder: onEvent - frameNumber: 2 timestampUs: 37856170158 - rendering.
2019-10-02 07:13:10.584 6456-7137/com.otaliastudios.cameraview.demo I/OverlayLayout: draw target: VIDEO_SNAPSHOT canvas: 1080x1440 view: 1080x1440 widthScale: 1.0 heightScale: 1.0
2019-10-02 07:13:10.590 6456-7137/com.otaliastudios.cameraview.demo W/cameraview.dem: 0xebadde09 skipped times: 0
2019-10-02 07:13:10.591 6456-7137/com.otaliastudios.cameraview.demo A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x6300750064006f in tid 7137 (VideoEncoder), pid 6456 (cameraview.demo)
2019-10-02 07:13:10.691 7155-7155/? I/crash_dump64: performing dump of process 6456 (target tid = 7137)
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG: Revision: '12'
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG: ABI: 'arm64'
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG: pid: 6456, tid: 7137, name: VideoEncoder  >>> com.otaliastudios.cameraview.demo <<<
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x6300750064006f
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x0  0000000000000000  x1  0000000000000000  x2  0000000000000000  x3  00000007fecdf710
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x4  00000077d7bda830  x5  0000000000000005  x6  0000000000000227  x7  8080808080808080
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x8  ffffffffffe2d754  x9  0000000000000001  x10 0000000000000001  x11 8080808080808080
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x12 8080808080808080  x13 8080808080808080  x14 0000000000000001  x15 00000007fecdf740
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x16 0000000000000600  x17 0000007879d6a7e8  x18 00000077d4ebd6f8  x19 00000077d85d6ff0
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x20 006300750064006f  x21 00000007fec15000  x22 00000077d869a080  x23 0000000000000000
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x24 0000000000000001  x25 0000000000630069  x26 00000077d9b0a388  x27 0000000000000001
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     x28 0000000000000001  x29 00000077d4ebd940
2019-10-02 07:13:10.706 7155-7155/? A/DEBUG:     sp  00000077d4ebd500  lr  00000077e330fe90  pc  00000077e330f498
2019-10-02 07:13:10.834 7155-7155/? A/DEBUG: backtrace:
2019-10-02 07:13:10.834 7155-7155/? A/DEBUG:     #00 pc 000000000026a498  /vendor/lib64/egl/libGLESv2_adreno.so (EsxBltLib::CopyHostPtrBltSw(EsxBltCopyHostPtr const*)+416)
2019-10-02 07:13:10.834 7155-7155/? A/DEBUG:     #01 pc 0000000000261ff0  /vendor/lib64/egl/libGLESv2_adreno.so (EsxResource::UpdateSubResourceDirect(EsxContext*, unsigned int, int, EsxBox const*, EsxSurfaceAccessDesc const*, int)+1112)
2019-10-02 07:13:10.834 7155-7155/? A/DEBUG:     #02 pc 0000000000262d48  /vendor/lib64/egl/libGLESv2_adreno.so (EsxTextureObject::PerformShadowCopy(EsxContext*)+608)
2019-10-02 07:13:10.835 7155-7155/? A/DEBUG:     #03 pc 000000000024fa4c  /vendor/lib64/egl/libGLESv2_adreno.so (EsxContext::ValidateProgramTexUnitBoundObjs(EsxProgram*, int)+724)
2019-10-02 07:13:10.835 7155-7155/? A/DEBUG:     #04 pc 000000000024c648  /vendor/lib64/egl/libGLESv2_adreno.so (EsxContext::ValidateGfxState(EsxDrawDescriptor const*)+520)
2019-10-02 07:13:10.835 7155-7155/? A/DEBUG:     #05 pc 00000000002a4350  /vendor/lib64/egl/libGLESv2_adreno.so (EsxContext::DrawArraysInstanced(EsxPrimType, int, unsigned int, unsigned int)+200)
2019-10-02 07:13:10.835 7155-7155/? A/DEBUG:     #06 pc 0000000000441a28  /system/framework/arm64/boot-framework.oat (offset 0x41e000) (android.opengl.GLES10.glDrawArrays [DEDUPED]+168)
2019-10-02 07:13:10.835 7155-7155/? A/DEBUG:     #07 pc 00000000000573e0  /dev/ashmem/dalvik-jit-code-cache_6456_6456 (deleted) (com.otaliastudios.cameraview.filter.BaseFilter.onDraw+80)
2019-10-02 07:13:10.835 7155-7155/? A/DEBUG:     #08 pc 0000000000059560  /dev/ashmem/dalvik-jit-code-cache_6456_6456 (deleted) (com.otaliastudios.cameraview.filter.BaseFilter.draw+288)
2019-10-02 07:13:10.835 7155-7155/? A/DEBUG:     #09 pc 0000000000038cf8  /dev/ashmem/dalvik-jit-code-cache_6456_6456 (deleted) (com.otaliastudios.cameraview.internal.egl.EglViewport.drawFrame+392)
2019-10-02 07:13:10.835 7155-7155/? A/DEBUG:     #10 pc 00000000000840d8  /dev/ashmem/dalvik-jit-code-cache_6456_6456 (deleted) (com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onFrame+2808)
2019-10-02 07:13:10.835 7155-7155/? A/DEBUG:     #11 pc 0000000000559388  /system/lib64/libart.so (art_quick_invoke_stub+584)
2019-10-02 07:13:10.835 7155-7155/? A/DEBUG:     #12 pc 00000000000d02c8  /system/lib64/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+200)
2019-10-02 07:13:10.835 7155-7155/? A/DEBUG:     #13 pc 0000000000280cc0  /system/lib64/libart.so (art::interpreter::ArtInterpreterToCompiledCodeBridge(art::Thread*, art::ArtMethod*, art::ShadowFrame*, unsigned short, art::JValue*)+344)
[... cut ...]
2019-10-02 07:13:10.836 7155-7155/? A/DEBUG:     #38 pc 000000000045fe0c  /system/lib64/libart.so (art::InvokeVirtualOrInterfaceWithJValues(art::ScopedObjectAccessAlreadyRunnable const&, _jobject*, _jmethodID*, jvalue*)+424)
2019-10-02 07:13:10.836 7155-7155/? A/DEBUG:     #39 pc 000000000048b22c  /system/lib64/libart.so (art::Thread::CreateCallback(void*)+1120)
2019-10-02 07:13:10.836 7155-7155/? A/DEBUG:     #40 pc 0000000000091c3c  /system/lib64/libc.so (__pthread_start(void*)+36)
2019-10-02 07:13:10.836 7155-7155/? A/DEBUG:     #41 pc 0000000000024200  /system/lib64/libc.so (__start_thread+68)
```

lucazin, 2019-10-02:
But what kind of error is it ?

If this not work in my device..maybe not work in others new devices too..
Happen in official demo, in my app,  in RAN overlay app,  and just happer after i recording for many times with frontal camera..  its like lotery

Does have nothing we can do?  I will use this feature with some modification for my clients.
omg..  the overlay recording for me is essential =/

lucazin, 2019-10-02:
This happens just with takevideosnapshot ...with takevideo not happens

We can implement a overlay feature to takevideo method for i can put overlay please my friend ?

lucazin, 2019-10-02:
I need create something like storys about instagram.... with cameraview i was close to do that..
but with this error i dont know now..

But think with me..in instagram i can create story normally put all overlays e record de video ..with same mobile..

What can u suggest i do ?

lucazin, 2019-10-02:
@natario1 

I think I'll have to give up ..

How do I make the video recorded in takevideo mode in full screen of the phone?

because it only stays up to 1 part of the screen ..

I will have to record with takevideo even without overlay = /