natario1/CameraView #607: onConfigureFailed! Session android.hardware.camera2.impl.CameraCaptureSessionImpl@3a3e30a

bayunvnt, 2019-10-01:
Hallo, im using this library for my college project, than I have this issue **"onConfigureFailed! Session android.hardware.camera2.impl.CameraCaptureSessionImpl@3a3e30a"**, this error mostly occured on xiaomi device, can I have any solution? i hope this solve or you can give me any clue, thank you

natario1, 2019-10-07:
Can you fill the bug report information? Thanks!

coderJohnZhang, 2019-11-14:
I have met this issue, too.

11-14 16:46:21.587 19645 19645 E AndroidRuntime: FATAL EXCEPTION: main
11-14 16:46:21.587 19645 19645 E AndroidRuntime: Process: com.tcl.tvcamera, PID: 19645
11-14 16:46:21.587 19645 19645 E AndroidRuntime: java.lang.RuntimeException: onConfigureFailed! Session android.hardware.camera2.impl.CameraCaptureSessionImpl@74bb74e
11-14 16:46:21.587 19645 19645 E AndroidRuntime:        at com.tcl.cameraview.engine.Camera2Engine$4.onConfigureFailed(Camera2Engine.java:531)
11-14 16:46:21.587 19645 19645 E AndroidRuntime:        at android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy.lambda$onConfigureFailed$1(CallbackProxies.java:64)
11-14 16:46:21.587 19645 19645 E AndroidRuntime:        at android.hardware.camera2.impl.-$$Lambda$CallbackProxies$SessionStateCallbackProxy$gvbTsp9UPpKJAbdycdci_ZW5BeI.run(Unknown Source:4)
11-14 16:46:21.587 19645 19645 E AndroidRuntime:        at android.os.Handler.handleCallback(Handler.java:873)
11-14 16:46:21.587 19645 19645 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:99)
11-14 16:46:21.587 19645 19645 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:193)
11-14 16:46:21.587 19645 19645 E AndroidRuntime:        at android.os.HandlerThread.run(HandlerThread.java:65)

nsekhar90, 2020-02-05:
Im getting this same crash on Nexus 6P.

Is it possible for the library to not throw the exception (resulting in app crash) but report to client that initialization failed & the app can handle it gracefully ?

natario1, 2020-02-05:
@nsekhar90 do you still get this error if you choose a smaller picture/video/preview stram/frame processing size?

nsekhar90, 2020-02-05:
Im just using the default options and only for taking a picture (no video). I can try reducing the quality/size.

natario1, 2020-02-05:
Yeah please do. With camera.setPictureSize(). By default it's the biggest available. You can try camera.setPictureSize(SizeSelectors.smallest()) as an extreme example.

nsekhar90, 2020-02-05:
That fixes the issue. Is there anything I can do on the client side to adjust this based on device ?

natario1, 2020-02-05:
Nice to know. What if you **remove that line**, but instead you add `setPreviewStreamSize(SizeSelectors.smallest())` ? Does this fix the issue as well?

You can adjust based on the device with these methods, just pass a SizeSelector that fixes the issue

nsekhar90, 2020-02-05:
yeah, the `setPreviewStreamSize(SizeSelectors.smallest())` fixes the issue too. So, should I just check for a particular device (Nexus 6p) in this case & add the line `setPreviewStreamSize(SizeSelectors.smallest())` ?

natario1, 2020-02-05:
I would call something like 
`setPreviewStreamSize(SizeSelectors.and(SizeSelectors.maxWidth(MAX_WIDTH), SizeSelectors.biggest())`.

You should find the value of `MAX_WIDTH` by trying on your nexus (1000? 1200? 1400? 1900?). If you find an exact value please let me know.

nsekhar90, 2020-02-05:
1000 works best, can we calculate this runtime instead of hard coding it ?

nsekhar90, 2020-02-05:
Im asking about run time calculation because this crash might occur on other devices right ? I noticed that other people posted about this crash in the top.

Frayo44, 2020-03-11:
Are there any updates? Also got this on Redmi Note 7 & 8 & Galaxy S8.

natario1, 2020-03-11:
I'm not actively working on this but my suggestion is to limit the preview stream size with `setPreviewStreamSize()`. If you represent a company feel free to [sponsor](https://github.com/sponsors/natario1) me to let me work on this sooner

Frayo44, 2020-03-11:
I am not represent a company :/
How can I calculate MAX_WIDTH at runtime?

natario1, 2020-03-11:
I don't know - you should find something that works for your app, as it likely depends on the picture/video size that you are choosing, whether you are using frame processing or not, what is the frame processing size, ...

nxnyk, 2020-04-08:
I have the same issue on Redmi 8A

tomer8007, 2020-11-15:
@natario1 Have an idea of what's the root cause/what flows lead to the `onConfigureFailed` error? Or how to deterministically reproduce this?

natario1, 2020-11-16:
@tomer8007 it happens when the set of output sizes that we chose (preview stream size, frame processing size if you're using it, picture size when in Mode.PICTURE, video size when in Mode.VIDEO) is rejected by the device. We respect all limits for each size individually, but sometimes the combination of them is too much.

nestoleh, 2021-06-04:
Does anyone have a working solution for this problem? I got this error on devices Xiaomi (Redmi Note 7, Redmi Note 8, Redmi Note 8T, Redmi 8A) and Samsung (Galaxy A02s, Galaxy A01, Galaxy A10s)

1singhmanmeet, 2021-07-26:
It comes mostly on Red devices

munir700, 2022-01-03:
It comes frequently on Galaxy A02s device

sumit007, 2023-02-22:
Does anyone able to find a solution for this. I had tried the solution but I am still getting this issue in Redmi 10T 5G devices. 

> I would call something like `setPreviewStreamSize(SizeSelectors.and(SizeSelectors.maxWidth(MAX_WIDTH), SizeSelectors.biggest())`.
> 
> You should find the value of `MAX_WIDTH` by trying on your nexus (1000? 1200? 1400? 1900?). If you find an exact value please let me know.
> 
> Later we might improve this within the lib.