natario1/CameraView #610: AddView  Overlay

lucazin, 2019-10-01:
### How do I?
I have the last version 2.3.0   so i want to add a overlay view dynamic..
im doing like this

   ImageView image = new ImageView(this);
        image.setLayoutParams(new android.view.ViewGroup.LayoutParams(100,100));
        image.setImageURI(Uri.parse("https://scene7.zumiez.com/is/image/zumiez/pdp_hero/Married-To-The-Mob-Birdie-Sticker-_261461.jpg"));
        
        camera.addView(image,1,image.getLayoutParams());

but not show anything?

How can i do?

### Version used
2.3.0

lucazin, 2019-10-01:
i got it

`   final Bitmap[] myBitmap = {null};
    
        try {
            URL url = new URL("https://scene7.zumiez.com/is/image/zumiez/pdp_hero/Married-To-The-Mob-Birdie-Sticker-_261461.jpg");
            HttpURLConnection connection = (HttpURLConnection) url.openConnection();
            connection.setDoInput(true);
            connection.connect();
            InputStream input = connection.getInputStream();
            myBitmap[0] = BitmapFactory.decodeStream(input);
    
            ImageView image = new ImageView(CameraActivity.this);
            image.setLayoutParams(new android.view.ViewGroup.LayoutParams(200, 200));
            image.setImageBitmap(myBitmap[0]);
    
            image.setOnTouchListener(CameraActivity.this);
    
            camera.addView(image, 1, image.getLayoutParams());
            camera.onViewAdded(image);
        }
        catch (Exception e)
        {
        
        }`

But now i cant move it with draglistener .. how can i move this in runtime?

lucazin, 2019-10-01:
i check now.. when i record i video snapshot  my imageview dont show in video recorded.

just in camera preview, not in recorded state

lucazin, 2019-10-01:
`            ImageView image = new ImageView(CameraActivity.this);
            image.setLayoutParams(new android.view.ViewGroup.LayoutParams(200, 200));
            image.setImageBitmap(myBitmap[0]);
    
            image.setOnTouchListener(CameraActivity.this);
      
            OverlayLayout.LayoutParams params = new OverlayLayout.LayoutParams(200,200);
            
            camera.addView(image, 1, image.getLayoutParams());`

If i put image.getlayoutParams  its show my image in camera runtime .. but not after recorded snapshot

but if i put OverlayLayout params.. not work

natario1, 2019-10-02:
@lucazin please read the documentation about overlays https://natario1.github.io/CameraView/docs/watermarks-and-overlays.html . 

You must change `params. drawOnPreview`, `params. drawOnPictureSnapshot`, `params. drawOnVideoSnapshot` to the values that you want, before calling addView.

closed 2019-10-02 by natario1

lucazin, 2019-10-02:
Thanks natario... I can now work it overlays... My problem is with frontal camera exception