natario1/CameraView #615: Preview Rotation

Sv4nstaygeR, 2019-10-02:
Im developing an app for smartwatches with built in camera.

Normally this wouldn't be an issue but when i look the watch with the camera preview opened is
a bit weird because the camera its located at the right of the watch, so the frames in the
preview aren't very intuitive for looking cause they're rotated.

What im asking for is a method to rotate de Surface/Texture/GL view -> 0, 90, 180 , 270,
Im asking this because when i use android:rotation="90" in my xml layout file it don't do anything at all, even if include all the xml in another layout and trying to rotate this one, all the views get rotated but the cameraView persists the same way. 

Would be nice to have a method to check and set this, any other solution is accepted.

Thank you for all this project, its one of the best tools for dealing easy with cameras for devs. ^^

natario1, 2019-10-02:
Have you tried to use android:rotation with cameraPreview=texture?

Sv4nstaygeR, 2019-10-02:
As i mostly need the Shapeness filter i cant choose the texture or the surface.

natario1, 2019-10-02:
Then the only thing you can do is create your own `Filter` just like `NoFilter` but override `onPreDraw()` and modify the `transformMatrix` to apply your rotation.

I don't know if this will add distortions but maybe not.

Sv4nstaygeR, 2019-10-02:
Okay i'll try your aproach!

closed 2019-10-15 by natario1