natario1/CameraView #620: Changing the Flash value while camera2 is previewing not working

Ma7moudHatem, 2019-10-05:
### Describe the bug
I'm using the Camera2 engine, and while the camera opened and previewing I changed the Flash mode to ON, then called method CameraView.takePictureSnapshot, the Flash has not appeared while capturing the photo.
- CameraView version: *2.3.4*
- Camera engine used: *camera2*
- Reproducible in official demo app: *yes*
- Device / Android version: *HUAWEI Mate 9, API 24*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. While the camera opened and previewing, change the flash to ON
2. Call method CameraView.takePictureSnapshot
3. See error

### Expected behavior
The flash should be on while capturing the photo

natario1, 2019-10-05:
Snapshot is just a snapshot of the preview. If you want it to be similar to a picture, you should add `cameraPictureSnapshotMetering="true"`.

Ma7moudHatem, 2019-10-06:
How to use Snapshot video with flash, is it possible?

natario1, 2019-10-07:
If you want the flash to be on, just set it to `Flash.TORCH`

closed 2019-10-15 by natario1