natario1/CameraView #621: take snapshot NullPointerException

wosiu, 2019-10-06:
### Describe the bug
Exception after take snapshot.

- CameraView tested versions: 
  - 2.2.0 take snapshot works
  - 2.3.0 take snapshot does not work
- Camera engine used: camera2
- Reproducible in official demo app: *yes/no*
- Device / Android version: Huawei P10 Lite / Android 7.0

### To Reproduce
call takeSnapshot
Cannot reproduce in demo APP

### Logs
```
E/CameraEngine: uncaughtException: Unexpected exception: java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.Object android.hardware.camera2.CaptureResult.get(android.hardware.camera2.CaptureResult$Key)' on a null object reference
    java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.Object android.hardware.camera2.CaptureResult.get(android.hardware.camera2.CaptureResult$Key)' on a null object reference
        at com.otaliastudios.cameraview.picture.Snapshot2PictureRecorder.<init>(Snapshot2PictureRecorder.java:109)
        at com.otaliastudios.cameraview.engine.Camera2Engine.onTakePictureSnapshot(Camera2Engine.java:689)
        at com.otaliastudios.cameraview.engine.CameraEngine$22.run(CameraEngine.java:1164)
        at android.os.Handler.handleCallback(Handler.java:761)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:156)
        at android.os.HandlerThread.run(HandlerThread.java:61)
E/CameraEngine: Probably some deadlock in destroy. Current thread: Thread[main,5,main] Handler thread:  Thread[CameraViewEngine,5,]
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: eu.yesse.readerdemo.debug, PID: 32220
    java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.Object android.hardware.camera2.CaptureResult.get(android.hardware.camera2.CaptureResult$Key)' on a null object reference
        at com.otaliastudios.cameraview.picture.Snapshot2PictureRecorder.<init>(Snapshot2PictureRecorder.java:109)
        at com.otaliastudios.cameraview.engine.Camera2Engine.onTakePictureSnapshot(Camera2Engine.java:689)
        at com.otaliastudios.cameraview.engine.CameraEngine$22.run(CameraEngine.java:1164)
        at android.os.Handler.handleCallback(Handler.java:761)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:156)
        at android.os.HandlerThread.run(HandlerThread.java:61)
```

I haven't done any investigation yet. And because it is not reproducable in a demo app, the issue might be specific to my usage only. I'll try to get into that in a week or so.

natario1, 2019-10-15:
@wosiu it would be nice if you could provide more info.

natario1, 2019-10-16:
I think you are just calling takePictureSnapshot() very soon, before the first preview frame. I have removed the crash now.

closed 2019-10-16 by natario1

wosiu, 2019-10-16:
I haven't tested your changes yet, but I added a backoff time (1500ms) before I call snapshot after onCameraOpened and you're right - it helped.

commit 06fbf93f (2019-10-16): Make Holder.getLastResult nullable. Fixes #621
commit a63cd847 (2019-10-16): Bug fixes (#651)

* Use CopyOnWriteArrayList for Camera2 action list. Fixes #649

* Listen to suspicious display rotation events. Fixes #623

* Correctly release GL objects on activity pause. Fixes #648

* Add synchronization in SnapshotVideoRecorder. Fixes #642

* Make Holder.getLastResult nullable. Fixes #621

* Fix tests