natario1/CameraView #622: Front Camera Video Recording should have option to flip horizontally

usmanrana07, 2019-10-07:
### Problem to be addressed
Many applications are using front camera video recording as flipped horizontally i.e. SnapChat, Instagram etc

### Describe the solution you'd like
There should be feature in this library that will distinguish it from others, so user can capture videos flipped or not based upon his choice.

### Additional context
I've searched on it and some people recommended to work on low level apis and some suggested post processing on captured media file. Altering the captured frames not in  a good way may lead to slow rendering and latency. You've wrote this library and being experienced on camera side, you may have appropriate solution for that or atleast the direction that may lead towards good solution. 

Thanks

natario1, 2019-10-07:
You can use `takeVideoSnapshot()` and the video will be automatically flipped. Please take a look at the documentation

usmanrana07, 2019-10-07:
I've checked that, it still shows video in mirrored mode, not horizontally flipped like SnapChat does.

closed 2019-10-07 by usmanrana07