natario1/CameraView #623: camera preview is upside down when switch to reverse landscape mode.

SachinCTI, 2019-10-07:
If we enable the device rotation, and rotate our device to landscape to 180 landscape, the preview we get is upside down.

natario1, 2019-10-15:
@SachinCTI can you fill the bug information? Thanks.

SachinCTI, 2019-10-16:
**Describe the bug**
CameraView version: 2.2.0
Camera engine used: *camera2
Reproducible in official demo app: no
Device / Android version: xiaomi 6 pro pro, API 28

step1: Enable the auto rotation of device.
step2: Open the app in landscape mode. Preview we get is fine.
step3: Now flip the device to landscape to reverse landscape. (180 degree)
step4: Then the preview we get is upside down. 

[image]
[image]

commit a63cd847 (2019-10-16): Bug fixes (#651)

* Use CopyOnWriteArrayList for Camera2 action list. Fixes #649

* Listen to suspicious display rotation events. Fixes #623

* Correctly release GL objects on activity pause. Fixes #648

* Add synchronization in SnapshotVideoRecorder. Fixes #642

* Make Holder.getLastResult nullable. Fixes #621

* Fix tests
closed 2019-10-16 by natario1