natario1/CameraView #631: Preview becomes black and stucks in Note5 Android 7.0 in Demo app

usmanrana07, 2019-10-08:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.3.0*
- Camera engine used: *both*
- Reproducible in official demo app: *yes*
- Device / Android version: *Note 5 API 24*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Go to Settings
2. Choose Preview Surface 'gl surface'
3. Preview will become black and stucked 

### Expected behavior
It is working fine (but rendering is slow) in Huawei Y9 2019 Android Api 28. But in Note 5 it isn't working at all in gl surface mode.

natario1, 2019-10-15:
Can you fill the bug report information? Thanks.

usmanrana07, 2019-10-15:
It was misbehaving that day but didn't face that issue later on and in other devices. May be device was low at performance at that time.

closed 2019-10-15 by usmanrana07