natario1/CameraView #633: Camera2 RawFrameProcessor

xmas79, 2019-10-08:
Current implementation of FrameProcessor receive data in NV21 format, converted by the **convertToNV21** function. 

I'd like to receive raw frame bytes not converted by the **convertToNV21** in order to process them with custom functions.

natario1, 2019-10-15:
Hi, we do not plan to add RAW support but if you subclass Camera2Engine you should be able to get what you want.

closed 2019-10-15 by natario1

xmas79, 2019-11-08:
Hello, I subclassed the Camera2Engine and override the methods without any issues. However I'm not able to instantiate this new class because the constructor requires a "CameraCallbacks" class which is package private, so I'm unable to instantiate this class from my package scope and call the setEngine() function on CameraView. How should I proceed? Thanks.