natario1/CameraView #636: Calling 'takeVideoSnapShot' right after switching mode doesn't record video

usmanrana07, 2019-10-10:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.3.0*
- Camera engine used: *both*
- Reproducible in official demo app: *yes (but code needs some change)*
- Device / Android version: *All devices*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. set  Mode=PICTURE before calling 'takePictureSnapShot' on click on a button or set  Mode=VIDEO before calling 'takeVideoSnapShot' on long click on a button.
2. It will fail to record video or capture image in first try because camera closes and restarts when mode is changed. As recording method is called in very next line after changing the mode. So, it might not be ready and skips the recording and no action is done. 

### Expected behavior
There shouldn't be need of switching mode at our end each time before calling 'takePictureSnapShot' or 'takeVideoSnapShot' because if we're calling 'takeVideo' that means mode should be video and if it isn't that already then change it  (May be there is some concern behind it that I couldn't understand). Besides that it shouldn't fail recording when we setMode and call takePicture/takeVideo without giving  a delay.

natario1, 2019-12-28:
I believe this should work now.

closed 2019-12-28 by natario1

BhavyaRattan, 2020-04-03:
@natario1 I am facing the same issue with library version "2.6.2" can you guide me in the right direction?