natario1/CameraView #639: SnapShot capturing is low in quality than preview

usmanrana07, 2019-10-11:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.3.1
- Camera engine used: *camera1/camera2*
- Reproducible in official demo app: *yes*
- Device / Android version: *Note 8,API 28*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
Call 'takePictureSnapShot' method and you will see the captured picture has a bit lower quality/ blurry edges of objects. Although resolution is showing as correct.

usmanrana07, 2019-10-14:
@natario1 I just multiplied the resulted cropped width,height with 2 during debugging and result was improved (the resolution and size of captured image was doubled as expected).  Can you please add a method setSnapShotQualityMultiplexer method in this so, we can use it for that purpose. Most importantly if it is the right approach as you have broader idea of CameraView and there could be some more precise solution.

My change was in SnapshotGlPictureRecorder:

```
protected void onRendererTextureCreated(int textureId) {
        mTextureId = textureId;
        mViewport = new EglViewport();
        // Need to crop the size.
        Rect crop = CropHelper.computeCrop(mResult.size, mOutputRatio);
        mResult.size = new Size(crop.width()*2, crop.height()*2);  // doubled the size for testing
        mTransform = new float[16];
        Matrix.setIdentityM(mTransform, 0);

        if (mHasOverlay) {
            mOverlayDrawer = new OverlayDrawer(mOverlay, mResult.size);
        }
    }
```

natario1, 2019-10-15:
@usmanrana07 the demo app will crop down the image to `1000` pixels, in `PicturePreviewActivity`. But the real picture is probably better quality. You can try to change that

usmanrana07, 2019-10-15:
I'm using takePictureSpanShot method in my app , and I didn't use any crop property.

usmanrana07, 2019-10-23:
@natario1 can you please look into this

natario1, 2019-10-23:
@usmanrana07 Sorry I don't know what I could look at.

The camera sends a preview frame of fixed size say 100x100 then this *same frame* is passed to the preview and to the picture snapshot. So it is the very same source for two outputs.

If you set output size to 200x200 it will just be upscaled (missing pixels are interpolated / guessed) but it will not add any detail. The only thing that make sense is to downscale the output (e.g. 50x50) so you get a smaller picture, and for this we already have an API: `setSnapshotMaxWidth` and `setSnapshotMaxheight`.

usmanrana07, 2019-10-23:
the width and height is match_parent already. And calculation is correct. But the quality drops because of the point that if screen dimens are 1080x1920 then aspect ratio is 0.56 but as default camera orientation is landscaped so, in Preview, 1080 is considered as height and 1920 as width . When resulted image is cropped from it then output is 607x1080 maintaining the aspect ratio of 0.56

This causes quality to be lower than actual preview. the output should be 1080x1920 as preview is match_parent.

natario1, 2019-10-23:
@usmanrana07 I understand! You can use `wrap_content` to avoid aspect ratio cropping . You will just have black bands in the preview. But the resulting image will be as large as the preview stream.

Also you can use `setPreviewStreamSize()` to choose a bigger preview stream size, if the current one does not satisfy you.

manith-p, 2019-10-30:
@natario1 when I take a photo with takePictureSnapshot() the flash isn't triggered even I set the camera flash to Flash.On How can I trigger the flash when takePictureSnapShot()?

natario1, 2019-10-31:
@manith-p you should set pictureSnapshotMetering=true

kingbash90, 2019-11-05:
Flash not working with takePictureSnapshot(),even if when set cameraView.setPictureSnapshotMetering(true).

closed 2019-11-12 by natario1