natario1/CameraView #640: No encoders for type:video/avc

colorgold, 2019-10-11:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.3.1*
- Camera engine used: *both*
- Reproducible in official demo app: *yes*
- Device / Android version: *HUAWEI Y5 2018, Version: 8.1.0*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Go to 'CameraView'
2. Click on 'button to record video'
3. See error

### Expected behavior
The app will crash with an error dialog. The stack trace shows this problem:
Fatal Exception: java.lang.RuntimeException 
No encoders for type:video/avc
keyboard_arrow_up 
arrow_right 
com.otaliastudios.cameraview.internal.DeviceEncoders.findDeviceEncoder (DeviceEncoders.java:205) 

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
 <com.otaliastudios.cameraview.CameraView
                    android:id="@+id/addStory_cameraView"
                    android:layout_width="wrap_content"
                    android:layout_height="wrap_content"
                    android:layout_centerInParent="true"
                    android:keepScreenOn="true"
                    app:cameraGesturePinch="zoom"
                    app:cameraGestureScrollHorizontal="filterControl1"
                    app:cameraGestureScrollVertical="exposureCorrection"
                    app:cameraPlaySounds="true"
                    app:cameraGestureTap="autoFocus">

                    <TextView
                        android:layout_width="wrap_content"
                        android:layout_height="wrap_content"
                        android:id="@+id/addStory_tvRecordTime"
                        android:drawableStart="@drawable/ic_red_circle"
                        android:text="@string/time"
                        android:layout_gravity="end"
                        android:textSize="18sp"
                        android:textStyle="bold"
                        android:textColor="@color/white"
                        app:layout_drawOnPreview="true"
                        android:visibility="gone"/>
                </com.otaliastudios.cameraview.CameraView>
```

### Logs
Fatal Exception: java.lang.RuntimeException 
No encoders for type:video/avc
keyboard_arrow_up 
arrow_right 
com.otaliastudios.cameraview.internal.DeviceEncoders.findDeviceEncoder (DeviceEncoders.java:205) 

com.otaliastudios.cameraview.internal.DeviceEncoders.<init> (DeviceEncoders.java:109) 

com.otaliastudios.cameraview.video.FullVideoRecorder.onPrepareMediaRecorder (FullVideoRecorder.java:114) 

com.otaliastudios.cameraview.video.Full1VideoRecorder.onPrepareMediaRecorder (Full1VideoRecorder.java:45) 

com.otaliastudios.cameraview.video.FullVideoRecorder.prepareMediaRecorder (FullVideoRecorder.java:46) 

com.otaliastudios.cameraview.video.FullVideoRecorder.onStart (FullVideoRecorder.java:199) 

com.otaliastudios.cameraview.video.VideoRecorder.start (VideoRecorder.java:76) 

com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo (Camera1Engine.java:356) 

com.otaliastudios.cameraview.engine.CameraEngine$23.run (CameraEngine.java:1212) 

android.os.Handler.handleCallback (Handler.java:795) 

android.os.Handler.dispatchMessage (Handler.java:99) 

android.os.Looper.loop (Looper.java:166) 

android.os.HandlerThread.run (HandlerThread.java:65)

natario1, 2019-10-15:
Can you post logs from the demo app? From start to crash.

closed 2019-11-11 by stale[bot]

kiran4228, 2020-01-08:
How to resolve the above issue. I am facing the error like this.
java.lang.RuntimeException: No encoders for type:video/avc
	at com.otaliastudios.cameraview.f.b.a(SourceFile:15)
	at com.otaliastudios.cameraview.f.b.<init>(SourceFile:4)
	at com.otaliastudios.cameraview.l.h.a(SourceFile:27)
	at com.otaliastudios.cameraview.j.j$a.onDrawFrame(SourceFile:18)
	at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1575)
	at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1271)