natario1/CameraView #641: setParameters fails (and crash) when setting white balance

adrianhall, 2019-10-12:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: "com.otaliastudios:cameraview:2.3.1"
- Camera engine used: Engine.CAMERA1 (although engine is set to Engine.CAMERA2)
- Reproducible in official demo app: Don't know
- Device / Android version: Emulator - Pixel3, API 29, minAPI = 26, targetAPI = 29

### To Reproduce
Setting `camera.setWhiteBalance(WhiteBalance.FLOURESCENT)` results in the following crash:

```
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.tailwind.app.photos, PID: 12739
    java.lang.RuntimeException: setParameters failed
        at android.hardware.Camera.native_setParameters(Native Method)
        at android.hardware.Camera.setParameters(Camera.java:2117)
        at com.otaliastudios.cameraview.engine.Camera1Engine$3.run(Camera1Engine.java:516)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:67)
```

The referenced line in Camera1Engine is the setParameters line in setWhiteBalance.  I've tried both app:cameraEngine="camera1" and app:cameraEngine="camera2" - both result in the same crash (referencing Camera1Engine)

Note that this cannot be trapped through cameraError callback or try/catch block.

### Expected behavior

1) I'd expect to be able to set the WhiteBalance to any legal value.  If the value is not legal, it should throw at the setting point, allowing me to adjust
2) It should not crash my app.

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/take_picture_camera"
        android:layout_width="0dp"
        android:layout_height="0dp"
        android:layout_marginStart="64dp"
        android:keepScreenOn="true"
        app:cameraAudio="off"
        app:cameraEngine="camera2"
        app:layout_constraintBottom_toBottomOf="parent"
        app:layout_constraintEnd_toEndOf="parent"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintTop_toTopOf="parent" />
```

### Screenshots
If applicable, add screenshots to help explain your problem.

### Logs
Use `CameraLogger.setLogLevel(LEVEL_VERBOSE)` to see all logs into LogCat.

Use `CameraLogger.registerLogger()` to export to file or crash reporting service.

```
I/CameraEngine: Start: posting runnable. State: 0
W/CameraEngine: Start: executing runnable. AllState is 0
I/Step: ALL doStart Called. Enqueuing.
I/Step: ALL doStart About to start. Setting state to STARTING
    ENGINE doStart Called. Enqueuing.
I/Step: ENGINE doStart About to start. Setting state to STARTING
I/Camera1Engine: collectCameraInfo Facing: BACK Internal: 0 Cameras: 2
I/Angles: Angles changed: sensorOffset: 90 displayOffset: 0 deviceOrientation: 0
I/Choreographer: Skipped 100 frames!  The application may be doing too much work on its main thread.
I/Camera1Engine: onStartEngine: Applying default parameters.
I/OpenGLRenderer: Davey! duration=1692ms; Flags=0, IntendedVsync=7205537921522, Vsync=7207204588122, OldestInputEvent=9223372036854775807, NewestInputEvent=0, HandleInputStart=7207219765733, AnimationStart=7207219987733, PerformTraversalsStart=7207222586733, DrawStart=7207223318733, SyncQueued=7207224375733, SyncStart=7207225892733, IssueDrawCommandsStart=7207225931733, SwapBuffers=7207226699733, FrameCompleted=7207232219733, DequeueBufferDuration=243000, QueueBufferDuration=3897000, 
[... 66 lines cut ...]
    onMeasure: both are MATCH_PARENT or fixed value. We adapt. This means CROP_CENTER. (1080x1786)
I/CameraEngine: destroy: state: 2 thread: Thread[main,5,main]
I/CameraEngine: Stop: posting runnable. State: 2
E/CameraEngine: Probably some deadlock in destroy. Current thread: Thread[main,5,main] Handler thread:  Thread[CameraViewEngine,5,]
D/AndroidRuntime: Shutting down VM
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.tailwind.app.photos, PID: 13098
    java.lang.RuntimeException: setParameters failed
        at android.hardware.Camera.native_setParameters(Native Method)
        at android.hardware.Camera.setParameters(Camera.java:2117)
        at com.otaliastudios.cameraview.engine.Camera1Engine$3.run(Camera1Engine.java:516)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:67)
I/Process: Sending signal. PID: 13098 SIG: 9
```

(This is all logs from the first mention of Camera).

### APK
Link to a Github repo where the bug is reproducible.

https://github.com/adrianhall/tailwind-photos-for-android/tree/cv-bug-1
(although this has some pollution due to my diagnosis of the bug - for instance, I added an explicit engine set which didn't set the engine)

adrianhall, 2019-10-12:
Follow-up:
When setting

```
app:cameraExperimental="true"
app:cameraEngine="camera2"
```

This does not happen.

natario1, 2019-10-15:
@adrianhall thanks for reporting.

We check that everything is supported before applying, but sometimes the device lies about it. My answer to these crashes lately has been "please switch to Camera2".

Besides that, it might be an emulator-only issue.

adrianhall, 2019-10-15:
Agreed - thanks for the confirmation!

I might suggest selecting camera2 on >= MARSHMALLOW by default.

closed 2019-10-15 by adrianhall