natario1/CameraView #642: NullPointerException and RuntimeException caused crash

usmanrana07, 2019-10-12:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.3.1*
- Camera engine used: *camera2*
- Device / Android version: *Huawei, API 28*

During normal usage of app it suddenly stopped and gave following crash logs:

```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void com.otaliastudios.cameraview.video.encoding.MediaEncoderEngine.stop()' on a null object reference
       at com.otaliastudios.cameraview.video.SnapshotVideoRecorder.onStop + 92(SnapshotVideoRecorder.java:92)
       at com.otaliastudios.cameraview.video.VideoRecorder.stop + 90(VideoRecorder.java:90)
       at com.otaliastudios.cameraview.engine.Camera1Engine.onStopPreview + 250(Camera1Engine.java:250)
       at com.otaliastudios.cameraview.engine.CameraEngine$11.call + 543(CameraEngine.java:543)
       at com.otaliastudios.cameraview.engine.CameraEngine$11.call + 540(CameraEngine.java:540)
       at com.otaliastudios.cameraview.engine.Step$4.then + 155(Step.java:155)
       at com.otaliastudios.cameraview.engine.Step$4.then + 150(Step.java:150)
       at com.google.android.gms.tasks.zzf.run + 2(:2)
       at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run + 129(WorkerHandler.java:129)
       at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute + 104(WorkerHandler.java:104)
       at com.google.android.gms.tasks.zze.onComplete + 6(:6)
       at com.google.android.gms.tasks.zzr.zza + 19(:19)
       at com.google.android.gms.tasks.zzu.zze + 132(:132)
       at com.google.android.gms.tasks.zzu.continueWithTask + 74(:74)
       at com.otaliastudios.cameraview.engine.Step.doStop + 150(Step.java:150)
       at com.otaliastudios.cameraview.engine.Step.doStop + 143(Step.java:143)
       at com.otaliastudios.cameraview.engine.CameraEngine.stopPreview + 540(CameraEngine.java:540)
       at com.otaliastudios.cameraview.engine.CameraEngine.access$700 + 123(CameraEngine.java:123)
       at com.otaliastudios.cameraview.engine.CameraEngine$18$1.call + 777(CameraEngine.java:777)
       at com.otaliastudios.cameraview.engine.CameraEngine$18$1.call + 774(CameraEngine.java:774)
       at com.otaliastudios.cameraview.engine.Step$4.then + 155(Step.java:155)
       at com.otaliastudios.cameraview.engine.Step$4.then + 150(Step.java:150)
       at com.google.android.gms.tasks.zzf.run + 2(:2)
       at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run + 129(WorkerHandler.java:129)
       at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute + 104(WorkerHandler.java:104)
       at com.google.android.gms.tasks.zze.onComplete + 6(:6)
       at com.google.android.gms.tasks.zzr.zza + 19(:19)
       at com.google.android.gms.tasks.zzu.zze + 132(:132)
       at com.google.android.gms.tasks.zzu.continueWithTask + 74(:74)
       at com.otaliastudios.cameraview.engine.Step.doStop + 150(Step.java:150)
       at com.otaliastudios.cameraview.engine.Step.doStop + 143(Step.java:143)
       at com.otaliastudios.cameraview.engine.CameraEngine$18.run + 774(CameraEngine.java:774)
       at android.os.Handler.handleCallback + 907(Handler.java:907)
       at android.os.Handler.dispatchMessage + 105(Handler.java:105)
       at android.os.Looper.loop + 216(Looper.java:216)
       at android.os.HandlerThread.run + 65(HandlerThread.java:65)
```

```
Fatal Exception: java.lang.RuntimeException: Error during glCreateShader type=35633 glError 0x 501
       at com.otaliastudios.cameraview.internal.GlUtils.checkError + 31(GlUtils.java:31)
       at com.otaliastudios.cameraview.internal.GlUtils.loadShader + 45(GlUtils.java:45)
       at com.otaliastudios.cameraview.internal.GlUtils.createProgram + 61(GlUtils.java:61)
       at com.otaliastudios.cameraview.internal.egl.EglViewport.createProgram + 38(EglViewport.java:38)
       at com.otaliastudios.cameraview.internal.egl.EglViewport.drawFrame + 85(EglViewport.java:85)
       at com.otaliastudios.cameraview.picture.SnapshotGlPictureRecorder.takeFrame + 239(SnapshotGlPictureRecorder.java:239)
       at com.otaliastudios.cameraview.picture.SnapshotGlPictureRecorder$2.run + 156(SnapshotGlPictureRecorder.java:156)
       at android.os.Handler.handleCallback + 907(Handler.java:907)
       at android.os.Handler.dispatchMessage + 105(Handler.java:105)
       at android.os.Looper.loop + 216(Looper.java:216)
       at android.os.HandlerThread.run + 65(HandlerThread.java:65)
```

natario1, 2019-10-15:
Are these two separate issues or the same one?

usmanrana07, 2019-10-15:
It gave these 2 logs in fabric for a single crash

natario1, 2019-10-16:
Thanks for reporting!

commit a63cd847 (2019-10-16): Bug fixes (#651)

* Use CopyOnWriteArrayList for Camera2 action list. Fixes #649

* Listen to suspicious display rotation events. Fixes #623

* Correctly release GL objects on activity pause. Fixes #648

* Add synchronization in SnapshotVideoRecorder. Fixes #642

* Make Holder.getLastResult nullable. Fixes #621

* Fix tests
closed 2019-10-16 by natario1