natario1/CameraView #645: setParameters failed and crash on Android 4.4.4

fanxxxx, 2019-10-14:
### Describe the bug
- CameraView version: 2.2.0
- Camera engine used: camera1
- Reproducible in official demo app: no (no device)
- Device / Android version: Xiaomi Mi3. Android 4.4.4(level 19)

I got a lot of crash report on Xiaomi Mi3, here's crash log:

### Logs
```
1 android.hardware.Camera.native_setParameters(Native Method)
2 android.hardware.Camera.setParameters(Camera.java:1671)
3 com.otaliastudios.cameraview.engine.Camera1Engine.boolean applyZoom(android.hardware.Camera$Parameters,float)(Camera1Engine.java:16)
4 com.otaliastudios.cameraview.engine.Camera1Engine.void applyAllParameters(android.hardware.Camera$Parameters)(Camera1Engine.java:37)
5 com.otaliastudios.cameraview.engine.Camera1Engine.com.google.android.gms.tasks.Task onStartEngine()(Camera1Engine.java:6)
6 com.otaliastudios.cameraview.engine.CameraEngine$3.com.google.android.gms.tasks.Task call()(CameraEngine.java:3)
7 com.otaliastudios.cameraview.engine.CameraEngine$3.java.lang.Object call()(CameraEngine.java:1)
8 com.otaliastudios.cameraview.engine.Step$2.com.google.android.gms.tasks.Task then(com.google.android.gms.tasks.Task)(Step.java:4)
9 com.otaliastudios.cameraview.engine.Step$2.java.lang.Object then(com.google.android.gms.tasks.Task)(Step.java:1)
10 com.google.android.gms.tasks.zzf.void run()(Unknown Source)
11 ##_parent_##4##_parent_##
12 ##_child_## void run()##_child_##
13 ##_child_## void run()##_child_##
14 ##_child_## void run()##_child_##
15 ##_child_## void run()##_child_##
16 com.otaliastudios.cameraview.internal.utils.WorkerHandler.void run(java.lang.Runnable)(WorkerHandler.java:2)
17 com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.void execute(java.lang.Runnable)(WorkerHandler.java:1)
18 com.google.android.gms.tasks.zze.void onComplete(com.google.android.gms.tasks.Task)(Unknown Source)
19 com.google.android.gms.tasks.zzr.void zza(com.google.android.gms.tasks.zzq)(Unknown Source)
20 ##_parent_##6##_parent_##
21 ##_child_## void zza(com.google.android.gms.tasks.Task)##_child_##
22 ##_child_## void zza(com.google.android.gms.tasks.Task)##_child_##
23 ##_child_## void zza(com.google.android.gms.tasks.Task)##_child_##
24 ##_child_## void zza(com.google.android.gms.tasks.Task)##_child_##
25 ##_child_## void zza(com.google.android.gms.tasks.Task)##_child_##
26 ##_child_## void zza(com.google.android.gms.tasks.Task)##_child_##
27 com.google.android.gms.tasks.zzu.void zze()(Unknown Source)
28 ##_parent_##1##_parent_##
29 ##_child_## void zze()##_child_##
30 com.google.android.gms.tasks.zzu.com.google.android.gms.tasks.Task continueWithTask(com.google.android.gms.tasks.Continuation)(Unknown Source)
31 ##_parent_##1##_parent_##
32 ##_child_## com.google.android.gms.tasks.Task continueWithTask(java.util.concurrent.Executor,com.google.android.gms.tasks.Continuation)##_child_##
33 com.otaliastudios.cameraview.engine.Step.com.google.android.gms.tasks.Task doStart(boolean,java.util.concurrent.Callable,java.lang.Runnable)(Step.java:7)
34 com.otaliastudios.cameraview.engine.CameraEngine.com.google.android.gms.tasks.Task startBind()(CameraEngine.java:2)
35 com.otaliastudios.cameraview.engine.CameraEngine.com.google.android.gms.tasks.Task access$1100(com.otaliastudios.cameraview.engine.CameraEngine)(CameraEngine.java:1)
36 com.otaliastudios.cameraview.engine.CameraEngine$17$1.com.google.android.gms.tasks.Task call()(CameraEngine.java:2)
37 com.otaliastudios.cameraview.engine.CameraEngine$17$1.java.lang.Object call()(CameraEngine.java:1)
38 com.otaliastudios.cameraview.engine.Step$2.com.google.android.gms.tasks.Task then(com.google.android.gms.tasks.Task)(Step.java:4)
39 com.otaliastudios.cameraview.engine.Step$2.java.lang.Object then(com.google.android.gms.tasks.Task)(Step.java:1)
40 com.google.android.gms.tasks.zzf.void run()(Unknown Source)
41 ##_parent_##4##_parent_##
42 ##_child_## void run()##_child_##
43 ##_child_## void run()##_child_##
44 ##_child_## void run()##_child_##
45 ##_child_## void run()##_child_##
46 com.otaliastudios.cameraview.internal.utils.WorkerHandler.void run(java.lang.Runnable)(WorkerHandler.java:2)
47 com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.void execute(java.lang.Runnable)(WorkerHandler.java:1)
48 com.google.android.gms.tasks.zze.void onComplete(com.google.android.gms.tasks.Task)(Unknown Source)
49 com.google.android.gms.tasks.zzr.void zza(com.google.android.gms.tasks.zzq)(Unknown Source)
50 ##_parent_##6##_parent_##
51 ##_child_## void zza(com.google.android.gms.tasks.Task)##_child_##
52 ##_child_## void zza(com.google.android.gms.tasks.Task)##_child_##
53 ##_child_## void zza(com.google.android.gms.tasks.Task)##_child_##
54 ##_child_## void zza(com.google.android.gms.tasks.Task)##_child_##
55 ##_child_## void zza(com.google.android.gms.tasks.Task)##_child_##
56 ##_child_## void zza(com.google.android.gms.tasks.Task)##_child_##
57 com.google.android.gms.tasks.zzu.void zze()(Unknown Source)
58 ##_parent_##1##_parent_##
59 ##_child_## void zze()##_child_##
60 com.google.android.gms.tasks.zzu.com.google.android.gms.tasks.Task continueWithTask(com.google.android.gms.tasks.Continuation)(Unknown Source)
61 ##_parent_##1##_parent_##
62 ##_child_## com.google.android.gms.tasks.Task continueWithTask(java.util.concurrent.Executor,com.google.android.gms.tasks.Continuation)##_child_##
63 com.otaliastudios.cameraview.engine.Step.com.google.android.gms.tasks.Task doStart(boolean,java.util.concurrent.Callable,java.lang.Runnable)(Step.java:7)
64 com.otaliastudios.cameraview.engine.Step.com.google.android.gms.tasks.Task doStart(boolean,java.util.concurrent.Callable)(Step.java:5)
65 com.otaliastudios.cameraview.engine.CameraEngine$17.void run()(CameraEngine.java:2)
66 android.os.Handler.handleCallback(Handler.java:733)
67 android.os.Handler.dispatchMessage(Handler.java:95)
68 android.os.Looper.loop(Looper.java:136)
69 android.os.HandlerThread.run(HandlerThread.java:61)
```

natario1, 2019-10-15:
No... these things happen on Camera1 devices and there's no clue about what's wrong. Probably a device specific issue. I'm not aware of any solution and I don't own this particular device.

fanxxxx, 2019-10-15:
It's possible to add try catch at `setParameters` and callback on `onCameraError`?

closed 2019-11-11 by stale[bot]