natario1/CameraView #646: Watermark/ Overlay Touch events

colorgold, 2019-10-14:
### How do I?
I have been trying to drag the watermark when taking a video but the touch events are consumed by the cameraview. I followed this link, [https://ran3000.github.io/2019/09/14/cameraview-interactivity.html](url), to forward touch events to the watermark. However, the image cannot be dragged, only clicked on. Is there a way I can accomplish this?

### Version used
2.3.1

natario1, 2019-10-15:
Maybe the author can help you, @RAN3000 

You could try calling `cameraView.requestDisallowInterceptTouchEvent(true)`. Note that gestures might not work after that.

RAN3000, 2019-10-15:
Hi!

I haven't tested this but you can try to override `onDragEvent(DragEvent)` in `ForwardTouchesActivity` from the article, in the same way as `onTouchEvent(MotionEvent)`.

Reference: [https://developer.android.com/guide/topics/ui/drag-drop#AboutDragListeners](https://developer.android.com/guide/topics/ui/drag-drop#AboutDragListeners)

closed 2019-11-11 by stale[bot]

ddyos, 2020-09-07:
@natario1 
Hi.
I meet a similar problem  like this issue.
The onInterceptTouchEvent() in CameraView return true, and this override block child view to handle touch event.

Question: Why not  add a config to control the return value of the method onInterceptTouchEvent()?
Thank you！