natario1/CameraView #648: I user MultiFilter ，when onResume，app crash

caiyoufei, 2019-10-15:
java.lang.RuntimeException: Error during glUniformMatrix4fv glError 0x 501
        at com.otaliastudios.cameraview.internal.GlUtils.checkError(GlUtils.java:25)
        at com.otaliastudios.cameraview.filter.BaseFilter.onPreDraw(BaseFilter.java:192)
        at com.otaliastudios.cameraview.filter.BaseFilter.draw(BaseFilter.java:182)
        at com.otaliastudios.cameraview.filter.MultiFilter.draw(MultiFilter.java:260)
        at com.otaliastudios.cameraview.internal.egl.EglViewport.drawFrame(EglViewport.java:93)
        at com.otaliastudios.cameraview.preview.GlCameraPreview$Renderer.onDrawFrame(GlCameraPreview.java:215)
        at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1590)
        at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1286)

caiyoufei, 2019-10-15:
[image]
change demo code

natario1, 2019-10-16:
Thanks for reporting!

closed 2019-10-16 by natario1

harikumar007, 2020-01-06:
Fatal Exception: java.lang.RuntimeException
Error during glCreateShader type=35633 glError 0x 501
com.otaliastudios.cameraview.internal.GlUtils.checkError (GlUtils.java:31)

Getting this error when using takeSnapShot option...

commit a63cd847 (2019-10-16): Bug fixes (#651)

* Use CopyOnWriteArrayList for Camera2 action list. Fixes #649

* Listen to suspicious display rotation events. Fixes #623

* Correctly release GL objects on activity pause. Fixes #648

* Add synchronization in SnapshotVideoRecorder. Fixes #642

* Make Holder.getLastResult nullable. Fixes #621

* Fix tests