natario1/CameraView #650: NullPointerException setPreviewTexture

stefanJi, 2019-10-16:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: 2.3.1
- Camera engine used: camera2
- Reproducible in official demo app:no
- Device / Android version: *HUAWEI MLA-AL10 ANDROID 6.0*

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/bell_cameraView"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        app:cameraAudio="off"
        app:cameraFacing="front"
        app:cameraFlash="off"
        app:cameraPreview="glSurface"
        app:cameraEngine="camera2"
        app:cameraExperimental="true"
        app:cameraPlaySounds="false" />
```

### Logs

```
java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.setPreviewTexture(android.graphics.SurfaceTexture)' on a null object reference
	at com.otaliastudios.cameraview.b.bAY(SourceFile:120)
	at com.otaliastudios.cameraview.b.d(SourceFile:27)
	at com.otaliastudios.cameraview.b$10.run(SourceFile:78)
	at com.otaliastudios.cameraview.b$8.run(SourceFile:63)
	at android.os.Handler.handleCallback(Handler.java:743)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:150)
	at android.os.HandlerThread.run(HandlerThread.java:61)
```

stefanJi, 2019-10-18:
I am also confused. I declare to use Camera2, but it looks like Camera1 is used from the stack trace.

natario1, 2019-10-18:
This means you have not set cameraExperimental to true, OR API level is < 21 (or both)

closed 2019-11-14 by stale[bot]