natario1/CameraView #652: App crashes when using camera2 Engine due to concurency

113408, 2019-10-17:
### Describe the bug
I recently moved from a similar library to yours, I find it great in processing the pictues (mettering option lacked on the other one) however I face a critical issue when using `camera.takePictue()` inside a timer. The app crashed as it tries to get results before the previous request was completed. I checked a bit the code and saw that the `baseEngine` used some kind of lock to *skip/abort* taking a picture when another one is still being processed and that's the behavior I'm expecting.
The same code works perfectly using `camera1Engine`. 

As pointed below on the logs, a null check is missing in this part of the code (this is the quick fix without digging deeper). LastResult is nullable and should be safely check when considering the state?
```
@Override
    protected boolean checkShouldSkip(@NonNull ActionHolder holder) {
        Integer aeState = holder.getLastResult(this)?.get(CaptureResult.CONTROL_AE_STATE);
        boolean result = aeState != null && aeState == CaptureResult.CONTROL_AE_STATE_CONVERGED;
        LOG.i("checkShouldSkip:", result);
        return result;
    }
```
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version:  *2.3.1
- Camera engine used: *camera2*
- Reproducible in official demo app: *n/a*
- Device / Android version: *All devices/all versions*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Setup clean/new project with the bear minimum
2. init the camera and the timer like below
```
class MainActivity : AppCompatActivity() {

    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        setContentView(R.layout.activity_main)
        camera.setLifecycleOwner(this)
        camera.addCameraListener(object : CameraListener() {
            override fun onCameraOpened(options: CameraOptions) {
                Timer().scheduleAtFixedRate(object: TimerTask(){
                    override fun run() {
                        camera.takePicture()
                    }
                },0,1)
            }
            override fun onPictureTaken(result: PictureResult) {
                Log.d("TEST","Picture was taken")
                val now = System.currentTimeMillis()
                result.toFile(File(Environment.getExternalStorageDirectory(),now.toString())) {
                        file -> Log.d("TEST","Picture was saved to: ${file?.absolutePath}")
                }
            }
        })
    }
}
```
3. Make sure to use Camera 2 engine and and experimental on the XML file
### Expected behavior
The expected behavior is somehow similar to the *Snapshot* one however I need a full resolution image. If the camera process is still running it should ignore any other call to `takePictue()` until it's job is done.

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```
<com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:keepScreenOn="true"
        android:layout_height="0dp"
        android:layout_width="0dp"
        app:cameraEngine="camera2"
        app:cameraExperimental="true"
        app:cameraPlaySounds="false"
        app:cameraAudio="off"
        app:cameraMode="picture"
        app:layout_constraintBottom_toBottomOf="parent"
        app:layout_constraintLeft_toLeftOf="parent"
        app:layout_constraintRight_toRightOf="parent"
        app:layout_constraintTop_toTopOf="parent" />
```

### Logs
```
java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.Object android.hardware.camera2.TotalCaptureResult.get(android.hardware.camera2.CaptureResult$Key)' on a null object reference
        at com.otaliastudios.cameraview.engine.meter.ExposureMeter.checkShouldSkip(ExposureMeter.java:53)
        at com.otaliastudios.cameraview.engine.meter.BaseMeter.onStart(BaseMeter.java:34)
        at com.otaliastudios.cameraview.engine.action.TogetherAction.onStart(TogetherAction.java:47)
        at com.otaliastudios.cameraview.engine.action.ActionWrapper.onStart(ActionWrapper.java:37)
        at com.otaliastudios.cameraview.engine.meter.MeterAction.onStart(MeterAction.java:77)
        at com.otaliastudios.cameraview.engine.action.ActionWrapper.onStart(ActionWrapper.java:37)
        at com.otaliastudios.cameraview.engine.action.TimeoutAction.onStart(TimeoutAction.java:35)
        at com.otaliastudios.cameraview.engine.action.BaseAction.start(BaseAction.java:42)
        at com.otaliastudios.cameraview.engine.Camera2Engine.onTakePicture(Camera2Engine.java:723)
        at com.otaliastudios.cameraview.engine.CameraEngine$21.run(CameraEngine.java:1134)
        at android.os.Handler.handleCallback(Handler.java:789)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)
```
I hope this will help you resolve this bug and thanks for the great work

natario1, 2019-10-17:
This was fixed by #651, but thanks very much for detailed report!

closed 2019-10-17 by natario1

113408, 2019-10-21:
Indeed the #651 solves the issue related to `LastResult` however in the same scenario the `holder` itself is sometimes null if metering is turned on. If metering is off then there are no issues. I tried to dig the issue and wrap the holder in a null check (after making it nullable) but the preview just stops at some point (freeze).
Without checking the nullable holder I get an exception as follow:
```
java.lang.NullPointerException: Attempt to invoke interface method 'void com.otaliastudios.cameraview.engine.action.ActionHolder.removeAction(com.otaliastudios.cameraview.engine.action.Action)' on a null object reference
        at com.otaliastudios.cameraview.engine.action.BaseAction.setState(BaseAction.java:106)
        at com.otaliastudios.cameraview.engine.meter.WhiteBalanceMeter.onCaptureCompleted(WhiteBalanceMeter.java:81)
        at com.otaliastudios.cameraview.engine.action.TogetherAction.onCaptureCompleted(TogetherAction.java:83)
        at com.otaliastudios.cameraview.engine.action.ActionWrapper.onCaptureCompleted(ActionWrapper.java:65)
        at com.otaliastudios.cameraview.engine.action.ActionWrapper.onCaptureCompleted(ActionWrapper.java:65)
        at com.otaliastudios.cameraview.engine.action.TimeoutAction.onCaptureCompleted(TimeoutAction.java:42)
        at com.otaliastudios.cameraview.engine.Camera2Engine$1.onCaptureCompleted(Camera2Engine.java:292)
        at android.hardware.camera2.impl.CameraCaptureSessionImpl$1.lambda$onCaptureCompleted$3(CameraCaptureSessionImpl.java:640)
        at android.hardware.camera2.impl.-$$Lambda$CameraCaptureSessionImpl$1$OA1Yz_YgzMO8qcV8esRjyt7ykp4.run(Unknown Source:8)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:65)
```

Any idea as to where the check should happen or if the right signal would be to abort the session in case of null holder ?

natario1, 2019-10-22:
Hi @113408 could you open a new issue? Providing `INFO` logs if you can.