natario1/CameraView #658: Multiple fragment using camera instance

ManuEpi, 2019-10-23:
### How do I?

Hello, I have a big issue actually, I hope that someone could help me, I'm stuck on it since days.

I'm using Android Navigation component (Jetpack) . And I have a lateral menu (com.google.android.material.navigation.NavigationView) which will call the corresponding Fragment depending on choice that we make.

I'm in Fragment A which has cameraView in full screen
I click on my lateral menu and select Fragment B.
It calls `nav_view.setupWithNavController(findNavController(R.id.navHost))`

And this what happen :
FragmentB : onResume ( so cameraView.open() )
FragmentA: onPause ( cameraView.close() )
FragmentA: onDestroy ( cameraView.destroy() )

It will clearly open the FragmentB camera before that FragmentA release his camera Instance

And I have an exception : 

`E/Camera1Engine: onStartEngine: Failed to connect. Maybe in use by another app?
E/CameraEngine: uncaughtException: Got CameraException: com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Fail to connect to camera service on engine state: 1
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Fail to connect to camera service
        at com.otaliastudios.cameraview.engine.Camera1Engine.onStartEngine(Camera1Engine.java:152)
        at com.otaliastudios.cameraview.engine.CameraEngine$3.call(CameraEngine.java:382)
        at com.otaliastudios.cameraview.engine.CameraEngine$3.call(CameraEngine.java:375)
        at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:117)
        at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:112)
        at com.google.android.gms.tasks.zzf.run(Unknown Source)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:129)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute(WorkerHandler.java:104)
        at com.google.android.gms.tasks.zze.onComplete(Unknown Source)
        at com.google.android.gms.tasks.zzr.zza(Unknown Source)
        at com.google.android.gms.tasks.zzu.zze(Unknown Source)
        at com.google.android.gms.tasks.zzu.continueWithTask(Unknown Source)
        at com.otaliastudios.cameraview.engine.Step.doStart(Step.java:112)
        at com.otaliastudios.cameraview.engine.CameraEngine.startEngine(CameraEngine.java:375)
        at com.otaliastudios.cameraview.engine.CameraEngine.access$1100(CameraEngine.java:123)
        at com.otaliastudios.cameraview.engine.CameraEngine$17$1.call(CameraEngine.java:722)
        at com.otaliastudios.cameraview.engine.CameraEngine$17$1.call(CameraEngine.java:719)
        at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:117)
        at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:112)
        at com.google.android.gms.tasks.zzf.run(Unknown Source)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:129)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute(WorkerHandler.java:104)
        at com.google.android.gms.tasks.zze.onComplete(Unknown Source)
        at com.google.android.gms.tasks.zzr.zza(Unknown Source)
        at com.google.android.gms.tasks.zzu.zze(Unknown Source)
        at com.google.android.gms.tasks.zzu.continueWithTask(Unknown Source)
        at com.otaliastudios.cameraview.engine.Step.doStart(Step.java:112)
        at com.otaliastudios.cameraview.engine.Step.doStart(Step.java:105)
        at com.otaliastudios.cameraview.engine.CameraEngine$17.run(CameraEngine.java:719)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:145)
        at android.os.HandlerThread.run(HandlerThread.java:61)
     Caused by: java.lang.RuntimeException: Fail to connect to camera service
        at android.hardware.Camera.<init>(Camera.java:565)
        at android.hardware.Camera.open(Camera.java:405)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onStartEngine(Camera1Engine.java:149)
        at com.otaliastudios.cameraview.engine.CameraEngine$3.call(CameraEngine.java:382) 
        at com.otaliastudios.cameraview.engine.CameraEngine$3.call(CameraEngine.java:375) 
        at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:117) 
        at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:112) 
        at com.google.android.gms.tasks.zzf.run(Unknown Source) 
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:129) 
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute(WorkerHandler.java:104) 
        at com.google.android.gms.tasks.zze.onComplete(Unknown Source) 
        at com.google.android.gms.tasks.zzr.zza(Unknown Source) 
        at com.google.android.gms.tasks.zzu.zze(Unknown Source) 
        at com.google.android.gms.tasks.zzu.continueWithTask(Unknown Source) 
        at com.otaliastudios.cameraview.engine.Step.doStart(Step.java:112) 
        at com.otaliastudios.cameraview.engine.CameraEngine.startEngine(CameraEngine.java:375) 
        at com.otaliastudios.cameraview.engine.CameraEngine.access$1100(CameraEngine.java:123) 
        at com.otaliastudios.cameraview.engine.CameraEngine$17$1.call(CameraEngine.java:722) 
        at com.otaliastudios.cameraview.engine.CameraEngine$17$1.call(CameraEngine.java:719) 
        at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:117) 
        at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:112) 
        at com.google.android.gms.tasks.zzf.run(Unknown Source) 
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:129) 
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute(WorkerHandler.java:104) 
        at com.google.android.gms.tasks.zze.onComplete(Unknown Source) 
        at com.google.android.gms.tasks.zzr.zza(Unknown Source) 
        at com.google.android.gms.tasks.zzu.zze(Unknown Source) 
        at com.google.android.gms.tasks.zzu.continueWithTask(Unknown Source) 
        at com.otaliastudios.cameraview.engine.Step.doStart(Step.java:112) 
        at com.otaliastudios.cameraview.engine.Step.doStart(Step.java:105) 
        at com.otaliastudios.cameraview.engine.CameraEngine$17.run(CameraEngine.java:719) 
        at android.os.Handler.handleCallback(Handler.java:739) 
        at android.os.Handler.dispatchMessage(Handler.java:95) 
        at android.os.Looper.loop(Looper.java:145) 
        at android.os.HandlerThread.run(HandlerThread.java:61) `

Hope that it's clear. Thanks

### Version used
Latest

natario1, 2019-10-23:
You must make sure that `close()`, or even better, `destroy()`, is called before the other fragment's `open()`.

About how to do that that'd be just an implementation detail of your app

ManuEpi, 2019-10-23:
Thanks for your response.

I don't really know where to do the implementation because it's the MainActivity who make the switch between FragmentA and FragmentB , it doesn't have instance of cameraView. 

Only both fragments have it , and in FragmentA , the onPause() is called AFTER the onResume() of the new FragmentB so I cannot make sure to close() destroy() before that the other fragment open().

Do you have any idea ?

natario1, 2019-10-23:
FragmentB.onResume() can ask the activity to check that FragmentA has released the camera, or you can use some navigation callback of the navigation library, or probably other ways!

ManuEpi, 2019-10-23:
Oh yes I didn't think about that , that some good ideas.
I will try them , thanks for helping

ManuEpi, 2019-10-24:
Okey I fixed it by using a boolean MutableLiveData.

When I'm in FragmentA onDestroy I set "isReleased" boolean to true
And in onResume of FragmentB I'm using .observe() on the boolean.
Inside the observe I check if the boolean == true . If it is , I just open my camera
And then I set my boolean to false because it's not longer released

Thanks