natario1/CameraView #660: Hi! - sponsoring CameraView development

natario1, 2019-10-23:
Hi! 👋

I am Mattia, the main developer and maintainer here, and I have enabled the GitHub Sponsor button on this repository which will let you support the library development by sponsoring my open source work through GitHub itself.

In the last years, I have spent countless unpaid hours on this and other projects, designing and developing tools that solve real problems, for everyone to use. I am very happy to know that companies of *all* sizes benefit from my work everyday.

GitHub is now offering you the opportunity to thank back with a sponsorship. For individuals, it could be a small amount. For companies that are powered by my tools, it could be a fairly small percentage of their revenue. In any case, your support will be greatly appreciated (and for bigger amounts, you get things like private support hours, or something else if you prefer).

https://github.com/sponsors/natario1

I will experiment leaving this as a pinned issue - for a while at least - since it's where people are most likely to see it. Sorry if this bothers you, and thank you all for your help.

HBiSoft, 2019-11-03:
I feel more people should be sponsoring. This is an amazing library with great support.

If a team would have to implement all the functions in this library, it would take them at least a month, if not more.

Thank you for the awesome library. I will be sponsoring again soon.

karamjeetzeroit, 2024-10-07:
I hope this message finds you well! I’d like to suggest a new feature for our video recording functionality: the ability to change the background while recording with the front camera. This could enhance user experience and allow for more creative video content.

Thank you for considering this idea!