natario1/CameraView #667: CameraException

FitApps7, 2019-10-30:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: 2.3.1
- Camera engine used: *camera1*
- Reproducible in official demo app: *not know*
- Device / Android version: Huawei Mate 20 lite (HWSNE), Android 9

com.otaliastudios.cameraview.CameraException: 
at com.otaliastudios.cameraview.engine.Camera1Engine.onError (Camera1Engine.java:102)
at android.hardware.Camera$EventHandler.handleMessage (Camera.java:1319)
at android.os.Handler.dispatchMessage (Handler.java:112)
at android.os.Looper.loop (Looper.java:216)
at android.os.HandlerThread.run (HandlerThread.java:65)
Caused by: java.lang.RuntimeException: 
at com.otaliastudios.cameraview.engine.Camera1Engine.onError (Camera1Engine.java:90)
at android.hardware.Camera$EventHandler.handleMessage (Camera.java:1319)
at android.os.Handler.dispatchMessage (Handler.java:112)
at android.os.Looper.loop (Looper.java:216)
at android.os.HandlerThread.run (HandlerThread.java:65)

HBiSoft, 2019-11-03:
I have the same issue. 

Unfortunately I'm not able to provide device specifications because the crash is not caught by Crashlytics and Google Console's log only provides me the following:

    com.otaliastudios.cameraview.CameraException:
    at com.otaliastudios.cameraview.engine.Camera1 Engine.onError(Camera1Engine.java:5)
    at android.hardware. Camera $EventHandler.handleMessage(Camera.java:1319)
    at android.os.Handler.dispatch Message(Handler.java: 112)
    at android.os.Looper.loop(Looperjava:216)
    at android.os.HandlerThread.run(HandlerThread.java:65)
    Caused by: java.lang.RuntimeException:
    at com.otaliastudios.cameraview.engine.Camera1 Engine.onError(Camera1Engine.java:4)

What I can say is, I'm using `CameraOptions` to determine what camera APIs are available on the device. If `CAMERA2` is available, I select it as default. I also give the user the option to pick between `CAMERA1` and `CAMERA2`. I do this for all the camera options including preview frame rate, white balance etc..

HBiSoft, 2019-11-26:
Here is the log a user sent me:

```text
com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Fail to connect to camera service
	at com.otaliastudios.cameraview.engine.Camera1Engine.g(Camera1Engine.java:13)
	at com.otaliastudios.cameraview.engine.CameraEngine$3.call(CameraEngine.java:3)
	at com.otaliastudios.cameraview.engine.CameraEngine$3.call(CameraEngine.java:1)
	at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:4)
	at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:1)
	at com.google.android.gms.tasks.zzf.run(Unknown Source)
	at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:2)
	at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute(WorkerHandler.java:1)
	at com.google.android.gms.tasks.zze.onComplete(Unknown Source)
	at com.google.android.gms.tasks.zzr.zza(Unknown Source)
	at com.google.android.gms.tasks.zzu.zze(Unknown Source)
	at com.google.android.gms.tasks.zzu.continueWithTask(Unknown Source)
	at com.otaliastudios.cameraview.engine.Step.a(Step.java:7)
	at com.otaliastudios.cameraview.engine.CameraEngine.startEngine(CameraEngine.java:2)
	at com.otaliastudios.cameraview.engine.CameraEngine.b(CameraEngine.java:1)
	at com.otaliastudios.cameraview.engine.CameraEngine$17$1.call(CameraEngine.java:2)
	at com.otaliastudios.cameraview.engine.CameraEngine$17$1.call(CameraEngine.java:1)
	at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:4)
	at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:1)
	at com.google.android.gms.tasks.zzf.run(Unknown Source)
	at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:2)
	at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute(WorkerHandler.java:1)
	at com.google.android.gms.tasks.zze.onComplete(Unknown Source)
	at com.google.android.gms.tasks.zzr.zza(Unknown Source)
	at com.google.android.gms.tasks.zzu.zze(Unknown Source)
	at com.google.android.gms.tasks.zzu.continueWithTask(Unknown Source)
	at com.otaliastudios.cameraview.engine.Step.a(Step.java:7)
	at com.otaliastudios.cameraview.engine.Step.a(Step.java:5)
	at com.otaliastudios.cameraview.engine.CameraEngine$17.run(CameraEngine.java:2)
	at android.os.Handler.handleCallback(Handler.java:733)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:157)
	at android.os.HandlerThread.run(HandlerThread.java:61)
Caused by: java.lang.RuntimeException: Fail to connect to camera service
	at android.hardware.Camera.native_setup(Native Method)
	at android.hardware.Camera.<init>(Camera.java:420)
	at android.hardware.Camera.open(Camera.java:372)
	at com.otaliastudios.cameraview.engine.Camera1Engine.g(Camera1Engine.java:1)
	... 32 more
```

HBiSoft, 2019-11-29:
This issue looks to be the same as #440 where it was closed because the logs were updated.

>Will improve error messages in #425 ... Without this there's not much we can do. If you debug yourself and find an issue, please reopen or open a new issue

---

**All of the crashes are on HUAWEI devices**, running Android 9.

I've had 97 crashes the past 7 days.

Here is a list of all the devices which experiences this issue:
- Honor 8X (HWJSN-H)
- HUAWEI Y9 2018 (HWFLA-H)
- HUAWEI P30 lite (HWMAR)
- P30 Pro (HWVOG)
- P20 Pro (HWCLT)
- Mate 9 (HWMHA)

thoaihuynhnguyen, 2019-11-30:
I used a external camera support by Android, but CameraException, Can you help me. Thansks,

closed 2020-01-05 by stale[bot]

HBiSoft, 2020-01-08:
I still see this issue in crashlytics. Bump

wslaimin, 2021-01-07:
I also see the issue