natario1/CameraView #673: Is there a way to use HEVC (H_265) video codec?

amitelman, 2019-11-07:
I think this library is efficient and very easy to use, but it seems that only H_264 \ H_263 is supported.

For large-sized Video files it is much better to use the HEVC video codec.. which result in files 4 times smaller. 

15 sec video in high resolution on the H_264 codec returns a 80 Mb file.
same video encoded with the HEVC returns a 20 Mb file.

So is there a way to use the HEVC codec?

Thanks in advance, and much appreciation for the hard work.

natario1, 2019-11-12:
It's not possible. You can see here that even in Android 10 devices are not required to have a HEVC encoder : https://developer.android.com/guide/topics/media/media-formats , only a decoder (from API 21 only).

amitelman, 2019-11-12:
I see, 
But what about devices that do support HEVC encoding ? there is a wide range of them. Of course all newly devices that comes out.
Can't it be an optional feature to effect supported devices? 
I know that the iOS for example, is encoding HEVC by default if the device is supporting the codec. 

I mean, there is no reason to decode in H264 if the device supports HEVC.

Thank you for the replay.

natario1, 2019-11-12:
We can add `VideoCodec.HEVC` option, feel free to open a PR. But that will fail on a huge number of devices.

We already have `VideoCodec.DEVICE_DEFAULT` so if the device manufacturer thinks they have a very good HEVC encoder, the video will be HEVC.

About "there is no reason to encode in H264 if the device supports HEVC", there can be many reasons, like the fact that the final video will not be readable by devices that do not have a HEVC decoder (like all Android before API 21).

amitelman, 2019-11-13:
Alright, Thank you very much for replaying..!

closed 2019-11-13 by amitelman