natario1/CameraView #674: MultiFilter and preview size change will cause preview disorder

candrwow, 2019-11-09:
Hi,I want add multiFilter in camera,when I use it and change camera layoutparam height,it will cause preview disorder.
[image]
[image]

Step:
1.camera is fill screen,use match_parent.
```
<com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera"
            android:layout_width="match_parent"
            android:layout_height="0dp"
            android:background="@color/appColorPrimaryDark"
            app:cameraExperimental="true"
            app:cameraFacing="back"
            app:cameraPictureMetering="true"
            app:cameraPictureSnapshotMetering="false"
            app:cameraPreview="glSurface"
            android:keepScreenOn="true"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintTop_toTopOf="parent" />
```
2.add multifilter
```
val hueFilter = HueFilter()
val brightnessFilter = BrightnessFilter()
brightnessFilter.brightness = 1.2f
camera.filter = MultiFilter(hueFilter, brightnessFilter)
```
3.change camera's height,use for 1:1 or 4:3
```
val lp = camera.layoutParams as ConstraintLayout.LayoutParams
            if (lp.dimensionRatio == null && !iv_square.isSelected) {
                lp.dimensionRatio = "1:1"
                lp.bottomToTop = R.id.layout_control
                lp.bottomToBottom = UNSET
                iv_square.isSelected = true
            } else {
                lp.dimensionRatio = null
                lp.bottomToBottom = PARENT_ID
                lp.bottomToTop = UNSET
                iv_square.isSelected = false
            }
```
This will happen on any emulator and my two phone,Nexus5 API 23 and S8+ 9.0

natario1, 2019-11-12:
Can you reproduce this on the demo app? It has controls to change the CameraView size.

closed 2019-12-09 by stale[bot]

grigM, 2020-07-23:
https://github.com/natario1/CameraView/issues/837#issuecomment-662995295