natario1/CameraView #676: render CameraView's texture into other gl context

troy-lamerton, 2019-11-09:
We have an opengl game with camera features such as showing your face inside the game.

To render the camera (in the game's opengl context) I can use `addFrameProcessor` to get a `byte[]` every frame. This means copying the bytes every frame. Surely, there is a better way.

### How do I?
Use camera view's gl texture in my game's gl context.

Is there an RGB texture of the camera preview?
Perhaps I can hack it by writing a filter that renders to my game's open gl context, instead of applying an effect.

### Version used
CameraView 2.4.0

troy-lamerton, 2019-11-12:
Using addFrameProcessor and copying the `byte[]` each frame does work. I load the nv21 bytes into unity and use a shader to render them as rgb.

The performance is good when camera is at 30fps. This is still using cpu though when an approach described in the issue would probably be a lot better.

Is there certain java files I can modify to expose the gl texture of camera view?

natario1, 2019-11-12:
CameraView renders onto its own surface which is a `CameraPreview` implementation. You should look into camera previews and either create your own (like `GlCameraPreview`, but with your own thread/context) or try to use `GlCameraPreview` itself and render your game into it.

To apply the new preview you should subclass `CameraView` and override `instantiateCameraPreview()`.

closed 2019-12-09 by stale[bot]