natario1/CameraView #677: Flash.ON don`t works in video mode

alxgord, 2019-11-12:
### Describe the bug
Flash.ON don`t works in video mode(I using takeVide() method, not snapshot), in photo mode works as expected
- CameraView version: 2.4.0
- Camera engine used: both
- Device / Android version: *Essential Phone, API 29*

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
<com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="0dp"
        android:layout_height="0dp"
        android:keepScreenOn="true"
        android:soundEffectsEnabled="false"
        app:cameraAudio="on"
        app:cameraFlash="on"
        app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
        app:cameraEngine="camera2"
        app:cameraFacing="back"
        app:cameraGesturePinch="zoom"
        app:cameraGestureScrollHorizontal="zoom"
        app:cameraGestureTap="autoFocus"
        app:cameraGrid="off"
        app:cameraMode="picture"
        app:cameraPlaySounds="false"
        app:cameraPreview="glSurface"
        app:cameraUseDeviceOrientation="true"
        app:layout_constraintBottom_toBottomOf="parent"
        app:layout_constraintEnd_toEndOf="parent"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintTop_toTopOf="parent" />
```

natario1, 2019-11-12:
If you want the flash always on during a video you should use `TORCH`.

alxgord, 2019-11-12:
Okay. Can you please explain how works Flash.ON mode when video recording?

closed 2019-12-09 by stale[bot]