natario1/CameraView #680: java.lang.RuntimeException: onConfigureFailed! Session

ashishnegi962, 2019-11-14:
java.lang.RuntimeException: onConfigureFailed! Session android.hardware.camera2.impl.CameraCaptureSessionImpl@eb02b3a
        at com.otaliastudios.cameraview.engine.Camera2Engine$4.onConfigureFailed(Camera2Engine.java:536)
        at android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy.lambda$onConfigureFailed$1(CallbackProxies.java:64)
        at android.hardware.camera2.impl.-$$Lambda$CallbackProxies$SessionStateCallbackProxy$gvbTsp9UPpKJAbdycdci_ZW5BeI.run(Unknown Source:4)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:201)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2019-11-14 10:45:57.244 29621-29621/northeast.ecological.wii.ecological_north E/ACRA: ACRA caught a RuntimeException for northeast.ecological.wii.ecological_north
    java.lang.RuntimeException: onConfigureFailed! Session android.hardware.camera2.impl.CameraCaptureSessionImpl@eb02b3a
        at com.otaliastudios.cameraview.engine.Camera2Engine$4.onConfigureFailed(Camera2Engine.java:536)
        at android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy.lambda$onConfigureFailed$1(CallbackProxies.java:64)
        at android.hardware.camera2.impl.-$$Lambda$CallbackProxies$SessionStateCallbackProxy$gvbTsp9UPpKJAbdycdci_ZW5BeI.run(Unknown Source:4)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:201)
        at android.os.HandlerThread.run(HandlerThread.java:65)

ashishnegi962, 2019-11-22:
sir please kindly provide suggestion to solve the problem. Camera2Engine on Configure Failed

bayunvnt, 2019-11-22:
I have this problem too

huaxiaos, 2019-12-01:
I have this problem too Redmi note 8

soulrelay, 2019-12-01:
I have this problem too Redmi note 8

deckyfx, 2019-12-04:
Also got this error on Samsung A10

closed 2019-12-31 by stale[bot]

ermanishks, 2020-01-06:
i have this problem too on LG Q6 android version 8.1.0

dilrajsingh1997, 2020-01-06:
I have the same issue on Redmi note 7

com.otaliastudios.cameraview.engine.Camera2Engine$4.onConfigureFailed + 581 (Camera2Engine.java:581)

CodeIgniter00, 2020-01-20:
I have the same issue on HUAWEI Y7

gyqGitHub, 2020-04-03:
I have the same issue on Redmi note 7

gyqGitHub, 2020-04-03:
1 com.otaliastudios.cameraview.engine.Camera2Engine$4.onConfigureFailed(Camera2Engine.java:2)
--
2 java.lang.reflect.Method.invoke(Native Method)
3 android.hardware.camera2.dispatch.InvokeDispatcher.dispatch(InvokeDispatcher.java:39)
4 android.hardware.camera2.dispatch.HandlerDispatcher$1.run(HandlerDispatcher.java:65)
5 android.os.Handler.handleCallback(Handler.java:761)
6 android.os.Handler.dispatchMessage(Handler.java:98)
7 android.os.Looper.loop(Looper.java:156)
8 android.os.HandlerThread.run(HandlerThread.java:61)

nxnyk, 2020-04-08:
I have the same issue on Redmi 8A

moujunqiang, 2020-08-26:
Readmi8上面碰到这个问题

lamlt95, 2020-12-18:
I have the same issue on Redmi Redmi 7A
Does anyone have an answer to this issue yet

farzadhallaji, 2021-01-06:
I have the same issue

ashishnegi962, 2021-02-15:
Redmi Redmi 7A hardware issus in camera.

Ahsanwarsi, 2021-03-18:
Hi @natario1 
Same issue on **Pixel 2 XL** & **Galaxy S20 5G**.

`Fatal Exception: java.lang.RuntimeException: onConfigureFailed! Session android.hardware.camera2.impl.CameraCaptureSessionImpl@7918db5
       at com.otaliastudios.cameraview.engine.Camera2Engine$4.onConfigureFailed(Camera2Engine.java:584)
       at android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy.lambda$onConfigureFailed$1$CallbackProxies$SessionStateCallbackProxy(CallbackProxies.java:64)
       at android.hardware.camera2.impl.-$$Lambda$CallbackProxies$SessionStateCallbackProxy$gvbTsp9UPpKJAbdycdci_ZW5BeI.run(:4)
       at android.os.Handler.handleCallback(Handler.java:938)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loop(Looper.java:246)
       at android.os.HandlerThread.run(HandlerThread.java:67)`

ashishnegi962, 2021-03-20:
camera 2 working in hard ware configration