natario1/CameraView #682: Camera engine not starting on Huawei P20 Pro

SarunasDaubaris, 2019-11-14:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.4.0*
- Camera engine used: *camera1*
- Reproducible in official demo app: *no*
- Device / Android version: *Huawei P20 Pro, 8.1.0*

I'm using CameraView with Flutter's Platform View. It is instantiated programmatically. It works as expected on Samsung A50 Android 9.

The bug occurs on Huawei P20 Pro. I've run my app with debugger - Camera1Engine is used by the library. It seems that Camera1Engine methods `onStartEngine()`, `onStartBind()`, `onStartPreview()` are not even being called.

### To Reproduce
Steps to reproduce the behavior:
1. Create CameraView instance
``` 
cameraView.layoutParams = FrameLayout.LayoutParams(FrameLayout.LayoutParams.MATCH_PARENT, FrameLayout.LayoutParams.MATCH_PARENT)
cameraView.keepScreenOn = true
```

2. Once Flutter AndroidView is initialised, open the camera
```
cameraView.open()
```

### Expected behaviour
Camera preview is displayed

### Current behaviour
Black background is visible and camera is not previewed

nxnyk, 2019-11-21:
the same problem

HBiSoft, 2019-12-03:
@SarunasDaubaris 
I've had all sorts of problems on Huawei devices (Huawei P20 Pro being one of them). Have a look at my comment [here](https://github.com/natario1/CameraView/issues/667#issuecomment-559843209).

closed 2019-12-30 by stale[bot]

cleancontrol, 2020-01-22:
I got this exception : on huawei tablet 
`
2020-01-22 21:30:30.779 22037-23316/com.naqicontrol.clean W/MessageQueue: Handler (android.os.Handler) {37d621b} sending message to a Handler on a dead thread
    java.lang.IllegalStateException: Handler (android.os.Handler) {37d621b} sending message to a Handler on a dead thread
        at android.os.MessageQueue.enqueueMessage(MessageQueue.java:603)
        at android.os.Handler.enqueueMessage(Handler.java:653)
        at android.os.Handler.sendMessageAtTime(Handler.java:622)
        at android.os.Handler.sendMessageDelayed(Handler.java:592)
        at android.os.Handler.post(Handler.java:345)
        at android.hardware.camera2.impl.CameraDeviceImpl.close(CameraDeviceImpl.java:991)
        at com.otaliastudios.cameraview.k.b.Q(SourceFile:740)
        at com.otaliastudios.cameraview.k.d$i.call(SourceFile:414)
        at com.otaliastudios.cameraview.k.d$i.call(SourceFile:411)
        at com.otaliastudios.cameraview.k.l.c$b.call(SourceFile:69)
        at com.otaliastudios.cameraview.k.l.c$b.call(SourceFile:60)
        at com.otaliastudios.cameraview.k.l.a$b.onComplete(SourceFile:100)
        at com.google.android.gms.tasks.zzj.run(SourceFile:4)
        at com.otaliastudios.cameraview.p.f.g.c(SourceFile:137)
        at com.otaliastudios.cameraview.p.f.g$b.execute(SourceFile:112)
        at com.google.android.gms.tasks.zzi.onComplete(SourceFile:10)
        at com.google.android.gms.tasks.zzr.zza(SourceFile:19)
        at com.google.android.gms.tasks.zzu.trySetResult(SourceFile:94)
        at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(SourceFile:10)
        at com.otaliastudios.cameraview.k.l.a$b$a.onComplete(SourceFile:116)
        at com.otaliastudios.cameraview.k.l.a$d.run(SourceFile:195)
        at com.otaliastudios.cameraview.p.f.g.c(SourceFile:137)
        at com.otaliastudios.cameraview.k.l.a.b(SourceFile:192)
        at com.otaliastudios.cameraview.k.l.a.a(SourceFile:32)
        at com.otaliastudios.cameraview.k.l.a$b.onComplete(SourceFile:101)
        at com.google.android.gms.tasks.zzj.run(SourceFile:4)
        at com.otaliastudios.cameraview.p.f.g.c(SourceFile:137)
        at com.otaliastudios.cameraview.p.f.g$b.execute(SourceFile:112)
        at com.google.android.gms.tasks.zzi.onComplete(SourceFile:10)
        at com.google.android.gms.tasks.zzr.zza(SourceFile:19)
        at com.google.android.gms.tasks.zzu.trySetResult(SourceFile:94)
        at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(SourceFile:10)
        at com.otaliastudios.cameraview.k.l.a$b$a.onComplete(SourceFile:116)
        at com.otaliastudios.cameraview.k.l.a$d.run(SourceFile:195)
        at com.otaliastudios.cameraview.p.f.g.c(SourceFile:137)
        at com.otaliastudios.cameraview.k.l.a.b(SourceFile:192)
        at com.otaliastudios.cameraview.k.l.a.a(SourceFile:32)
        at com.otaliastudios.cameraview.k.l.a$b.onComplete(SourceFile:101)
        at com.otaliastudios.cameraview.k.l.a$d.run(SourceFile:195)
        at android.os.Handler.handleCallback(Handler.java:761)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:156)
        at android.os.HandlerThread.run(HandlerThread.java:61)
2020-01-22 21:30:30.779 22037-23316/com.naqicontrol.clean I/d: onStopEngine: Clean up. Released camera.
2020-01-22 21:30:30.779 22037-23316/com.naqicontrol.clean I/d: onStopEngine: Aborting actions.
2020-01-22 21:30:30.780 22037-23316/com.naqicontrol.clean W/d: onStopEngine: Returning.
2020-01-22 21:30:30.780 22037-23316/com.naqicontrol.clean I/a: ENGINE << OFF - Finished.
2020-01-22 21:30:30.781 22037-22037/com.naqicontrol.clean E/d: EXCEPTION: Got CameraException. Dispatching to callback.`

cleancontrol, 2020-01-22:
please help me.

HBiSoft, 2020-01-22:
@cleancontrol please add your log in my question mentioned above(758). This question is closed and will most likely not be seen.