natario1/CameraView #687: How to capture frame by frame images from Android video recording in real time ?

gabriel-TheCode, 2019-11-21:
### How do I?
I tried with the method below to get the frames of a video stream in real time but It doesnt work

```java
        cameraView.addFrameProcessor(new FrameProcessor() {
            private long lastTime = System.currentTimeMillis();

            @Override
            public void process(@NonNull Frame frame) {
                long newTime = frame.getTime();
                long delay = newTime - lastTime;
                lastTime = newTime;
                Log.d("Frame delayMillis:", "delay "+ "FPS: "+ "1000 / "+ delay);
                if (DECODE_BITMAP) {
                    YuvImage yuvImage = new YuvImage(frame.getData(), ImageFormat.NV21,
                            frame.getSize().getWidth(),
                            frame.getSize().getHeight(),
                            null);
                    ByteArrayOutputStream jpegStream = new ByteArrayOutputStream();
                    yuvImage.compressToJpeg(new Rect(0, 0,
                            frame.getSize().getWidth(),
                            frame.getSize().getHeight()), 100, jpegStream);
                    byte[] jpegByteArray = jpegStream.toByteArray();
                    Bitmap bitmap = BitmapFactory.decodeByteArray(jpegByteArray, 0, jpegByteArray.length);
                    //noinspection ResultOfMethodCallIgnored
                    Log.d("FRAME >>>>", bitmap.toString());
                }
            }
        });
```

### Version used
2.0.0-beta04

closed 2019-12-18 by stale[bot]

abhishektiwari-kiwi, 2020-07-08:
Any update on this? We need solution. Help us out.
@gabriel-TheCode how did you manage to achieve this?

gabriel-TheCode, 2020-07-08:
Sorry @abhishektiwari-kiwi , I finally opted for another approach, I used an RTMP client that was more judicious for my problem and my requirements.

abhishektiwari-kiwi, 2020-07-08:
I want to achieve video recording along with frame extraction with the help of this library.
Guess I need to change the approach.
@gabriel-TheCode By the way, Thank you for the idea.