natario1/CameraView #713: Crash with versions newer than 2.3.1

brnogz, 2019-12-19:
### Describe the bug
In my app, some dialogs shown over cameraview. To prevent permission requests blocks these dialogs, permissions are asked manually after dialogs dismissed. When user grant permissions, setLifecycle method called. Cameraview causes crash after permissions granted in first launch. When app relaunched, everything work normal. Both test device does not supports camera2 api(I guess), even if camera2 is selected, crash coming from camera1 engine.

- CameraView version: 2.4.0, 2.5.0
- Camera engine used: camera1
- Reproducible in official demo app: no
- Device / Android version: Mi A2 (Android 9), Redmi Note 5 (Androıd 9)

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
        <com.otaliastudios.cameraview.CameraView
            android:id="@+id/cvMain"
            android:layout_width="0dp"
            android:layout_height="0dp"
            android:keepScreenOn="true"
            app:cameraAudio="off"
            app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
            app:cameraGesturePinch="zoom"
            app:cameraGestureTap="autoFocus"
            app:cameraMode="picture"
            app:cameraPlaySounds="false" />
```

### Logs
````
2019-12-19 12:26:41.873 7943-7943/com.translately.android E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.translately.android, PID: 7943
    java.lang.RuntimeException: Unknown CameraPreview output class.
        at com.otaliastudios.cameraview.engine.Camera1Engine.onStartBind(Camera1Engine.java:185)
        at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:467)
        at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:463)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
        at com.google.android.gms.tasks.zzj.run(Unknown Source:4)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler$2.execute(WorkerHandler.java:112)
        at com.google.android.gms.tasks.zzi.onComplete(Unknown Source:10)
        at com.google.android.gms.tasks.zzr.zza(Unknown Source:19)
        at com.google.android.gms.tasks.zzu.trySetResult(Unknown Source:94)
        at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(Unknown Source:10)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
```

vladislav1nikov, 2019-12-20:
1. Create variable isSetLifecycleOwner = false
2. After calling setLifecycleOwner set the isSetLifecycleOwner to true
3. In onPause add code:
if (!isSetLifecycleOwner) {
    camera.close();
}

sidneywang, 2019-12-20:
same crash with cameraview 2.4.0 on device redmi 6.

brnogz, 2019-12-20:
> 2\. setLifecycleOwner

It works for now, I will do some tests

brnogz, 2019-12-20:
> 1. Create variable isSetLifecycleOwner = false
> 2. After calling setLifecycleOwner set the isSetLifecycleOwner to true
> 3. In onPause add code:
>    if (!isSetLifecycleOwner) {
>    camera.close();
>    }
> 
> Do not call setLifecycleOwner more than once

It works in a quick test, I will do some tests before closing this issue

natario1, 2019-12-21:
I will add `app:cameraRequestPermissions="true|false"` option in new release, so you can set the lifecycle owner without worrying about automatic permission requests.

natario1, 2019-12-28:
This was merged in #718

closed 2019-12-28 by natario1