natario1/CameraView #714: Camera2 + Flashlight.ON or AUTO

comradewalker, 2019-12-19:
### Describe the bug

- CameraView version: worked fine till v2.20
- Camera engine used: camera2
- Reproducible in official demo app: yes
- Device / Android version:  Samsung Galaxy S10+, API 28

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Go to camera2 engine and set Flashlight ON (or AUTO)
2. Take a picture.

### Expected behavior
Everything was fine, but since v2.3.0 after you click on capturing photo button nothing happens (preview is still on, I can hear shutter sound, but photo is never taken). If I leave app, for some time I can't even open preview on device stock camera, then it resets. I believe there's some deadlock. 

Camera1 engine works fine.

natario1, 2019-12-28:
I believe this is fixed in latest versions.

closed 2019-12-28 by natario1

comradewalker, 2020-01-08:
Sorry for a late reply. I've tried with latest version (v2.6.0) and the issue is still here. Camera 2 api doesn't work with flashlight ON.