natario1/CameraView #721: when audio is enabled (Audio.ON) the RECORD_AUDIO permission should be added to the app manifest file.

alexei-28, 2019-12-25:
Android Studio 3.6

I want to use this lib only for capture pictures/images.

app/build.gradle:

```
dependencies {
    implementation fileTree(dir: 'libs', include: ['*.jar'])
    implementation 'com.google.android.material:material:1.2.0-alpha03'
    implementation 'com.google.android.gms:play-services-maps:17.0.0'
    implementation "com.otaliastudios:cameraview:2.5.0"
    implementation('com.crashlytics.sdk.android:crashlytics:2.10.1@aar') { transitive = true; }
    implementation 'com.google.firebase:firebase-ml-vision:24.0.1'
    implementation "org.jetbrains.kotlin:kotlin-stdlib-jdk7:$kotlin_version"
    implementation 'com.synnapps:carouselview:0.1.4'
    implementation 'androidx.appcompat:appcompat:1.1.0'
```

In my activity:

class ScanCheckActivity : RootActivity() {
    private lateinit var dataBinding: ScanCheckActivityBinding

    companion object {
        val TAG = ScanCheckActivity::class.java.name
    }

    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        dataBinding = DataBindingUtil.setContentView(this, R.layout.scan_check_activity)
        init()
    }

    private fun init() {
        setSupportActionBar(findViewById(R.id.toolBar))
        getSupportActionBar()?.setDisplayHomeAsUpEnabled(true)
        getSupportActionBar()?.setHomeButtonEnabled(true)
        getSupportActionBar()?.setDisplayShowTitleEnabled(false)

        dataBinding.cameraView.setLifecycleOwner(this);
     
    }
override fun onResume() {
        super.onResume()
        dataBinding.cameraView.open();
    }

In xml:

```
<?xml version="1.0" encoding="utf-8"?>
<layout xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:app="http://schemas.android.com/apk/res-auto"
    xmlns:tools="http://schemas.android.com/tools">

    <data>

    </data>

    <androidx.constraintlayout.widget.ConstraintLayout
        android:layout_width="match_parent"
        android:layout_height="match_parent">

        <include
            android:id="@+id/profileDetailsToolbar"
            layout="@layout/tool_bar"
            app:layout_constraintEnd_toEndOf="parent"
            app:layout_constraintStart_toStartOf="parent"
            app:layout_constraintTop_toTopOf="parent"
            app:title="@{@string/scan_check}" />

        <com.otaliastudios.cameraview.CameraView
            android:id="@+id/cameraView"
            android:layout_width="0dp"
            android:layout_height="0dp"
            android:keepScreenOn="true"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintEnd_toEndOf="parent"
            app:layout_constraintStart_toStartOf="parent"
            app:layout_constraintTop_toBottomOf="@+id/profileDetailsToolbar" />

        <com.google.android.material.button.MaterialButton
            android:id="@+id/detectButton"
            android:layout_width="0dp"
            android:layout_height="@dimen/button_height"
            android:layout_margin="@dimen/default_margin"
            android:text="@string/get"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintEnd_toEndOf="parent"
            app:layout_constraintStart_toStartOf="parent" />

    </androidx.constraintlayout.widget.ConstraintLayout>

</layout>
```

But when I start app I get error in this line:

```
override fun onResume() {
        super.onResume()
        dataBinding.cameraView.open();
    }
```

error:

```
12-25 11:49:55.302 E/AndroidRuntime(26323): java.lang.RuntimeException: Unable to resume activity {com.myproject.debug/com.myproject.ui.activity.ScanCheckActivity}: java.lang.IllegalStateException: Permission error: when audio is enabled (Audio.ON) the RECORD_AUDIO permission should be added to the app manifest file.
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3103)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3134)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2481)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at android.app.ActivityThread.-wrap11(ActivityThread.java)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1344)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at android.os.Handler.dispatchMessage(Handler.java:102)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at android.os.Looper.loop(Looper.java:148)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at android.app.ActivityThread.main(ActivityThread.java:5417)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at java.lang.reflect.Method.invoke(Native Method)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
12-25 11:49:55.302 E/AndroidRuntime(26323): Caused by: java.lang.IllegalStateException: Permission error: when audio is enabled (Audio.ON) the RECORD_AUDIO permission should be added to the app manifest file.
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at com.otaliastudios.cameraview.CameraView.checkPermissionsManifestOrThrow(CameraView.java:796)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at com.otaliastudios.cameraview.CameraView.checkPermissions(CameraView.java:759)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at com.otaliastudios.cameraview.CameraView.open(CameraView.java:742)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at com.myproject.ui.activity.ScanCheckActivity.onResume(ScanCheckActivity.kt:252)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1258)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at android.app.Activity.performResume(Activity.java:6327)
12-25 11:49:55.302 E/AndroidRuntime(26323): 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3092)
```

I'm not need  **RECORD_AUDIO** .  I need only capture images.

alexei-28, 2019-12-25:
I found solution 👍 
`app:cameraAudio="off"`

natario1, 2019-12-28:
Correct! set audio to off

closed 2019-12-28 by natario1

bibin2415, 2020-12-13:
I added app:cameraAudio="off" still app is crashing. CameraView: Permission error: when audio is enabled (Audio.ON) the RECORD_AUDIO permission should be added to the app manifest file.

Manifest  file:
 <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-feature android:name="android.hardware.camera" />

XML file:
 <com.otaliastudios.cameraview.CameraView
                        android:id="@+id/cameraView"
                        android:layout_width="match_parent"
                        android:layout_height="match_parent"
                        android:keepScreenOn="true"
                        app:cameraAudio="off" />

please let me know the reason.

bibin2415, 2020-12-13:
cameraView.setAudio(Audio.OFF); 

worked!😊

Denesh11, 2020-12-16:
Hi bibin2415,
where could I add that code.

Denesh11, 2020-12-16:
Thanks  bibin . Its helps for me