natario1/CameraView #728: ML Kit: Can't get bitmap by FrameProcessor: MCT_EVENT_MODULE_BUF_DIVERT

alexei-28, 2019-12-31:
Android Studio 3.6

I need to recognize text on image by Firebase ML Kit.

So to do this I use FrameProcessor like this:

In my activity:

```
import com.otaliastudios.cameraview.CameraUtils
import com.otaliastudios.cameraview.frame.Frame
import com.otaliastudios.cameraview.frame.FrameProcessor
import com.google.firebase.ml.vision.FirebaseVision
import com.google.firebase.ml.vision.common.FirebaseVisionImage
import com.google.firebase.ml.vision.text.FirebaseVisionText
 dataBinding.cameraView.addFrameProcessor(object : FrameProcessor {
            override fun process(frame: Frame) {
                Debug.d(TAG, "addFrameProcessor: process:")
                var bitmap: Bitmap? = null
                if (frame.dataClass == ByteArray::class.java) {
                    Debug.d(TAG, "addFrameProcessor: process_dataClass: ByteArray")
                    val data: ByteArray = frame.getData()
                    bitmap = AndroidUtil.byteArray2Bitmap(data)
                } else if (frame.dataClass == Image::class.java) {
                    Debug.d(TAG, "addFrameProcessor: process_dataClass: Image")
                    val image: Image = frame.getData()
                    bitmap = AndroidUtil.image2Bitmap(image)
                }
                Debug.d(TAG, "addFrameProcessor: process: bitmap_result = $bitmap")
                if (bitmap != null) {
                    runDetector(bitmap)
                   
                }
            }
        })
```
I bring the phone camera to the photo but  result `bitmap_result  `is always **null**.

Here logcat:
```
12-31 16:03:09.690 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process_dataClass: ByteArray
12-31 16:03:09.690 D/skia    (22214): --- SkImageDecoder::Factory returned null
12-31 16:03:09.690 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process: bitmap_result = null
12-31 16:03:09.721 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process:
12-31 16:03:09.721 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process_dataClass: ByteArray
12-31 16:03:09.721 D/skia    (22214): --- SkImageDecoder::Factory returned null
12-31 16:03:09.722 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process: bitmap_result = null
12-31 16:03:09.725 E/mm-camera(  217): module_faceproc_port_event_func:886] MCT_EVENT_MODULE_BUF_DIVERT 65538, Cannot start FD, active 10003, frameid 289 0, native 1, mapped 1 1
12-31 16:03:09.764 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process:
12-31 16:03:09.764 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process_dataClass: ByteArray
12-31 16:03:09.764 D/skia    (22214): --- SkImageDecoder::Factory returned null
12-31 16:03:09.764 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process: bitmap_result = null
12-31 16:03:09.767 E/mm-camera(  217): module_faceproc_port_event_func:886] MCT_EVENT_MODULE_BUF_DIVERT 65538, Cannot start FD, active 10003, frameid 290 0, native 1, mapped 1 1
12-31 16:03:09.797 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process:
12-31 16:03:09.797 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process_dataClass: ByteArray
12-31 16:03:09.797 D/skia    (22214): --- SkImageDecoder::Factory returned null
12-31 16:03:09.798 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process: bitmap_result = null
12-31 16:03:09.809 E/mm-camera(  217): module_faceproc_port_event_func:886] MCT_EVENT_MODULE_BUF_DIVERT 65538, Cannot start FD, active 10003, frameid 291 0, native 1, mapped 1 1
12-31 16:03:09.840 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process:
12-31 16:03:09.840 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process_dataClass: ByteArray
12-31 16:03:09.840 D/skia    (22214): --- SkImageDecoder::Factory returned null
12-31 16:03:09.840 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process: bitmap_result = null
12-31 16:03:09.851 E/mm-camera(  217): module_faceproc_port_event_func:886] MCT_EVENT_MODULE_BUF_DIVERT 65538, Cannot start FD, active 10003, frameid 292 0, native 1, mapped 1 1
12-31 16:03:09.880 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process:
12-31 16:03:09.880 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process_dataClass: ByteArray
12-31 16:03:09.880 D/skia    (22214): --- SkImageDecoder::Factory returned null
12-31 16:03:09.880 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process: bitmap_result = null
12-31 16:03:09.886 D/audio_hw_primary(  198): out_set_parameters: enter: usecase(1: low-latency-playback) kvpairs: routing=2
12-31 16:03:09.893 E/mm-camera(  217): module_faceproc_port_event_func:886] MCT_EVENT_MODULE_BUF_DIVERT 65538, Cannot start FD, active 10003, frameid 293 0, native 1, mapped 1 1
12-31 16:03:09.896 D/audio_hw_primary(  198): select_devices: out_snd_device(2: speaker) in_snd_device(0: none)
12-31 16:03:09.896 D/msm8974_platform(  198): platform_send_audio_calibration: sending audio calibration for snd_device(2) acdb_id(15)
12-31 16:03:09.896 D/audio_hw_primary(  198): enable_snd_device: snd_device(2: speaker)
12-31 16:03:09.900 D/audio_hw_primary(  198): enable_audio_route: apply and update mixer path: low-latency-playback
12-31 16:03:09.908 I/ActivityManager(  778): START u0 {act=android.intent.action.MAIN cat=[android.intent.category.HOME] flg=0x10200000 cmp=com.google.android.googlequicksearchbox/com.google.android.launcher.GEL (has extras)} from uid 1000 on display 0
12-31 16:03:09.932 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process:
12-31 16:03:09.932 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process_dataClass: ByteArray
12-31 16:03:09.932 D/skia    (22214): --- SkImageDecoder::Factory returned null
12-31 16:03:09.932 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process: bitmap_result = null
12-31 16:03:09.934 E/mm-camera(  217): module_faceproc_port_event_func:886] MCT_EVENT_MODULE_BUF_DIVERT 65538, Cannot start FD, active 10003, frameid 294 0, native 1, mapped 1 1
12-31 16:03:09.968 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process:
12-31 16:03:09.968 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process_dataClass: ByteArray
12-31 16:03:09.968 D/skia    (22214): --- SkImageDecoder::Factory returned null
12-31 16:03:09.968 D/com.myproject.activity.ScanCheckActivity(22214): addFrameProcessor: process: bitmap_result = null
```
As you can see the `bitmap_result ` is **null** and always show this error:

`E/mm-camera(  217): module_faceproc_port_event_func:886] MCT_EVENT_MODULE_BUF_DIVERT 65538, Cannot start FD, active 10003, frameid 293 0, native 1, mapped 1 1`

alexei-28, 2019-12-31:
I found solution:

 ```
private fun initLogic() {
        dataBinding.cameraView.addFrameProcessor(object : FrameProcessor {
            override fun process(frame: Frame) {
                Debug.d(TAG, "addFrameProcessor: process: frame = $frame")
                var bitmap: Bitmap? = null
                if (frame.dataClass == ByteArray::class.java) {
                    Debug.d(TAG, "addFrameProcessor: process_dataClass: ByteArray")
                    val out = ByteArrayOutputStream()
                    val yuvImage = YuvImage(
                        frame.getData(),
                        ImageFormat.NV21,
                        frame.getSize().getWidth(),
                        frame.getSize().getHeight(),
                        null
                    )
                    yuvImage.compressToJpeg(
                        Rect(0, 0, frame.getSize().getWidth(), frame.getSize().getHeight()),
                        90,
                        out
                    )
                    val imageBytes = out.toByteArray()
                    bitmap = BitmapFactory.decodeByteArray(imageBytes, 0, imageBytes.size)
                } else if (frame.dataClass == Image::class.java) {
                    Debug.d(TAG, "addFrameProcessor: process_dataClass: Image")
                    val image: Image = frame.getData()
                    bitmap = AndroidUtil.image2Bitmap(image)
                }
                Debug.d(TAG, "addFrameProcessor: process: bitmap_result = $bitmap")
            }
        })
```

and now `bitmap_result  `is **not null**

closed 2020-01-27 by stale[bot]

abhishektiwari-kiwi, 2020-07-21:
Why for camera1 bitmap converted from byteArray is corrupted?
ML model couldn't recognize.
Please help me out.

[image]