natario1/CameraView #731: Camera is restarted on orientation flip

sewar, 2020-01-02:
### Describe the bug
Camera is restarted on orientation flip of 180 degrees.

- CameraView version: *2.6.0*
- Camera engine used: *camera1(default)*
- Reproducible in official demo app: *no*
- Device / Android version: *Pixel 3a, API 29*

### To Reproduce
1. In an Activity with `android:configChanges="keyboard|keyboardHidden|orientation|screenLayout|uiMode|screenSize|smallestScreenSize"`and `android:screenOrientation="sensorLandscape"`.
2. Flip device while keeping orientation (e.g. landscape -> reverseLandscape)
3. Camera is restarted, preview goes black for a few seconds.

### Expected behavior
Camera should not be restarted, preview should not have any effect.

### Logs
```
CameraCallbacks  I  onDisplayOffsetChanged 90 recreate: false
CameraCallbacks  W  onDisplayOffsetChanged restarting the camera.
…
CameraCallbacks  I  onDisplayOffsetChanged 270 recreate: false
CameraCallbacks  W  onDisplayOffsetChanged restarting the camera.
```

### Bug
The bug seems to be 
`if (isOpened() && !willRecreate)`
should be
`if (isOpened() && willRecreate)`
here:
https://github.com/natario1/CameraView/blob/9c984f0c216bf722c00c7e5f56838469029b09aa/cameraview/src/main/java/com/otaliastudios/cameraview/CameraView.java#L2203-L2214

natario1, 2020-01-02:
Thanks for actionable report. Can you use `landscape` instead of `sensorLandscape` ?

sewar, 2020-01-02:
> Thanks for actionable report. Can you use `landscape` instead of `sensorLandscape` ?

I tried that and the screen doesn't rotate anymore, I will have to rotate the view manually, is that the recommended way?

natario1, 2020-01-02:
If you need to let the whole UI rotate (as with `sensorLandscape`), unfortunately we need to restart. If you git blame those lines, you can easily find the commit, PR and GitHub issue that they fix.

If you want to fix the UI in landscape position (as with `landscape`), you can use `CameraListener.onOrientationChanged()` to manually rotate UI elements.

sewar, 2020-01-03:
@natario1 Thanks for the details.

closed 2020-01-03 by sewar