natario1/CameraView #736: Camera2 + Flashlight.ON or AUTO

comradewalker, 2020-01-08:
I can't reopen issue ( #714 ) so here a new one:

Describe the bug
CameraView version: worked fine till v2.2.0 (latest working version)
Camera engine used: camera2
Reproducible in official demo app: yes
Device / Android version: Samsung Galaxy S10+, API 28, 29
To Reproduce
Steps to reproduce the behavior, possibly in the demo app:

Go to camera2 engine and set Flashlight ON (or AUTO)
Take a picture.
Expected behavior
Everything was fine, but since v2.3.0 after you click on capturing photo button nothing happens (preview is still on, I can hear shutter sound, flashlight flashes, but photo is never taken->stale). If I leave app, for some time I can't even start preview on device stock camera, then it resets. I believe there's some deadlock.

Camera1 engine works fine.

natario1, 2020-01-10:
Can you fill the bug information? Including logs from the demo app.

comradewalker, 2020-01-11:
Here are logs ([camera.log](https://github.com/natario1/CameraView/files/4049101/camera.log)) and video ([camera-video.zip](https://github.com/natario1/CameraView/files/4049104/camera-video.zip)) of issue. 

Video starts with camera2 api, I tried to take a shot with flashlight ON, shutter sound is delayed, preview is still on, nothing happens on subsequent camera shots. Changing to camera1 api everything is fine, returning to camera2 api same thing happens. I switched once more to camera1 and once I returned to camera2 api, I could take 3 camera shots but even though flashlight was ON you can see in video that flashlight didn't flash.

Here is xml layout:

`
    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:layout_gravity="center"
        android:layout_marginBottom="88dp"
        android:keepScreenOn="true"
        app:cameraExperimental="true"
        app:cameraEngine="camera2"
        app:cameraPreview="glSurface"
        app:cameraPlaySounds="true"
        app:cameraGrid="off"
        app:cameraFlash="on"
        app:cameraAudio="on"
        app:cameraFacing="back"
        app:cameraGestureTap="autoFocus"
        app:cameraGestureLongTap="none"
        app:cameraGesturePinch="zoom"
        app:cameraGestureScrollHorizontal="none"
        app:cameraGestureScrollVertical="exposureCorrection"
        app:cameraMode="picture"
        app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker">

        
        <ImageView
            android:id="@+id/watermark"
            android:layout_width="72dp"
            android:layout_height="72dp"
            android:layout_gravity="bottom|end"
            android:src="@mipmap/logo_foreground"
            app:layout_drawOnPictureSnapshot="true"
            app:layout_drawOnPreview="true"
            app:layout_drawOnVideoSnapshot="true" />

    </com.otaliastudios.cameraview.CameraView>`

comradewalker, 2020-01-11:
Sorry man, log in previous post was incomplete, here you are the new ones:

AS logcat: 
[camera.log](https://github.com/natario1/CameraView/files/4049189/camera.log)

Log from CameraLogger.registerLogger:
[camera2.log](https://github.com/natario1/CameraView/files/4049190/camera2.log)

Video of issue:
[camera-video2.zip](https://github.com/natario1/CameraView/files/4049191/camera-video2.zip)

natario1, 2020-01-11:
I think we don't have enough logs to see what's wrong. Can you add log lines in Full2PictureRecorder line 59 to check that onStart is called?

And also in Camera2Engine line 1640 and 1642 to see if it's called and if we get into the `if` condition.

comradewalker, 2020-01-11:
Sure, here you are:
AS logcat:
[Camera3logcatAS.log](https://github.com/natario1/CameraView/files/4049333/Camera3logcatAS.log)
CameraLogger.registerLogger:
[camera3fileOutput.log](https://github.com/natario1/CameraView/files/4049335/camera3fileOutput.log)

This is v2.6.0 source code

natario1, 2020-01-11:
Thanks, this is what I was looking for. Unfortunately I'm not able to figure out what's wrong 😕 I don't know why Full2PictureRecorder.onImageAvailable is not called in your case.

comradewalker, 2020-01-13:
No problem man, when I catch some tome I'll try to investigate that myself.
Btw, thanks for the nice lib and all the hard work you've put in it.

closed 2020-02-09 by stale[bot]

comradewalker, 2020-09-14:
@natario1 Sorry I couldn't return to this earlier, I finally managed to spare some time for this. My camera2 api knowledge is really bad so can you please hint me where should look for. I'm aware of CameraViewEngine thead and FallbackCameraThread (on this thread should be called Full2PictureRecorder.onImageAvailable()).

When I add this 2 Sleep lines in CameraBaseEngine.takePicture() (lines 506-527), Full2PictureRecorder.onImageAvailable is called:

`@Override
    public /* final */ void takePicture(final @NonNull PictureResult.Stub stub) {
        // Save boolean before scheduling! See how Camera2Engine calls this with a temp value.
        final boolean metering = mPictureMetering;
        try { Thread.sleep(500); } catch (InterruptedException ignore) {}
        getOrchestrator().scheduleStateful("take picture", CameraState.BIND,
                new Runnable() {
            @Override
            public void run() {
                LOG.i("takePicture:", "running. isTakingPicture:", isTakingPicture());
                if (isTakingPicture()) return;
                if (mMode == Mode.VIDEO) {
                    throw new IllegalStateException("Can't take hq pictures while in VIDEO mode");
                }
                stub.isSnapshot = false;
                stub.location = mLocation;
                stub.facing = mFacing;
                stub.format = mPictureFormat;
                onTakePicture(stub, metering);
            }
        });
        try { Thread.sleep(1000); } catch (InterruptedException ignore) {}
    }`

Shorter sleeps won't work. 

Btw, when metering is off, flash wont work at all. Is this expected behaviour?
This issue is happening on Samsung galaxy s10+, but strangely enough not on wife's Samsung A50 and obviously on lots of other devices. Thank you in advance for any help.

natario1, 2020-09-14:
> Btw, when metering is off, flash wont work at all. Is this expected behaviour?

Yes! Flash is part of the metering procedure.

> can you please hint me where should look for

This sleep workaround doesn't give me any new idea, sorry... If you believe there's something wrong going on with threads, you can use the Android Studio profiler to see if threads are busy and why.