natario1/CameraView #737: I am unable to make video. It is showing bug . Kindly reslove

kiran4228, 2020-01-08:
java.lang.RuntimeException: No encoders for type:video/avc
	at com.otaliastudios.cameraview.f.b.a(SourceFile:15)
	at com.otaliastudios.cameraview.f.b.<init>(SourceFile:4)
	at com.otaliastudios.cameraview.l.h.a(SourceFile:27)
	at com.otaliastudios.cameraview.j.j$a.onDrawFrame(SourceFile:18)
	at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1575)
	at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1271)

HBiSoft, 2020-01-10:
Asked by @natario1 in this question  - https://github.com/natario1/CameraView/issues/640#issuecomment-542073934

> Can you post logs from the demo app? From start to crash.

natario1, 2020-01-10:
Latest version should fix this as much as possible! However I'll keep this issue to change the behavior from crash to a controlled CameraException.

kiran4228, 2020-01-11:
When it will resolve. I am unable use. Kindly help

kiran4228, 2020-01-18:
> behavior

Update came but not resolved. Kindly help. It's urgent

Landi29, 2021-01-09:
@natario1 I will try to solve this issue if it is ok with you. Does the issue also need exception handling, or should I just return a CameraException?

Landi29, 2021-01-09:
It seems that ` tryConfigureVideo()` and `tryConfigureAudio()` in `DeviceEncoders.java` both throw the corresponding exceptions, so I don't see the problem.