natario1/CameraView #739: When I switch to video and switch back to the camera, the camera preview stops.

birkanfirat, 2020-01-09:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.6.0*
- Camera engine used: *camera1*
- Reproducible in official demo app: *no*
- Device / Android version: *Pixel, API 28*

### Expected behavior
Hi, I have an instagram-like screen in my application, gallery, photo and video transitions are done on a TabActivity.
When I updated to your latest version, I encountered a problem I had not seen before.
When the user selects the camera, **Cameraview** opens without any inconvenience, then the video preview works smoothly when the user switches to the video section, but when the user switches from the video to the camera, the camera preview stops and does not work. This issue did not occur in your older versions.

**Note:** I can take pictures when the camera stops, only preview does not work.

I have an attached video, hopefully it was descriptive.
### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

PhotoActivity
```xml
     <com.otaliastudios.cameraview.CameraView
                    android:id="@+id/camera"
                    android:layout_width="0dp"
                    android:layout_height="0dp"
                    android:layout_gravity="center"
                    android:keepScreenOn="true"
                    app:cameraAudio="on"
                    app:cameraFacing="back"
                    app:cameraFlash="off"
                    app:cameraGestureLongTap="none"
                    app:cameraGesturePinch="zoom"
                    app:cameraGestureScrollHorizontal="exposureCorrection"
                    app:cameraGestureScrollVertical="none"
                    app:cameraGestureTap="autoFocus"
                    app:cameraGrid="off"
                    app:cameraEngine="camera1"
                    app:cameraPlaySounds="true"
                    app:cameraMode="picture"
                    app:cameraRequestPermissions="false"
                    app:layout_constraintDimensionRatio="1"
                    app:layout_constraintLeft_toLeftOf="parent"
                    app:layout_constraintRight_toRightOf="parent"
                    app:layout_constraintTop_toTopOf="parent" />

```
VideoActivity
```xml
      <com.otaliastudios.cameraview.CameraView
                android:id="@+id/cameraVideo"
                android:layout_width="0dp"
                android:layout_height="0dp"
                android:layout_gravity="center"
                android:keepScreenOn="true"
                app:cameraAudio="on"
                app:cameraFacing="back"
                app:cameraFlash="off"
                app:cameraGestureLongTap="none"
                app:cameraGesturePinch="zoom"
                app:cameraGestureScrollHorizontal="exposureCorrection"
                app:cameraGestureScrollVertical="none"
                app:cameraGestureTap="autoFocus"
                app:cameraGrid="off"
                app:cameraEngine="camera1"
                app:cameraMode="video"
                app:cameraPlaySounds="true"
                app:cameraVideoSizeAspectRatio="1:1"
                app:cameraVideoSizeBiggest="true"
                app:cameraWhiteBalance="auto"
                app:cameraRequestPermissions="false"
                app:layout_constraintDimensionRatio="1"
                app:layout_constraintLeft_toLeftOf="parent"
                app:layout_constraintRight_toRightOf="parent"
                app:layout_constraintTop_toTopOf="parent" />
```

### Video
[video-output-5F76325B-CDDF-4376-80CF-F81CD5B60D06.MP4.zip](https://github.com/natario1/CameraView/files/4040847/video-output-5F76325B-CDDF-4376-80CF-F81CD5B60D06.MP4.zip)

### Logs
[cameraLog.zip](https://github.com/natario1/CameraView/files/4040778/cameraLog.zip)

ufukhawk, 2020-01-09:
I have this problem. :/

ghafarzadeh, 2020-01-10:
I have this problem.

esrefviduslu, 2020-01-10:
I have this problem

natario1, 2020-01-10:
@birkanfirat thanks for reporting! Unfortunately I don't see any issue in logs. They are hard to read because you have two CameraView instances and also I don't know when the clicks happen.

Anyway please just make sure that `cameraView1.destroy()` is always called before `cameraView2.open()`. We don't support multiple instances at the same time.

birkanfirat, 2020-01-11:
Hello @natario1 

videoActivity:
override fun onResume() {
		super.onResume()
		isCapturing = false
		designer.videoCamera.open()
	}

	override fun onPause() {
		super.onPause()
		isCapturing = false
		designer.videoCamera.close()
	}

	override fun onDestroy() {
		super.onDestroy()
		isCapturing = false
		designer.videoCamera.destroy()
	}

PhotoActivity:
override fun onResume() {
		super.onResume()
		designer.camera.open()
	}

	override fun onPause() {
		super.onPause()
		designer.camera.close()
	}

	override fun onDestroy() {
		super.onDestroy()
		designer.camera.destroy()
	}

If I do destroy on pause, I cannot run it because the view is not recompiled, because it remains on the gray screen because I use TabActivity and it does not re-create views after the first click. :(

birkanfirat, 2020-01-11:
Hello Again @natario1 ,
I would be very happy if you can review the sharing I did in a simple app where this case occurs

      https://github.com/birkanfirat/exampleApp

natario1, 2020-01-11:
What I see there is that you call open() on a second instance before destroying the first one, which is something we don't support. I suggest you just use a single UI and switch between `camera.setMode()`, or you keep it as is but you find a way to ensure the other view is properly destroyed (e.g. a `static WeakReference`).

birkanfirat, 2020-01-11:
@natario1 When I switch between activities, I do close () in the pause method of the running activity before open (), but I think it's not enough. I was using this way before Camerav2.0 did not have any problems

zCanOzseven, 2020-01-11:
i also have same issues . it was working to  previous version . something happened with this version. i hope , you would help us ..

natario1, 2020-01-11:
It's not enough, you must `destroy()` before any other `open()`. I don't know if this was working before but I don't plan to support it anymore, sorry. Anyway TabActivity is deprecated since 10 years ago so you might consider using another solution.