natario1/CameraView #740: Is enough to call cameraView.close() to stop call FrameProcessor.process() ?

alexei-28, 2020-01-09:
In my android application I use **FrameProcessor** to pass frame to Firebase Machine Learning Kit.
Nice. It's work fine.

in app/build.gradle:

`implementation "com.otaliastudios:cameraview:2.6.0"`

Here snippet:

xml layout:

```
 <com.otaliastudios.cameraview.CameraView
            android:id="@+id/cameraView"
            android:layout_width="0dp"
            android:layout_height="0dp"
            android:keepScreenOn="true"
            app:cameraAudio="off"
            app:cameraMode="picture"
            app:cameraPictureFormat="jpeg"
            app:cameraRequestPermissions="false"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintEnd_toEndOf="parent"
            app:layout_constraintStart_toStartOf="parent"
            app:layout_constraintTop_toBottomOf="@+id/profileDetailsToolbar" />

```

```
class ScanCheckActivity : AppCompatActivity() {

 override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        Debug.d(TAG, "onCreate")
        dataBinding = DataBindingUtil.setContentView(this, R.layout.scan_check_activity)

        dataBinding.cameraView.setLifecycleOwner(this)
        dataBinding.cameraView.addFrameProcessor(object : FrameProcessor {

            override fun process(frame: Frame) {
                Debug.d(TAG, "process_start: frame = $frame")
                scanCheckViewModel.doRecognizeCheck(frame)
            }
        })
    }

 override fun onResume() {
        super.onResume()
        Debug.d(TAG, "onResume: ")
        dataBinding.cameraView.open()
    }

    override fun onPause() {
        super.onPause()
        Debug.d(TAG, "onPause:")
        dataBinding.cameraView.close()
    }

    override fun onDestroy() {
        super.onDestroy()
        Debug.d(TAG, "onDestroy: ")
        dataBinding.cameraView.destroy();
    }
}
```

I need when call `onPause()` to stop `FrameProcessor.process()` . And when call `onResume() `to start `FrameProcessor.process()`.
To do this I use (as you can see) two methods:
```
- cameraView.close()
- cameraView.open()
```
Questions:
1. Is enough to call `cameraView.close()/cameraView.open()` to stop/start call `FrameProcessor.process()` ?
2. Why need method `removeFrameProcessor()` ? Maybe I need to call this method to stop `FrameProcessor.process() `?

natario1, 2020-01-10:
You can remove onResume, onPause and onDestroy, because you already use setLifecycleOwner(). This is enough for frame processors to work correctly.

closed 2020-02-06 by stale[bot]

alexei-28, 2020-02-25:
After I forward to another activity the method 

`override fun process(frame: Frame) `

is still calling. How I can stop it?

`dataBinding.cameraView.setLifecycleOwner(this)` not help

alexei-28, 2020-02-26:
here my solution (by `Tasks.await`)
Android Studio 3.6

on activity

```
 dataBinding.cameraView.addFrameProcessor(object : FrameProcessor {
            override fun process(frame: Frame) {
                Debug.d(TAG, "process_start: call_doRecognizeCheck()")
                scanCheckViewModel.doRecognizeCheck(frame)
            }
        })
```

on viewModel:

```
fun doRecognizeCheck(frame: Frame) {
       // some logic here
       runDetector(firebaseVisionImage)
    }

private fun runDetector(firebaseVisionImage: FirebaseVisionImage) {
        val detector = FirebaseVision.getInstance().onDeviceTextRecognizer
        try {
            Tasks.await(detector.processImage(firebaseVisionImage)
                .addOnSuccessListener { firebaseVisionText ->
               // some logic here
        } catch (ex: Exception) { 
            // Firebase task failed.           
        }
    }
```
and now when I forward to another activity then method 

`override fun process(frame: Frame)`

stop calling. 

Nice. It's exactly what I need.