natario1/CameraView #742: "EXCEPTION:", "Got CameraException. Dispatching to callback."

wy799172467, 2020-01-10:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: 2.6.0
- Camera engine used: both
- Reproducible in official demo app: yes
- Device / Android version: HUAWEI MATE PRO 10, API 28

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. select  video mode
2. Click on record
3. See error

### Expected behavior
record a video.

### Screenshots
(http://bossfs.sammbo.com/0/1/A_300099/2020/01/1_A_300099_E7158FC8E74C6519A2F8A33065AA6750.jpg?x-oss-process=image/resize,s_720)

### Logs
com.otaliastudios.cameraview.video.Full2VideoRecorder$PrepareException: java.io.IOException: prepare failed.

natario1, 2020-01-10:
Hi, we need complete logs to understand if there is something wrong or not.

iskugor, 2020-01-13:
@natario1 I can see the issue Samsung A3, with Android 6.0.1.

Here are logs:

`
I/CameraDeviceState: Legacy camera service transitioning to state IDLE
I/CameraEngine: onStartBind: Completed
I/CameraOrchestrator: ENGINE >> BIND - Finished.
    BIND >> PREVIEW - Executing.
I/CameraEngine: onStartPreview: Dispatching onCameraPreviewStreamSizeChanged.
I/CameraCallbacks: onCameraPreviewStreamSizeChanged: swallowing because the preview size has not changed. 720x960
I/CameraPreview: setStreamSize: desiredW= 720 desiredH= 960
I/CameraEngine: onStartPreview: Starting preview.
I/RequestQueue: Repeating capture request set.
I/CameraEngine: onStartPreview: Started preview.
    CameraCaptureSession.StateCallback reported onReady.
W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
    Only received metering rectangles with weight 0.
    convertRequestToMetadata - Ignoring android.lens.focusDistance false, only 0.0f is supported
D/ViewRootImpl: ViewPostImeInputStage processPointer 0
D/ViewRootImpl: #3 mView = null
D/ViewRootImpl: ViewPostImeInputStage processPointer 1
W/DemoApp: Recording for 5 seconds...
D/TextView: setTypeface with style : 0
I/CameraOrchestrator: TAKE VIDEO - Scheduling.
D/ViewRootImpl: #1 mView = android.widget.LinearLayout{3ba8d0c V.E...... ......I. 0,0-0,0}
D/mali_winsys: new_window_surface returns 0x3000,  [431x88]-format:1
D/ViewRootImpl: MSG_RESIZED_REPORT: ci=Rect(0, 0 - 0, 0) vi=Rect(0, 0 - 0, 0) or=1
D/libEGL: eglInitialize EGLDisplay = 0xde3f72f4
I/CameraDeviceState: Legacy camera service transitioning to state CAPTURING
D/libEGL: eglInitialize EGLDisplay = 0xee97a354
    eglTerminate EGLDisplay = 0xee97a3b4
D/gralloc: gralloc_lock_ycbcr success. format : 11, usage: 30, ycbcr.y: 0xdb34f000, .cb: 0xdb39a001, .cr: 0xdb39a000, .ystride: 640 , .cstride: 640, .chroma_step: 2
V/CameraEngine: onImageAvailable: trying to acquire Image.
D/gralloc: gralloc_lock_ycbcr success. format : 11, usage: 3, ycbcr.y: 0xdb34f000, .cb: 0xdb39a001, .cr: 0xdb39a000, .ystride: 640 , .cstride: 640, .chroma_step: 2
I/CameraEngine: onImageAvailable: Image acquired in wrong state. Closing it now.
D/libEGL: eglInitialize EGLDisplay = 0xde3f72f4
V/CameraEngine: aeMode: 2 aeLock: false aeState: null aeTriggerState: null afState: 0 afTriggerState: null
I/CameraOrchestrator: BIND >> PREVIEW - Finished.
    TAKE VIDEO - Executing.
I/CameraEngine: takeVideo: running. isTakingVideo: false
I/CameraEngine: onTakeVideo called.
W/CameraEngine: onTakeVideo calling restartBind.
I/CameraEngine: RESTART BIND: scheduled. State: PREVIEW
I/CameraOrchestrator: PREVIEW << BIND - Scheduling.
    BIND << ENGINE - Scheduling.
    ENGINE >> BIND - Scheduling.
    BIND >> PREVIEW - Scheduling.
    TAKE VIDEO - Finished.
    PREVIEW << BIND - Executing.
I/CameraEngine: onStopPreview: Started.
I/FrameManager: release: Clearing the frame and buffer queue.
I/CameraEngine: onStopPreview: Returning.
I/CameraOrchestrator: PREVIEW << BIND - Finished.
    BIND << ENGINE - Executing.
I/CameraEngine: onStopBind: About to clean up.
I/RequestQueue: Repeating capture request cancelled.
I/CameraEngine: onStopBind: Returning.
I/CameraOrchestrator: BIND << ENGINE - Finished.
    ENGINE >> BIND - Executing.
I/CameraEngine: onStartBind: Started
    computeCaptureSize: result: 720x960 flip: true mode: VIDEO
I/CameraEngine: computePreviewStreamSize: targetRatio: 3:4 targetMinSize: 708x944
I/CameraEngine: computePreviewStreamSize: result: 960x720 flip: true
D/libEGL: eglInitialize EGLDisplay = 0xde3f72f4
W/art: Suspending all threads took: 10.155ms
I/art: Background partial concurrent mark sweep GC freed 21993(924KB) AllocSpace objects, 1(1204KB) LOS objects, 46% free, 4MB/8MB, paused 11.166ms total 54.738ms
I/FullVideoRecorder: prepareMediaRecorder: Preparing on thread Thread[CameraViewEngine,5,main][15115]
E/Legacy-CameraDevice-JNI: LegacyCameraDevice_nativeGetSurfaceId: Could not retrieve native Surface from surface.
E/CameraDeviceGLThread-0: Received exception on GL render thread: 
[... 316 lines cut ...]
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
        at com.google.android.gms.tasks.zzj.run(Unknown Source)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler$2.execute(WorkerHandler.java:112)
        at com.google.android.gms.tasks.zzi.onComplete(Unknown Source)
        at com.google.android.gms.tasks.zzr.zza(Unknown Source)
        at com.google.android.gms.tasks.zzu.trySetResult(Unknown Source)
        at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(Unknown Source)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
[... cut ...]
I/Choreographer: Skipped 196 frames!  The application may be doing too much work on its main thread.
I/CameraCallbacks: dispatchOnCameraClosed
W/DemoApp: Got CameraException #3
D/TextView: setTypeface with style : 0
D/ViewRootImpl: #3 mView = null
    #1 mView = android.widget.LinearLayout{afe655 V.E...... ......I. 0,0-0,0}
D/mali_winsys: new_window_surface returns 0x3000,  [410x88]-format:1
D/ViewRootImpl: MSG_RESIZED_REPORT: ci=Rect(0, 0 - 0, 0) vi=Rect(0, 0 - 0, 0) or=1
D/ViewRootImpl: #3 mView = null
`

closed 2020-02-15 by stale[bot]

lovkush, 2021-11-19:
Trying to run demo app. Just cloned the repository and installed demo on my device

Got CameraException2 

2021-11-19 10:07:11.134 11671-11700/com.otaliastudios.cameraview.demo I/CameraEngine: onStartBind: Started
2021-11-19 10:07:11.135 11671-11700/com.otaliastudios.cameraview.demo I/CameraEngine: computeCaptureSize: result: 3016x4032 flip: true mode: PICTURE
2021-11-19 10:07:11.152 11671-11700/com.otaliastudios.cameraview.demo I/CameraEngine: computePreviewStreamSize: targetRatio: 377:504 targetMinSize: 720x1001
2021-11-19 10:07:11.155 11671-11700/com.otaliastudios.cameraview.demo I/CameraEngine: computePreviewStreamSize: result: 4032x3016 flip: true
2021-11-19 10:07:11.166 11671-11700/com.otaliastudios.cameraview.demo W/CameraDevice-JV-0: Stream configuration failed due to: endConfigure:505: Camera 0: Unsupported set of inputs/outputs provided
2021-11-19 10:07:11.167 11671-11700/com.otaliastudios.cameraview.demo E/CameraCaptureSession: Session 0: Failed to create capture session; configuration failed
2021-11-19 10:07:11.168 11671-11700/com.otaliastudios.cameraview.demo E/CameraEngine: onConfigureFailed! Session android.hardware.camera2.impl.CameraCaptureSessionImpl@ce27c72
2021-11-19 10:07:11.170 11671-11700/com.otaliastudios.cameraview.demo W/CameraOrchestrator: ENGINE >> BIND - Finished with ERROR. com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: onConfigureFailed! Session android.hardware.camera2.impl.CameraCaptureSessionImpl@ce27c72
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: onConfigureFailed! Session android.hardware.camera2.impl.CameraCaptureSessionImpl@ce27c72
        at com.otaliastudios.cameraview.engine.Camera2Engine$4.onConfigureFailed(Camera2Engine.java:596)
        at android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy.lambda$onConfigureFailed$1$CallbackProxies$SessionStateCallbackProxy(CallbackProxies.java:64)
        at android.hardware.camera2.impl.-$$Lambda$CallbackProxies$SessionStateCallbackProxy$gvbTsp9UPpKJAbdycdci_ZW5BeI.run(Unknown Source:4)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:223)
        at android.os.HandlerThread.run(HandlerThread.java:67)
     Caused by: java.lang.RuntimeException: onConfigureFailed! Session android.hardware.camera2.impl.CameraCaptureSessionImpl@ce27c72
        at com.otaliastudios.cameraview.engine.Camera2Engine$4.onConfigureFailed(Camera2Engine.java:594)
        at android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy.lambda$onConfigureFailed$1$CallbackProxies$SessionStateCallbackProxy(CallbackProxies.java:64) 
        at android.hardware.camera2.impl.-$$Lambda$CallbackProxies$SessionStateCallbackProxy$gvbTsp9UPpKJAbdycdci_ZW5BeI.run(Unknown Source:4) 
        at android.os.Handler.handleCallback(Handler.java:938) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:223) 
        at android.os.HandlerThread.run(HandlerThread.java:67) 
2021-11-19 10:07:11.170 11671-11700/com.otaliastudios.cameraview.demo E/CameraEngine: EXCEPTION: Scheduling on the crash handler...
2021-11-19 10:07:11.171 11671-11700/com.otaliastudios.cameraview.demo I/CameraOrchestrator: BIND >> PREVIEW - Executing.
2021-11-19 10:07:11.171 11671-11700/com.otaliastudios.cameraview.demo W/CameraOrchestrator: BIND >> PREVIEW - State mismatch, aborting. current: ENGINE from: BIND to: PREVIEW
2021-11-19 10:07:11.171 11671-11700/com.otaliastudios.cameraview.demo I/CameraOrchestrator: BIND >> PREVIEW - Finished because ABORTED.
2021-11-19 10:07:11.171 11671-11700/com.otaliastudios.cameraview.demo I/CameraOrchestrator: OFF >> ENGINE - Executing.
2021-11-19 10:07:11.172 11671-11700/com.otaliastudios.cameraview.demo W/CameraOrchestrator: OFF >> ENGINE - State mismatch, aborting. current: ENGINE from: OFF to: ENGINE
2021-11-19 10:07:11.172 11671-11700/com.otaliastudios.cameraview.demo I/CameraOrchestrator: OFF >> ENGINE - Finished because ABORTED.
2021-11-19 10:07:11.172 11671-11700/com.otaliastudios.cameraview.demo I/CameraOrchestrator: ENGINE >> BIND - Executing.

SweetD3v, 2021-12-10:
Facing the same error.
How I got this issue:

Open camera -> Switch to front camera -> Save the selected state in sharedpreference or wherever you like.
Then open the app again & you'll get the same error. If you're opening the front camera from the based on your last selection from sharedpreference, then it happens.

SweetD3v, 2021-12-10:
> Hi, we need complete logs to understand if there is something wrong or not.

Can you check as I mentioned in my comment pls? It's the steps to reproduce the error.