natario1/CameraView #743: com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.

wy799172467, 2020-01-10:
com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
        at com.otaliastudios.cameraview.engine.CameraBaseEngine.onVideoResult(CameraBaseEngine.java:635)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onVideoResult(Camera1Engine.java:434)
        at com.otaliastudios.cameraview.video.VideoRecorder.dispatchResult(VideoRecorder.java:138)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStop(FullVideoRecorder.java:339)
        at com.otaliastudios.cameraview.video.VideoRecorder.stop(VideoRecorder.java:98)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:300)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:80)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:391)
        at com.otaliastudios.cameraview.engine.CameraBaseEngine$5.run(CameraBaseEngine.java:571)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:75)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:72)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
        at com.google.android.gms.tasks.zzj.run(Unknown Source:4)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler$2.execute(WorkerHandler.java:112)
        at com.google.android.gms.tasks.zzi.onComplete(Unknown Source:10)
        at com.google.android.gms.tasks.zzr.zza(Unknown Source:19)
        at com.google.android.gms.tasks.zzu.trySetResult(Unknown Source:94)
        at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(Unknown Source:10)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)

wy799172467, 2020-01-12:
### Describe the bug

Please add a clear description of what the bug is, and fill the list below.

    CameraView version: 2.6.0
    Camera engine used: both
    Reproducible in official demo app: no
    Device / Android version: HUAWEI MATE PRO 10, API 28

### To Reproduce

### Steps to reproduce the behavior, possibly in the demo app:

    use the function:
    mCamera.setMode(Mode.VIDEO);
    mCamera.takeVideo(new File(FileAccessor.getVideoPathName()+"/aaa"));
    See error

### Expected behavior

record a video.

### Logs

com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
at com.otaliastudios.cameraview.engine.CameraBaseEngine.onVideoResult(CameraBaseEngine.java:635)
at com.otaliastudios.cameraview.engine.Camera1Engine.onVideoResult(Camera1Engine.java:434)
at com.otaliastudios.cameraview.video.VideoRecorder.dispatchResult(VideoRecorder.java:138)
at com.otaliastudios.cameraview.video.FullVideoRecorder.onStop(FullVideoRecorder.java:339)
at com.otaliastudios.cameraview.video.VideoRecorder.stop(VideoRecorder.java:98)
at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:300)
at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:80)
at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:391)
at com.otaliastudios.cameraview.engine.CameraBaseEngine$5.run(CameraBaseEngine.java:571)
at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:75)
at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:72)
at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
at com.google.android.gms.tasks.zzj.run(Unknown Source:4)
at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:137)
at com.otaliastudios.cameraview.internal.utils.WorkerHandler$2.execute(WorkerHandler.java:112)
at com.google.android.gms.tasks.zzi.onComplete(Unknown Source:10)
at com.google.android.gms.tasks.zzr.zza(Unknown Source:19)
at com.google.android.gms.tasks.zzu.trySetResult(Unknown Source:94)
at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(Unknown Source:10)
at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)

closed 2020-02-08 by stale[bot]

wadelzubair, 2021-07-05:
Same issue here : 
**Device** : Leneovo tablet TB-7304I
**Version :**  2.7.0
**Android:**  6.0
**Logs :** 

> E/CameraTakingFragment: onCameraError: 
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
        at com.otaliastudios.cameraview.engine.CameraBaseEngine.onVideoResult(CameraBaseEngine.java:670)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onVideoResult(Camera1Engine.java:468)
        at com.otaliastudios.cameraview.video.VideoRecorder.dispatchResult(VideoRecorder.java:138)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStop(FullVideoRecorder.java:359)
        at com.otaliastudios.cameraview.video.VideoRecorder.stop(VideoRecorder.java:98)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:320)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:80)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:426)
        at com.otaliastudios.cameraview.engine.CameraBaseEngine$5.run(CameraBaseEngine.java:605)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:84)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:81)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$3.run(CameraOrchestrator.java:152)
        at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.execute(CameraOrchestrator.java:147)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:34)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.run(CameraOrchestrator.java:137)
        at android.os.Handler.handleCallback(Handler.java:836)
        at android.os.Handler.dispatchMessage(Handler.java:103)
        at android.os.Looper.loop(Looper.java:203)
        at android.os.HandlerThread.run(HandlerThread.java:61)
     Caused by: java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:314)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:80) 
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:426) 
        at com.otaliastudios.cameraview.engine.CameraBaseEngine$5.run(CameraBaseEngine.java:605) 
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100) 
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:84) 
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:81) 
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$3.run(CameraOrchestrator.java:152) 
        at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137) 
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.execute(CameraOrchestrator.java:147) 
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:34) 
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.run(CameraOrchestrator.java:137) 
        at android.os.Handler.handleCallback(Handler.java:836) 
        at android.os.Handler.dispatchMessage(Handler.java:103) 
        at android.os.Looper.loop(Looper.java:203) 
        at android.os.HandlerThread.run(HandlerThread.java:61)

wadelzubair, 2021-07-05:
Solved for me by changing the VideoCodec for Leneovo Tablet 
jsut I change it to :
`cameraView.setVideoCodec(VideoCodec.H_263);`

MohammadWaqarYounas, 2024-05-03:
java.lang.RuntimeException: Error during glUniformMatrix4fv: glError 0x501: invalid value
    com.otaliastudios.opengl.core.Egloo.checkGlError(Egloo.java:63)
    com.otaliastudios.opengl.program.GlTextureProgram.onPreDraw(GlTextureProgram.java:42)
    com.otaliastudios.cameraview.filter.BaseFilter.onPreDraw(BaseFilter.java:42)
    com.otaliastudios.cameraview.filters.LomoishFilter.onPreDraw(LomoishFilter.java:0)
    com.otaliastudios.cameraview.filter.BaseFilter.draw(BaseFilter.java:20)
    com.otaliastudios.cameraview.filter.MultiFilter.draw(MultiFilter.java:81)
    com.otaliastudios.cameraview.internal.GlTextureDrawer.draw(GlTextureDrawer.java:66)
    com.otaliastudios.cameraview.preview.GlCameraPreview$Renderer.onDrawFrame(GlCameraPreview.java:96)
    android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1581)
    android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1280)

I am continuously getting this error. Is there any solution for it?