natario1/CameraView #746: Set accurate preview frame rate if possible

hualong-shen, 2020-01-13:
### Problem to be addressed
Now we have an option `setPreviewFrameRate`, internally the calling chain will go to `applyPreviewFrameRate`, the method gets available range options and checks whether the desired new frame rate is within any of the available ranges one by one. There may be more than one match, the method will only apply the first one meet the requirement. However, the first one may not be the best for me. For example, if I set the preview frame rate to 30, my assumption is that it can give me a stable 30 fps for the video. But when I test the fps is not stable. I tracked the code and I found the supported fps ranges are:
[15, 15]
[24, 24]
[7, 30]
[30, 30]
[10, 30]
So when i set preview frame rate to 30, [7, 30] will match and be set. Then the frame rate will vary between 7 and 30 by some factors (like brightness of environment). But what i really need is [30, 30].

### Describe the solution you'd like
In `applyPreviewFrameRate`, we can check whether there is a range that `lower == uppper == new frame rate`, if so then use it, if not then use the first one match.

### Additional context
I assume when developer uses this option what they really need if a stable precise frame rate.

hualong-shen, 2020-01-16:
Can we do this change? If so I'd like to create an PR.

natario1, 2020-01-16:
I'm not sure I agree... If you set 7-30 and 7 is being used, it's because higher values would result in underexposed preview. 

I'm OK to accept a PR if we add a boolean flag. `app:cameraPreviewFrameRateExact="true|false"` and `camera.setPreviewFrameRateExact(true|false)`.

If true, we sort fps ranges from largest to smallest and then take the first that matches.
If false, we sort fps ranges from smaller (possibly lower == upper) and then again take the first that matches.

And in my opinion this should default to `false` for the reasons stated above, ensuring flexibility of exposure. What do you think?

hualong-shen, 2020-01-16:
Thanks for quick reply @natario1 , it makes sense. We can't guarantee the frame rate even one setPreviewFrameRate, we can only try our best. I'll implement and raise an PR.

hualong-shen, 2020-01-16:
Hi thanks for quick reply @natario1 , I agree with adding a flag. However sorting the range not always guarantee you get the exact range (lower == higher). Like the example ranges i posted above.
If i sort from smallest to largest:
[15, 15]
[24, 24]
[7, 30]
[10, 30]
[30, 30]
In this case if I setPreviewFrameRate(24), i can get what I want. However if i setPreviewFrameRate(30), I will get [7, 30] which is not what i want.

If I sort from largest to smallest:
[30, 30]
[10, 30]
[7, 30]
[24, 24]
[15, 15]
In this case, if I setPreviewFrameRate(30), i can get what i want. However if i setPreviewFrameRate(24), i will get [10, 30] which is not what i want.

I think the best way is to scan all the ranges to look for whether there is a wanted range existing, if so then use it, if not then use the first one match.

natario1, 2020-01-16:
By small/large I mean to sort by the interval width. [15,15] [24,24], [30,30] have width 0 so they will appear first. [7,30] has width 23 so it will appear last.

hualong-shen, 2020-01-16:
Oh i got you, that's great, i'll implement it and raise an PR. thank you

closed 2020-02-05 by natario1

OmarMartinezMx, 2020-02-19:
When will this be available? I see it in master, I see it in the documentation, but not in version 2.6.1

natario1, 2020-02-19:
@OmarMartinezMx In next version, sorry for delay. In the meantime feel free to use jitpack to pull master-SNAPSHOT (check out their docs)

7colorapp, 2020-11-27:
support wide angle?